displaylink-rpm/displaylink-rpm #113: Error compiling from Source

mtjones1115, 2019-12-25:
I'm trying to compile this from the source for use with RHEL8 but when I do I get this error 

``warning: line 9: Possible unexpanded macro in: Version:	%{_version}
warning: line 10: Possible unexpanded macro in: Release:	%{_release}
error: Bad source: /home/USER/rpmbuild/SOURCES/v%{_version}.tar.gz: No such file or directory
``

Has anyone run into this before? This is my input: 
``$ rpmbuild -ba displaylink.spec``

elguero, 2020-02-10:
Not sure about support on RHEL8 yet.  The RPMs are usually built by running `make`.  There are some variables used by the .spec are defined through that.

What happens when you run `make`?

elguero, 2020-04-16:
Please see https://github.com/displaylink-rpm/displaylink-rpm/issues/122#issuecomment-598075630

sbluhm, 2020-09-08:
@mtjones1115 , I just saw this issue. You MUST create the rpm via "make"! The Makefile contains those parameters so rpmbuild can expand it. If you insist running rpmbuild manually, you have to define those macros via the --define flag. See the Makefile lines 119 to 130 and line 136.

I believe this issue can be closed as the rpm generation is not being used as intended.

elguero, 2020-09-10:
I am going to go ahead and close this out since there has been no response.

Thank you @sbluhm for your help.

closed 2020-09-10 by elguero