displaylink-rpm/displaylink-rpm #114: 32bit package requires 64bit libs

insaner, 2019-12-31:
rpm -Uvh fedora-30-displaylink-1.6.3-1.i386.rpm 

error: Failed dependencies:
        libc.so.6()(64bit) is needed by displaylink-1.6.3-1.i386
        libc.so.6(GLIBC_2.14)(64bit) is needed by displaylink-1.6.3-1.i386
        libc.so.6(GLIBC_2.2.5)(64bit) is needed by displaylink-1.6.3-1.i386

elguero, 2020-02-10:
@insaner Please paste the output from running `uname -a`.

insaner, 2020-02-10:
Unfortunately, that might not be so useful, as I am running a 64bit kernel in 32bit userland (as I try to migrate to pure 64bit).

$ uname -rm
5.2.7-100.fc29.x86_64   x86_64

insaner, 2020-02-10:
For reference, I did eventually get it installed, but as you can see, it's because I installed the 64bit glibc too.

$ rpm -q displaylink glibc
displaylink-1.6.3-1.i386
glibc-2.29-9.fc30.i686
glibc-2.29-9.fc30.x86_64

elguero, 2020-02-10:
Hmm.  That is an interesting setup.  I am seeing Fedora Core 29 kernel but you have glibc for Fedora 30 installed with Xorg 32 bit.

I am not sure if we can really support this type of setup.  Is it okay to close this issue?  It seems that you figured out how to get to work for your particular setup.

insaner, 2020-02-10:
Upto you, the problem was with trying to install the 32bit displaylink without glibc 64bit. I unfortunately can no longer uninstall glibc 64bit without likely breaking other things. I'm going to be doing some stuff with VM's, and might be able to give this a further test down the road if you're interested in leaving this open (in case it affects someone else later).

closed 2020-10-26 by elguero