displaylink-rpm/displaylink-rpm #115: Build failed on 5.4.7 due to missing Documentaion/Kconfig

Heidistein, 2020-01-06:
I already updated the makefile to build 1.6.4, which went without problem, but this the 'Documentation/Kconfig' is missing, says make:

```
DKMS make.log for evdi-1.6.4 for kernel 5.4.7-200.fc31.x86_64 (x86_64)
ma  6 jan 2020 10:45:55 CET
make KBUILD_VERBOSE=1 SUBDIRS=/var/lib/dkms/evdi/1.6.4/build SRCROOT=/var/lib/dkms/evdi/1.6.4/build CONFIG_MODULE_SIG= -C /lib/modules/5.4.7-200.fc31.x86_64/build modules
make[1]: Entering directory '/usr/src/kernels/5.4.7-200.fc31.x86_64'
make -f ./Makefile syncconfig
make -f ./scripts/Makefile.build obj=scripts/basic
rm -f .tmp_quiet_recordmcount
make -f ./scripts/Makefile.build obj=scripts/kconfig syncconfig
scripts/kconfig/conf  --syncconfig Kconfig
Kconfig:34: can't open file "Documentation/Kconfig"
make[3]: *** [scripts/kconfig/Makefile:73: syncconfig] Error 1
make[2]: *** [Makefile:567: syncconfig] Error 2
make[1]: *** [Makefile:678: include/config/auto.conf.cmd] Error 2
make[1]: Leaving directory '/usr/src/kernels/5.4.7-200.fc31.x86_64'
make: *** [Makefile:22: all] Error 2
```

NathanDeckinga, 2020-01-06:
It looks like evdi isn't working on 5.4 yet (or at lest the fix isn't fully incorporated yet).  I guess I will have to pause on upgrading kernels for a little while and keep running the latest 5.3 for Fedora/KDE.

https://github.com/DisplayLink/evdi/issues/172

Heidistein, 2020-01-06:
Yes, I found that also indeed. For my college is running arch-linux and uses evdi without trouble on the 5.4 branch.
May I should have done this before, but patching + compiling by hand (as per instructions provided in that report) lets compile and work evdi as a charm. I am typing this on 5.4.8 with edvi 1.6.4.

Why I did not try it before is because the missing Documentation/Kconfig is a very different problem. I truly do not unserstand where it comes from, exectly. 
So, the patch works, untested in the dkms.

I have tried to test the patch with DKMS, which keeps burning because of the missing Kconfig.

teutonius, 2020-01-06:
Same hier afer update to 5.4.7 dkms rebuild of the modules failed. The Documentation/Kconfig can be solved like this https://bugzilla.redhat.com/show_bug.cgi?id=1478726 scroll to the very bottom.
Still build fails with various issues like.
Building module:
cleaning build area....(bad exit status: 2)
make -j8 KERNELRELEASE=5.4.7-200.fc31.x86_64 all INCLUDEDIR=/lib/modules/5.4.7-200.fc31.x86_64/build/include KVERSION=5.4.7-200.fc31.x86_64 DKMS_BUILD=1....(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.4.7-200.fc31.x86_64 (x86_64)
Consult /var/lib/dkms/evdi/1.6.3/build/make.log for more information.

abucodonosor, 2020-01-07:
My patch doesn't have any DKMS fix, we don't use DKMS but see: 

https://github.com/DisplayLink/evdi/issues/172#issuecomment-561964789

elguero, 2020-01-13:
For anyone that is looking to build the RPM, the following branch updates the build to apply the patches that have been contributed to fix the evdi compile issue on kernel 5.4.

https://github.com/elguero/displaylink-rpm/tree/evdi1.6.4-kernel5.4-patch

Someone beat me to creating a PR for updating to the evdi 1.6.4 (#117) and this branch was based off of that branch that I created a few months back.  Once the curent evdi 1.6.4 PR has been accepted, I can rebase and submit a PR which will fix this until evdi is released with the fixes.

NathanDeckinga, 2020-01-16:
Thanks for the fix and the update, Elguero.  Although I find it comical that someone beat you to the PR for the RPM when you are the one who submitted the fix in evdi.

elguero, 2020-01-16:
I just updated the branch.  It was brought to my attention that I failed to commit all the changes in order to build the RPMs.  Only the patches were added by mistake.

evdi 1.6.4 builds fine on kernels prior to 5.4.  If you look at the history, I made the original change on a different branch in November 2019 but I held off due to the fact that I don't like how Fedora 31 and displaylink has a bad lag on the remote monitor.  I wasn't sure if it was evdi 1.6.4 that was causing it. 
 I was hoping that another version of evdi or displaylink would be out to fix that by now.

PR #118 (2020-02-10): Apply patches to get evdi to compile and install on kernel 5.4.
The patches were provided by abucodonosor and severach.

See https://github.com/DisplayLink/evdi/issues/172#issuecomment-561394805
See https://github.com/DisplayLink/evdi/issues/172#issuecomment-561964789

Fixes #115, fixes #116
closed 2020-02-10 by elguero