displaylink-rpm/displaylink-rpm #116: Failed to start DisplayLink Manager Service Fedora 31, kernel-5.4.7-200.fc31.x86_64

thaarbach, 2020-01-09:
After an `sudo dnf update` the displaylink service failed to start.
Systeminformation on Lenovo ThinkPad T590
```
Architektur:                     x86_64
CPU Operationsmodus:             32-bit, 64-bit
Byte-Reihenfolge:                Little Endian
Adressgrößen:                    39 bits physical, 48 bits virtual
CPU(s):                          8
Liste der Online-CPU(s):         0-7
Thread(s) pro Kern:              2
Kern(e) pro Socket:              4
Sockel:                          1
NUMA-Knoten:                     1
Anbieterkennung:                 GenuineIntel
Prozessorfamilie:                6
Modell:                          142
Modellname:                      Intel(R) Core(TM) i7-8565U CPU @ 1.80GHz
```

```
Name         : kernel-headers
Version      : 5.4.7
Release      : 200.fc31
Architecture : x86_64
Size         : 4.8 M
Quelle       : kernel-headers-5.4.7-200.fc31.src.rpm
Repository   : @System
Aus Paketque : updates
Summary      : Header files for the Linux kernel for use by glibc
URL          : http://www.kernel.org/
Lizenz       : GPLv2
Description  : Kernel-headers includes the C header files that specify the interface
             : between the Linux kernel and userspace libraries and programs.  The
             : header files define structures and constants that are needed for
             : building most standard programs and are also needed for rebuilding the
             : glibc package.
```
journalctl -xe
```
Jan 09 09:26:10 N81 systemd[1]: displaylink.service: Scheduled restart job, restart counter is at 1.
Jan 09 09:26:10 N81 systemd[1]: Stopped DisplayLink Manager Service.
Jan 09 09:26:10 N81 audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=displaylink comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res>
Jan 09 09:26:10 N81 audit[1]: SERVICE_STOP pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=displaylink comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=>
Jan 09 09:26:10 N81 systemd[1]: Starting DisplayLink Manager Service...
Jan 09 09:26:10 N81 modprobe[13454]: modprobe: FATAL: Module evdi not found in directory /lib/modules/5.4.7-200.fc31.x86_64
Jan 09 09:26:10 N81 audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=displaylink comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res>
Jan 09 09:26:10 N81 systemd[1]: displaylink.service: Control process exited, code=exited, status=1/FAILURE
Jan 09 09:26:10 N81 systemd[1]: displaylink.service: Failed with result 'exit-code'.
Jan 09 09:26:10 N81 systemd[1]: Failed to start DisplayLink Manager Service.
```

Runnig the maschine with Kernel `5.3.16-300.fc31.x86_64` displaylink service starts as expected.

Any suggestions?

NathanDeckinga, 2020-01-10:
I don't have an answer for you - other than there is at least 1 known issue with 5.4.7 upstream: https://github.com/DisplayLink/evdi/issues/172
And it looks like Fedora 31 is now trying to install 5.4.8 as of this morning, so I'm going to wait until I see an update here before installing that.  I'm in the same boat you are currently.

abucodonosor, 2020-01-12:
Use patch from https://github.com/DisplayLink/evdi/issues/172#issuecomment-561394805
Also for DKMS on top of the patch, you'll need https://github.com/DisplayLink/evdi/issues/172#issuecomment-561964789

hrabalvojta, 2020-01-13:
I would like to add some additional logs:

uname -a
```
Linux xxxxxxxx 5.4.8-200.fc31.x86_64 #1 SMP Mon Jan 6 16:44:18 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```
Installation fail
```
sudo rpm -ivh /home/vhrabal/Downloads/fedora-31-displaylink-1.6.3-1.x86_64.rpm 
Verifying...                          ################################# [100%]
Preparing...                          ################################# [100%]
Updating / installing...
   1:displaylink-1.6.3-1              ################################# [100%]
Job for displaylink.service failed because the control process exited with error code.
See "systemctl status displaylink.service" and "journalctl -xe" for details.
warning: %post(displaylink-1.6.3-1.x86_64) scriptlet failed, exit status 1
warning: %triggerin(displaylink-1.6.3-1.x86_64) scriptlet failed, exit status 10
```
You can see service didn't start

journalctl -xe
```
Jan 13 14:35:26 xxxxxxxx modprobe[22096]: modprobe: FATAL: Module evdi not found in directory /lib/modules/5.4.8-200.fc31.x86_64
Jan 13 14:35:26 xxxxxxxx systemd[1]: displaylink.service: Control process exited, code=exited, status=1/FAILURE
```
You can clearly see that displaylink.service doesn't start because missing evdi module.

First build log was missing two packages flex and bison. 
```
sudo dnf install flex bison
```

Here I am including full evdi build log
[evdi-build.log](https://github.com/displaylink-rpm/displaylink-rpm/files/4053816/evdi-build.log)

cat /var/lib/dkms/evdi/1.6.3/build/make.log | grep -i error
```
make[4]: *** [Makefile:43: /usr/src/kernels/5.4.8-200.fc31.x86_64/tools/objtool/fixdep-in.o] Error 2
make[3]: *** [/usr/src/kernels/5.4.8-200.fc31.x86_64/tools/build/Makefile.include:5: fixdep] Error 2
make[2]: *** [Makefile:67: objtool] Error 2
make[1]: *** [Makefile:1754: tools/objtool] Error 2
make[1]: *** [Makefile:1300: _modbuiltin_arch/x86] Error 2
make: *** [Makefile:22: all] Error 2
```

elguero, 2020-01-13:
For anyone interested in getting the RPM to be built with the patches see https://github.com/displaylink-rpm/displaylink-rpm/issues/115#issuecomment-573764279

thaarbach, 2020-01-22:
@elguero thanks for your comment. Did I understand correctly that the problem can be solved by installing your patch? Or do you have to pay attention to anything? Would it be possible to write a small guide on how to get DisplayLink with the kernel 5.4.7 to work with your patch?

alhalama, 2020-01-22:
Everything needed is on this branch: https://github.com/elguero/displaylink-rpm/tree/evdi1.6.4-kernel5.4-patch

To build the RPM package with the patches applied for 5.4 and dkms, simply 
1: clone the repo:
`git clone https://github.com/elguero/displaylink-rpm.git`
2. change directory into the repo:
`cd displaylink-rpm`
3. checkout that branch:
`git checkout evdi1.6.4-kernel5.4-patch`
4. run make:
`make`

The RPM to install will be under either the i386 or x86_64 directory depending on installation.  Once the RPM is installed it should work.

elguero, 2020-01-22:
@alhalama  Thanks for answering @thaarbach .

thaarbach, 2020-01-23:
@alhalama Thanks for the description.
@elguero Thanks for the patch,  I'll try it out as soon as I have time.

thaarbach, 2020-01-24:
@elguero 
Now it works with Kernel-5.4.7-200.fc31.x86_64. The first time I tried it, I created it with 5.3.16-300.fc31.x86_64, which worked, but after restarting with 5.4.7, the screens remained black. After a rebuild with 5.4.7 it worked. Then I tried to boot 5.4.12-200.fc31, but unfortunately it doesn't work. I rebuilt it with 5.4.12, but that didn't work. Do you need a patch for every new kernel?

@alhalama thx again for the little guide.

wernermaes, 2020-01-27:
Will the github repo (ttps://github.com/displaylink-rpm/displaylink-rpm/releases) also contain updated rpms in the near futuru?

For example

fedora-31-displaylink-1.6.4-1.x86_64.rpm

Kind regards

Werner

wernermaes, 2020-01-27:
I'm also using kernel-5.4.12-200.fc31

wernermaes, 2020-01-27:
There is a new kernel kernel-5.4.13-201.fc31.x86_64.
I will check this one out and let you know

wernermaes, 2020-01-27:
Hello

It does not work with 5.4.13-201 either

usr/src/kernels/5.4.13-201.fc31.x86_64/tools/build/Makefile.build:37: /usr/src/kernels/5.4.13-201.fc31.x86_64/tools/build/Build.include: No such file or directory
make[5]: *** No rule to make target '/usr/src/kernels/5.4.13-201.fc31.x86_64/tools/build/Build.include'.  Stop.
make[4]: *** [Makefile:43: /usr/src/kernels/5.4.13-201.fc31.x86_64/tools/objtool/fixdep-in.o] Error 2
make[3]: *** [/usr/src/kernels/5.4.13-201.fc31.x86_64/tools/build/Makefile.include:5: fixdep] Error 2
make[2]: *** [Makefile:67: objtool] Error 2
make[1]: *** [Makefile:1754: tools/objtool] Error 2
Has something changed in kernel-devel?

# rpm -ql kernel-devel-5.4.13-201.fc31.x86_64 |grep Build.inc
# rpm -ql kernel-devel-5.3.16-300.fc31.x86_64 |grep Build.inc
/usr/src/kernels/5.3.16-300.fc31.x86_64/tools/build/Build.include

thaarbach, 2020-01-27:
@wernermaes same as compling with 5.4.12.

wernermaes, 2020-01-27:
> @wernermaes same as compling with 5.4.12.

That's right. Luckily I still have kernel 5.3.16 installed.

JetStreamVelocity, 2020-01-27:
On my primary system I have applied a `versionlock` on kernel 5.3.16 to ensure the kernel isn't updated to 5.4 till there is confirmation this is working with 5.4.x

References:
- [Versionlock - perl.com article](https://www.perl.com/article/prevent-harmful-updates-with-versionlock/)
- [Fedoramagazine article on versionlock](https://fedoramagazine.org/boot-earlier-kernel/)

csilva235, 2020-01-27:
Having issues with 5.4.13 kernel.  Attempted the fixes listed above, but I'm not really sure I was doing them correctly. I tried to run the patch as noted in another comment, but the system just hangs. 

Any idea when the rpm will be updated to include the fix so I can get my monitors back? 

Here's the install output and journalctl -xe

=====================================================================================================================================
 Package                         Architecture               Version                           Repository                        Size
=====================================================================================================================================
Installing:
 displaylink                     x86_64                     1.6.3-1                           @commandline                     3.6 M
Installing dependencies:
 dkms                            noarch                     2.7.1-2.fc31                      fedora                            75 k

Transaction Summary
=====================================================================================================================================
Install  2 Packages

Total size: 3.7 M
Total download size: 75 k
Installed size: 10 M
Is this ok [y/N]: y
Downloading Packages:
dkms-2.7.1-2.fc31.noarch.rpm                                                                         141 kB/s |  75 kB     00:00    
-------------------------------------------------------------------------------------------------------------------------------------
Total                                                                                                106 kB/s |  75 kB     00:00     
Running transaction check
Transaction check succeeded.
Running transaction test
Transaction test succeeded.
Running transaction
  Preparing        :                                                                                                             1/1 
  Installing       : dkms-2.7.1-2.fc31.noarch                                                                                    1/2 
  Running scriptlet: dkms-2.7.1-2.fc31.noarch                                                                                    1/2 
Created symlink /etc/systemd/system/multi-user.target.wants/dkms.service → /usr/lib/systemd/system/dkms.service.

  Installing       : displaylink-1.6.3-1.x86_64                                                                                  2/2 
  Running scriptlet: displaylink-1.6.3-1.x86_64                                                                                  2/2 
Job for displaylink.service failed because the control process exited with error code.
See "systemctl status displaylink.service" and "journalctl -xe" for details.
warning: %post(displaylink-1.6.3-1.x86_64) scriptlet failed, exit status 1

Error in POSTIN scriptlet in rpm package displaylink
warning: %triggerin(displaylink-1.6.3-1.x86_64) scriptlet failed, exit status 10

Error in <unknown> scriptlet in rpm package displaylink
  Verifying        : dkms-2.7.1-2.fc31.noarch                                                                                    1/2 
  Verifying        : displaylink-1.6.3-1.x86_64                                                                                  2/2 

Installed:
  displaylink-1.6.3-1.x86_64                                         dkms-2.7.1-2.fc31.noarch                                        

Complete!

$ journalctl -xe
-- A start job for unit displaylink.service has finished with a failure.
-- 
-- The job identifier is 35854 and the job result is failed.
Jan 27 08:11:57 lbkdz5m2 audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='un>
Jan 27 08:12:02 lbkdz5m2 systemd[1]: displaylink.service: Scheduled restart job, restart counter is at 14.
-- Subject: Automatic restarting of a unit has been scheduled
-- Defined-By: systemd
-- Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- Automatic restarting of the unit displaylink.service has been scheduled, as the result for
-- the configured Restart= setting for the unit.
Jan 27 08:12:02 lbkdz5m2 audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='un>
Jan 27 08:12:02 lbkdz5m2 audit[1]: SERVICE_STOP pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='uni>
Jan 27 08:12:02 lbkdz5m2 systemd[1]: Stopped DisplayLink Manager Service.
-- Subject: A stop job for unit displaylink.service has finished
[... 12 lines cut ...]
-- 
-- The job identifier is 35950.
Jan 27 08:12:02 lbkdz5m2 modprobe[19196]: modprobe: FATAL: Module evdi not found in directory /lib/modules/5.4.13-201.fc31.x86_64
Jan 27 08:12:02 lbkdz5m2 systemd[1]: displaylink.service: Control process exited, code=exited, status=1/FAILURE
-- Subject: Unit process exited
-- Defined-By: systemd
-- Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- An ExecStartPre= process belonging to unit displaylink.service has exited.
-- 
-- The process' exit code is 'exited' and its exit status is 1.
Jan 27 08:12:02 lbkdz5m2 systemd[1]: displaylink.service: Failed with result 'exit-code'.
-- Subject: Unit failed
-- Defined-By: systemd
-- Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- The unit displaylink.service has entered the 'failed' state with result 'exit-code'.
Jan 27 08:12:02 lbkdz5m2 systemd[1]: Failed to start DisplayLink Manager Service.
-- Subject: A start job for unit displaylink.service has failed
-- Defined-By: systemd
-- Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- A start job for unit displaylink.service has finished with a failure.
-- 
-- The job identifier is 35950 and the job result is failed.

elguero, 2020-01-27:
@wernermaes The evdi project is currently working on patches to fix its compatibility with 5.4.  Once they release a new evdi package, the build here will be updated which would result in a new release with the rpms available like normal.  Right now, everyone is trying to help each other out with temporary solutions while we wait for the official maintainers to get things fixed.

In the meantime, you can either do as @JetStreamVelocity suggested and just not update your kernel and stick with using the RPMs that have currently been released.  Or you can build your own updated RPM as was outlined in https://github.com/displaylink-rpm/displaylink-rpm/issues/116#issuecomment-577255842

@csilva235 If you follow the steps in https://github.com/displaylink-rpm/displaylink-rpm/issues/116#issuecomment-577255842, there is no need to apply any patches.  The repo that is referenced in that comment takes care of patching evdi during the RPM build.

JetStreamVelocity, 2020-01-27:
I noticed the following on your journal messages:

`Module evdi not found in directory /lib/modules/5.4.13-201.fc31.x86_64`

I wonder if you need to build and install the evdi module as well?? Or if `displaylink-rpm` does that for you. Someone with more knowledge here can confirm.
 
> Here's the install output and journalctl -xe
> 
> [install.txt](https://github.com/displaylink-rpm/displaylink-rpm/files/4117475/install.txt)
> 
> [journal.txt](https://github.com/displaylink-rpm/displaylink-rpm/files/4117478/journal.txt)

elguero, 2020-01-27:
@JetStreamVelocity The evdi patch changes the name of the module.  This is handled in the repo https://github.com/elguero/displaylink-rpm/tree/evdi1.6.4-kernel5.4-patch.

csilva235, 2020-01-27:
> @csilva235 If you follow the steps in [#116 (comment)](https://github.com/displaylink-rpm/displaylink-rpm/issues/116#issuecomment-577255842), there is no need to apply any patches. The repo that is referenced in that comment takes care of patching evdi during the RPM build.

I completely missed that reply. I tried the steps there and it seems to be working now. 

Much appreciated.

thaarbach, 2020-01-28:
I've pinned kernel to 5.4.7-200.fc31.x86_64 and use the patch from 

> For anyone interested in getting the RPM to be built with the patches see [#115 (comment)](https://github.com/displaylink-rpm/displaylink-rpm/issues/115#issuecomment-573764279)

Works without any problems on my system.

ssaavedra, 2020-02-05:
@elguero please see your email, you should be able to push code directly and review other PRs directly. I've been out of GitHub issues for a little while and I would appreciate the help (and it seems the community would too).

ssaavedra, 2020-02-05:
I would not call this an "official" release, but I suppose it could be useful to publish it under a "pre-release" tag.

elguero, 2020-02-05:
@ssaavedra Okay.  Cool.  Thanks.  Glad to help in anyway I can.

I agree that I wouldn't create an official release.  This is just an interim fix for those who need it.  It looks like the evdi project is already working on bringing the patches in to their next release.

egyas, 2020-02-07:
> Everything needed is on this branch: https://github.com/elguero/displaylink-rpm/tree/evdi1.6.4-kernel5.4-patch
>
> To build the RPM package with the patches applied for 5.4 and dkms, simply
> 1: clone [...]

You sir, are owed a beer my yours truly!  I have Fedora 31, running kernel 5.4.15-200.fc31.x86_64.  I have a shaved head (well, stubble honestly), yet I was still tearing my hair out trying to make the darned DisplayLink unit in the docking station I bought work!  Your instructions saved not only my precious stubble, but quite possibly the dock as well from a painful death being hurled out the window.  ROFL

Seriously though sir, good on you for this!  :)
[image]

thaarbach, 2020-02-07:
@egyas what have you done, to get DisplayLink working with 5.4.15-200.fc31.x86_64? Only following the instructions from @alhalama https://github.com/displaylink-rpm/displaylink-rpm/issues/116#issuecomment-577255842? 
With 5.4.12-200.fc31 i was not able to get it work.

alhalama, 2020-02-07:
@thaarbach What was meant by rebuilt in your [previous comment](https://github.com/displaylink-rpm/displaylink-rpm/issues/116#issuecomment-578231385)?  

If the instructions that I provided are followed and you have installed the resulting RPM, it is expected that dkms would rebuild the evdi module for each kernel automatically.  If that isn't happening maybe there is an issue with the dkms configuration on your machine.  What is the output of `dkms status`?

egyas, 2020-02-07:
> @egyas what have you done, to get DisplayLink working with 5.4.15-200.fc31.x86_64? Only following the instructions from @alhalama [#116 (comment)](https://github.com/displaylink-rpm/displaylink-rpm/issues/116#issuecomment-577255842)?
> With 5.4.12-200.fc31 i was not able to get it work.

@thaarbach, I followed the instructions EXACTLY as @alhalama wrote them, and as I copied in my post above.  When I did the actual install of the RPM, i did a ..
`cd x86_64
sudo dnf localinstall displaylink-1.6.4-2.x86_64.rpm`

It did the install, but then seemed to freeze in the process.  Everything else worked fine, but the install never dumped out w/ a completed status.  But I **DID** hear my sound pop out and back in, and my network connection reset (my VPN connection I was using at the time dropped).  Since both sound and network are also going through that dock I'm using, it kind of made sense to me.  After a few minutes I did a ctrl-c on the dnf install, which dumped me back to a shell prompt.

At that point, I just went into ARandR, and noticed that those ports (in my case DVI-I-2-2 & DVI-I-3-3 were now detected.  I enabled them, and viola, there were my 2nd and 3rd external monitors! Now the laptop display, and all three external monitors are working fine.  

Hope that helps.  :)

thaarbach, 2020-02-07:
@alhalama As I wrote it. :-)
I followed your instructions but had booted the kernel 5.3.16-300.fc31.x86_64. Then I restart and boot  the kernel Kernel-5.4.7-200.fc31.x86_64 and the displays remain dark again.
Then I copied x86_64 to x86_64_firstdry,  executed `dnf remove displaylink`
Then follow your instructions https://github.com/displaylink-rpm/displaylink-rpm/issues/116#issuecomment-577255842  again, reboot and it worked

elguero, 2020-02-07:
@thaarbach Was the evdi driver installed manually at some point?  I would suggest uninstalling the RPM.  Then clear out anything left over under '/var/lib/dkms/evdi'.  Reboot into the latest kernel.  Then install the latest RPM following the instructions outlined above in https://github.com/displaylink-rpm/displaylink-rpm/issues/116#issuecomment-577255842

I just cleaned up that branch and created a pull request to get this in here.  Then we can build the RPMS for release but I think we are going to do a pre-release tag since the evdi team is currently working on incorporating these patches into evdi which will then mean we need to revert the patches once that happens.

Just to give you something for comparison:
```
ls -la /var/lib/dkms/evdi
total 12
drwxr-xr-x. 3 root root 4096 Feb  6 11:33 .
drwxr-xr-x. 3 root root 4096 Jan 13 12:11 ..
drwxr-xr-x. 9 root root 4096 Feb  6 11:33 1.6.4
lrwxrwxrwx. 1 root root   35 Jan 17 13:59 kernel-5.4.10-200.fc31.x86_64-x86_64 -> 1.6.4/5.4.10-200.fc31.x86_64/x86_64
lrwxrwxrwx. 1 root root   35 Jan 21 08:59 kernel-5.4.12-200.fc31.x86_64-x86_64 -> 1.6.4/5.4.12-200.fc31.x86_64/x86_64
lrwxrwxrwx. 1 root root   35 Jan 27 09:12 kernel-5.4.13-201.fc31.x86_64-x86_64 -> 1.6.4/5.4.13-201.fc31.x86_64/x86_64
lrwxrwxrwx. 1 root root   35 Jan 31 13:52 kernel-5.4.14-200.fc31.x86_64-x86_64 -> 1.6.4/5.4.14-200.fc31.x86_64/x86_64
lrwxrwxrwx. 1 root root   35 Feb  3 10:07 kernel-5.4.15-200.fc31.x86_64-x86_64 -> 1.6.4/5.4.15-200.fc31.x86_64/x86_64
lrwxrwxrwx. 1 root root   35 Feb  6 11:33 kernel-5.4.17-200.fc31.x86_64-x86_64 -> 1.6.4/5.4.17-200.fc31.x86_64/x86_64
lrwxrwxrwx. 1 root root   34 Jan 13 12:11 kernel-5.4.8-200.fc31.x86_64-x86_64 -> 1.6.4/5.4.8-200.fc31.x86_64/x86_64
```

thaarbach, 2020-02-08:
> @thaarbach Was the evdi driver installed manually at some point? I would suggest uninstalling the RPM. Then clear out anything left over under '/var/lib/dkms/evdi'. Reboot into the latest kernel. Then install the latest RPM following the instructions outlined above in [#116 (comment)](https://github.com/displaylink-rpm/displaylink-rpm/issues/116#issuecomment-577255842)

@elguero to install stuff on my maschine i just use dnf. Okay will clean up my system. 

Thanks for help @elguero @alhalama

thaarbach, 2020-02-08:
@elguero now it looks good
```
🐧  ~ $ ls -la /var/lib/dkms/evdi
insgesamt 12
drwxr-xr-x. 3 root root 4096  8. Feb 09:44 .
drwxr-xr-x. 3 root root 4096  9. Jan 10:09 ..
drwxr-xr-x. 3 root root 4096  8. Feb 09:44 1.6.4
lrwxrwxrwx. 1 root root   35  8. Feb 09:44 kernel-5.4.17-200.fc31.x86_64-x86_64 -> 1.6.4/5.4.17-200.fc31.x86_64/x86_64
```
```
🐧  ~ $ systemctl status displaylink.service 
● displaylink.service - DisplayLink Manager Service
   Loaded: loaded (/usr/lib/systemd/system/displaylink.service; static; vendor preset: disabled)
   Active: active (running) since Sat 2020-02-08 09:51:42 CET; 3s ago
  Process: 4466 ExecStartPre=/sbin/modprobe evdi (code=exited, status=0/SUCCESS)
 Main PID: 4468 (DisplayLinkMana)
    Tasks: 14 (limit: 18750)
   Memory: 8.8M
      CPU: 28ms
   CGroup: /system.slice/displaylink.service
           └─4468 /usr/libexec/displaylink/DisplayLinkManager

Feb 08 09:51:42 N81.isys-software.de systemd[1]: Starting DisplayLink Manager Service...
Feb 08 09:51:42 N81.isys-software.de systemd[1]: Started DisplayLink Manager Service.
```
But I can only say on Monday whether it really works because my dock is in my office.

closed 2020-02-10 by elguero

thaarbach, 2020-02-10:
@elguero thx for your help. It works now with 5.4.17-200.fc31.x86_64.

PR #118 (2020-02-10): Apply patches to get evdi to compile and install on kernel 5.4.
The patches were provided by abucodonosor and severach.

See https://github.com/DisplayLink/evdi/issues/172#issuecomment-561394805
See https://github.com/DisplayLink/evdi/issues/172#issuecomment-561964789

Fixes #115, fixes #116