displaylink-rpm/displaylink-rpm #119: Error compiling rawhide

itblaked, 2020-02-25:
Hi, cloned the repo down and ran sudo make rawhide to pickup the patch for kernel **5.5.x** but it fails to compile and I'm looking for help to understand why.

I'm running **Fedora31, 5.5.5-200.fc31.x86_64**.

During the make I'm prompted to reapply patches which already seem applied, I chose the default no. 

I also noticed it doesn't ask about the new patch that's upstream in evdi devel for the 5.5 kernel...

This is the output I get:

```bash
sudo make rawhide 
Checking last upstream commit date...
make RELEASE="`        echo -e "2" | tr -d '[:space:]'`.rawhide.`	curl -s https://api.github.com/repos/DisplayLink/evdi/branches/devel -H "Accept: application/vnd.github.full+json" | grep date | head -1 | cut -d: -f 2- | sed s/[^0-9TZ]//g`" devel all
make[1]: Entering directory '/home/../Software/displaylink/displaylink-rpm'
git clone --depth 1 -b devel https://github.com/DisplayLink/evdi.git /var/tmp/evdi-1.6.4
Cloning into '/var/tmp/evdi-1.6.4'...
remote: Enumerating objects: 45, done.
remote: Counting objects: 100% (45/45), done.
remote: Compressing objects: 100% (45/45), done.
remote: Total 45 (delta 1), reused 7 (delta 0), pack-reused 0
Unpacking objects: 100% (45/45), done.
cd /var/tmp/evdi-1.6.4 && git pull
Already up to date.
tar -z -c -f v1.6.4.tar.gz -C /var/tmp evdi-1.6.4
rpmbuild -bb --define "_topdir `pwd`" --define "_sourcedir `pwd`" --define "_rpmdir `pwd`" --define "_specdir `pwd`" --define "_srcrpmdir `pwd`" --define "_buildrootdir `mktemp -d /var/tmp/displayportXXXXXX`" --define "_builddir `mktemp -d /var/tmp/displayportXXXXXX`" --define "_release 2.rawhide.20200213T112235Z" --define "_daemon_version 5.2.14" --define "_version 1.6.4" --define "_tmppath `mktemp -d /var/tmp/displayportXXXXXX`"  displaylink.spec --target=i386
Building target platforms: i386
Building for target i386
setting SOURCE_DATE_EPOCH=1581033600
Executing(%prep): /bin/sh -e /var/tmp/displayport75BmxM/rpm-tmp.JEYNHp
+ umask 022
+ cd /var/tmp/displayport1S2ZSt
+ cd /var/tmp/displayport1S2ZSt
+ rm -rf displaylink-1.6.4
+ /usr/bin/mkdir -p displaylink-1.6.4
+ cd displaylink-1.6.4
+ /usr/bin/gzip -dc /home/../Software/displaylink/displaylink-rpm/v1.6.4.tar.gz
+ /usr/bin/tar -xof -
+ STATUS=0
+ '[' 0 -ne 0 ']'
+ /usr/bin/chmod -Rf a+rX,u+w,g-w,o-w .
+ cd evdi-1.6.4
+ echo 'Patch #0 (evdi-all-in-one-fixes.patch):'
Patch #0 (evdi-all-in-one-fixes.patch):
+ /usr/bin/patch --no-backup-if-mismatch -p1 --fuzz=0
patching file Makefile
Reversed (or previously applied) patch detected!  Assume -R? [n] 
Apply anyway? [n] 
Skipping patch.
1 out of 1 hunk ignored -- saving rejects to file Makefile.rej
patching file library/Makefile
Hunk #1 FAILED at 2.
1 out of 1 hunk FAILED -- saving rejects to file library/Makefile.rej
patching file module/Makefile
Hunk #1 FAILED at 43.
1 out of 1 hunk FAILED -- saving rejects to file module/Makefile.rej
patching file module/evdi_drv.c
Hunk #1 FAILED at 81.
1 out of 1 hunk FAILED -- saving rejects to file module/evdi_drv.c.rej
patching file module/evdi_drv.h
Hunk #1 FAILED at 20.
Hunk #2 FAILED at 51.
Hunk #3 FAILED at 112.
3 out of 3 hunks FAILED -- saving rejects to file module/evdi_drv.h.rej
patching file module/evdi_gem.c
Reversed (or previously applied) patch detected!  Assume -R? [n] 
Apply anyway? [n] 
Skipping patch.
4 out of 4 hunks ignored -- saving rejects to file module/evdi_gem.c.rej
error: Bad exit status from /var/tmp/displayport75BmxM/rpm-tmp.JEYNHp (%prep)

RPM build errors:
    Bad exit status from /var/tmp/displayport75BmxM/rpm-tmp.JEYNHp (%prep)
make[1]: *** [Makefile:132: i386/displaylink-1.6.4-2.rawhide.20200213T112235Z.i386.rpm] Error 1
make[1]: Leaving directory '/home/../Software/displaylink/displaylink-rpm'
make: *** [Makefile:78: rawhide] Error 2
```
```

thaarbach, 2020-02-26:
Have the same issue. It seems it is the same error as in #116 descripted. Now with Kernel 5.5

```
Feb 26 09:26:33 N81 modprobe[6450]: modprobe: FATAL: Module evdi not found in directory /lib/modules/5.5.5-200.fc31.x86_64
Feb 26 09:26:33 N81 systemd[1]: displaylink.service: Control process exited, code=exited, status=1/FAILURE
```

itblaked, 2020-02-26:
Yeah, was trying to get a HP USB C dock to work with my Lenovo X1 Gen 7 laptop, didn't succeed. I did get the displaylink service running successfully by building from the evdi devel branch https://github.com/DisplayLink/evdi, I could see displaylink listed in lsusb, but i couldn't see the 2 monitors connected to the dock via display settings in gnome or xrandr so removed it all and was kind of hoping this might work here instead... :(

alhalama, 2020-02-27:
When I installeda 5.5 kernel, I went back to the last commit before the patch (316206f) and then did a rawhide build.  These are the steps you can take to do that:

1. clone the repo: `git clone https://github.com/displaylink-rpm/displaylink-rpm.git`
2. change directory into the repo: `cd displaylink-rpm/`
3. check out commit 316206f: `git checkout 316206f -b withoutPatch`
4. build the rpm: `make rawhide`

The patches aren't necessary for the rawhide build as the changes are already done in the devel branch of evdi.

elguero, 2020-03-05:
Once evdi releases a new version, the current patches will be removed from the RPM spec.  If you are trying to work with the latest development branches, please do as @alhalama outlined for now.

kowalcj0, 2020-03-10:
Thanks @alhalama :+1: :muscle: 
Your instructions worked for me and I'm on Fedora 31 running `5.5.7-200.fc31.x86_64` kernel 

PS. will I have to rebuild and reinstall DisplayLink driver every single time kernel gets updated?

alhalama, 2020-03-10:
DKMS will build evdi for the new kernel as long as there aren't changes in the kernel that cause the build to fail.  In the case of the build failing then then updates to evdi would be needed.  Hopefully when that happens evdi has the updates and displaylink has already released the relevant updates so that we can just release a new RPM here so that you don't actually need to rebuild the displaylink-rpm yourself.

teutonius, 2020-03-12:
Instruction worked for me too Thx.

SergeyKons, 2020-03-17:
Maybe it's good idea to add these instruction to readme?

cbisnett, 2020-04-01:
I've been using the `v5.2.14-3-rc1` pre-release with the 5.4 kernel successfully. Upgrading to the 5.5 kernel caused the evdi module to fail when loading with `modprobe`. I found that building and installing evdi from master (https://github.com/DisplayLink/evdi/commits/b5c57910e4ef2b706249ddefd066f955e3899b8c) loaded and restored functionality.

It looks like the DisplayLink team hasn't cut a new release in a while, but it may make sense to cut a new RPM release using the newest upstream fixes.

alhalama, 2020-04-01:
@elguero I don't think these instructions should be in the readme though it might be better if rawhide builds worked again.  Maybe we should consider removing the patches before the next release from the DisplayLink team so that rawhide builds work for anyone that just clones the repo.  Note that I think we should probably wait until the official DisplayLink release before releasing a new RPM here and leave rawhide builds as the recommended approach for anyone looking to use a version of evdi that isn't officially released yet.

pabloalgo, 2020-04-01:
I was following the instructions without luck after installing the dependencies before "make rawhide" it looks that t installation was ok, but I don't have any signal on display, it installs but  say "Conflicts: xorg-x11-server-Xorg = 1.20.1" my kernel is 5.5.11-200.fc31.x86_64

After install it I have reboot the system and try it on gnome on wayland but I didn't see the display

by the way, I also got the typical issue with rmp packet for fedora 31

 displaylink.service - DisplayLink Manager Service
   Loaded: loaded (/usr/lib/systemd/system/displaylink.service; static; vendor preset: disabled)
   Active: activating (auto-restart) (Result: exit-code) since Wed 2020-04-01 19:11:22 CEST; 3s ago
  Process: 6360 ExecStartPre=/sbin/modprobe evdi (code=exited, status=1/FAILURE)

I am a newly with git clone, could someone help me ? I could install on Windows and ubuntu but just working on gnome and was a bit buggy some times and I prefer very much work on Fedora

Thank you

gdlin, 2020-04-14:
Hey everyone, I was using Fedora 30 with kernel 5.5 and I made it work after removing everything and following @alhalama instructions, and everything worked. Yesterday, I've upgraded to F31 and redid the steps since I'm having an issue. I think I already had it but maybe you are having it too, and I'd greatly appreciate some help. The keyboard entry has a delay, and the letters redraw multiple times on the screen. I have a discrete nVidia, so I already disabled and blacklisted nouveau, which was something that bringed up problems in the past, but this is still happening. Do you have any advice on this? 

Thanks.

alhalama, 2020-04-14:
@gdin, I have some delay with typing with some applications on the external monitors and I typically just leave those apps on the laptop display as that doesn't have the same issue and I haven't taken any time to look into the cause as it isn't in the apps that I am using most of the time.

gdlin, 2020-04-14:
@alhalama I have IntelliJ on my main screen which is connected via displaylink so it's a PAIN. I disabled GNOME animations and it's a bit better. I also don't have the problem on the laptop display. Thanks for your reply BTW, I see I'm not the only one.

I found this links, I'll try them later since I need to work and I can't break my system right now if something doesn't works as expected.

[Disable vsync post](https://askubuntu.com/a/884838)
[DisplayLink support](https://support.displaylink.com/knowledgebase/articles/1181623)

elguero, 2020-04-14:
@gdlin Yes, this has been an issue since Fedora 31.

See #112 and https://github.com/DisplayLink/evdi/issues/186

elguero, 2020-04-21:
The patches have been removed from master.  Rawhide builds should be working again.

closed 2020-04-21 by elguero