displaylink-rpm/displaylink-rpm #120: Fedora 31 - Single monitor stopped working.

egyas, 2020-02-27:
I previously had the drivers from this repo working.  And oh, the worked well.  I have a HP Envy x360 (Bang & Olufsen, Family: 103C_5335KV HP Envy) use a dock/port replicator (that had DisplayLink chips in it) to get additional USB, network, and display ports.  You can see from the picture below that at one point, I had all three external monitors working.

The current layout is...
Left monitor (DVI-I-3-3): Connected via DisplayPort cable to dock.  1920x1080
Top monitor (DVI-I-2-2): Connected via HDMI cable to dock.  1920x1080.
Right monitor (DVI-I-a-a): Connected via DisplayLink USB adapter to USB port on the dock.  1920x1080.

The top monitor is no longer displaying anything.  As you can see form the XrandR & ArandR pics below, both are recognized.  More over, the system seems to think that the top monitor is working, because when I move my mouse up there it disappears, and if I move it over to the left monitor, it comes out where it should be.  Apps that were set to spawn on that display previously still go there when I launch them, and I have to blindly grab for them and drag them down.

So the system sees that top monitor, and seems to think it's using it, AND I see the monitor's backlight so it''s not "dead", but there is no image displayed there.

Odder yet, if I disconnect that monitor, the left monitor (DVI-I-3-3) will not display.  I have to plug the other one back in in order to get it to come back to life.

Like I said, it WAS working fine, and I did NOT upgrade the kernel at all.  I rebooted one day and it just stopped working.

Unfortunately I can't swap cables or ports since one is DP/DVI only, and the other is HDMI only.  I reinstalled the drivers, but that had no impact.

Anyone have any ideas on how to restore full functionality?  Thanks!

Dock: MediaGear Dual Display USB-C docking station
https://www.mediagearpro.com/store/p33/DK0004.html
[image]
[image]
[image]

gdlin, 2020-04-16:
@egyas if your top monitor has 2 hdmi ports, try connecting one of those to the hdmi of the computer and check if it works. This is NOT to get the hdmi output. It happened to me (IDK why honestly) with one screen. Like it was a ground problem or something, but the behavior was similar to what you are describing. I ended up connecting and HDMI cable between two monitors. It makes NO SENSE, but it's working since!

elguero, 2020-05-14:
@egyas Were you able to get this resolved?

egyas, 2020-05-14:
I was not.  The monitor works fine on its own. But it still does not work in this configuration anymore.

elguero, 2020-05-14:
@egyas Have you been able to try the latest release out?

egyas, 2020-05-15:
@elguero I just tried it today.  Same as before.  :(

egyas, 2020-05-18:
@elguero So this was odd.  I had some issues with an unrelated software package (my company's VPN client), which forced me to reboot over the weekend.  When the laptop came back up, all of my display settings for the DisplayLink were gone, monitor's weren't recognized, etc.  I reinstalled the drivers, and this time it all worked.  So I'm back to all 4 screen working.  Ha!

egyas, 2020-05-18:
Thank you for the update!  :)  Closing issue.

closed 2020-05-18 by egyas

elguero, 2020-05-18:
@egyas Thanks for the update.  Glad that all is working now.