displaylink-rpm/displaylink-rpm #121: Cannot install on Fedora 31 with kernel 5.5.7-200

richardfearn, 2020-03-08:
Error occurs when installing `fedora-31-displaylink-1.6.4-2.x86_64.rpm`. [`/var/log/displaylink/displaylink.log`](https://github.com/displaylink-rpm/displaylink-rpm/files/4303056/displaylink.log) and 
[`/var/lib/dkms/evdi/1.6.4/build/make.log`](https://github.com/displaylink-rpm/displaylink-rpm/files/4303057/make.log) attached.

richardfearn, 2020-03-08:
It looks like another set of RPMs is needed, that includes commits like [this one](https://github.com/DisplayLink/evdi/commit/4722ad8cb55da60e3e1c000284fc0ecaffa17065#diff-ab843f41df51e462f2c71c5621b884ef).

BigZ1981, 2020-03-09:
I'm still learning the more advanced aspects of compiling stuff, but how can I install the latest commits if I clone it off the evdi repo?

alhalama, 2020-03-09:
Typically if you use `make rawhide` to build the RPM from this repository it will use the devel branch of evdi which has the changes needed for the 5.5. kernel.  Note that because the patches are automatically applied, you will want to checkout commit 316206f instead of the latest when making a rawhide build.   I provided more detailed steps to do this in[ a comment on issue 119](https://github.com/displaylink-rpm/displaylink-rpm/issues/119#issuecomment-592207142).

BigZ1981, 2020-03-09:
thanks for that.  I followed the steps & got this:

```
[root@Revan:displaylink-rpm]# make rawhide
Checking last upstream commit date...
make RELEASE="`        echo -e "1" | tr -d '[:space:]'`.rawhide.`	curl -s https://api.github.com/repos/DisplayLink/evdi/branches/devel -H "Accept: application/vnd.github.full+json" | grep date | head -1 | cut -d: -f 2- | sed s/[^0-9TZ]//g`" devel all
make[1]: Entering directory '/root/displaylink-rpm'
git clone --depth 1 -b devel https://github.com/DisplayLink/evdi.git /var/tmp/evdi-1.6.4
Cloning into '/var/tmp/evdi-1.6.4'...
remote: Enumerating objects: 45, done.
remote: Counting objects: 100% (45/45), done.
remote: Compressing objects: 100% (45/45), done.
remote: Total 45 (delta 1), reused 8 (delta 0), pack-reused 0
Unpacking objects: 100% (45/45), done.
cd /var/tmp/evdi-1.6.4 && git pull
Already up to date.
tar -z -c -f v1.6.4.tar.gz -C /var/tmp evdi-1.6.4
rpmbuild -bb --define "_topdir `pwd`" --define "_sourcedir `pwd`" --define "_rpmdir `pwd`" --define "_specdir `pwd`" --define "_srcrpmdir `pwd`" --define "_buildrootdir `mktemp -d /var/tmp/displayportXXXXXX`" --define "_builddir `mktemp -d /var/tmp/displayportXXXXXX`" --define "_release 1.rawhide.20200303T134631Z" --define "_daemon_version 5.2.14" --define "_version 1.6.4" --define "_tmppath `mktemp -d /var/tmp/displayportXXXXXX`"  displaylink.spec --target=i386
/bin/sh: rpmbuild: command not found
make[1]: *** [Makefile:132: i386/displaylink-1.6.4-1.rawhide.20200303T134631Z.i386.rpm] Error 127
make[1]: Leaving directory '/root/displaylink-rpm'
make: *** [Makefile:78: rawhide] Error 2
```

BigZ1981, 2020-03-09:
Got an aerror.  tried to start from scratch & re-clone the git.  Still errored out on me:

```[root@Revan:displaylink-rpm]# make rawhide
Checking last upstream commit date...
make RELEASE="`        echo -e "1" | tr -d '[:space:]'`.rawhide.`	curl -s https://api.github.com/repos/DisplayLink/evdi/branches/devel -H "Accept: application/vnd.github.full+json" | grep date | head -1 | cut -d: -f 2- | sed s/[^0-9TZ]//g`" devel all
make[1]: Entering directory '/root/displaylink-rpm'
cd /var/tmp/evdi-1.6.4 && git pull
Already up to date.
tar -z -c -f v1.6.4.tar.gz -C /var/tmp evdi-1.6.4
wget --post-data="fileId=1369    &accept_submit=Accept" -O DisplayLink\ USB\ Graphics\ Software\ for\ Ubuntu\ 5.2.14.zip \
	 https://www.displaylink.com/downloads/file?id=1369    
--2020-03-09 13:10:44--  https://www.displaylink.com/downloads/file?id=1369
Resolving www.displaylink.com (www.displaylink.com)... 34.252.102.94, 52.209.94.109, 52.214.95.115, ...
Connecting to www.displaylink.com (www.displaylink.com)|34.252.102.94|:443... connected.
HTTP request sent, awaiting response... 303 See Other
Location: http://assets.displaylink.com/live/downloads/software/f1369_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%205.2.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1583778045&Signature=i%2FarirBVkdjQQJw%2F4pU9yCXnqdM%3D [following]
--2020-03-09 13:10:45--  http://assets.displaylink.com/live/downloads/software/f1369_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%205.2.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1583778045&Signature=i%2FarirBVkdjQQJw%2F4pU9yCXnqdM%3D
Resolving assets.displaylink.com (assets.displaylink.com)... 52.218.100.123
Connecting to assets.displaylink.com (assets.displaylink.com)|52.218.100.123|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 12768730 (12M) [application/octet-stream]
Saving to: ‘DisplayLink USB Graphics Software for Ubuntu 5.2.14.zip’

DisplayLink USB Graphics Software 100%[==========================================================>]  12.18M   463KB/s    in 25s     

2020-03-09 13:11:11 (494 KB/s) - ‘DisplayLink USB Graphics Software for Ubuntu 5.2.14.zip’ saved [12768730/12768730]

rpmbuild -bb --define "_topdir `pwd`" --define "_sourcedir `pwd`" --define "_rpmdir `pwd`" --define "_specdir `pwd`" --define "_srcrpmdir `pwd`" --define "_buildrootdir `mktemp -d /var/tmp/displayportXXXXXX`" --define "_builddir `mktemp -d /var/tmp/displayportXXXXXX`" --define "_release 1.rawhide.20200303T134631Z" --define "_daemon_version 5.2.14" --define "_version 1.6.4" --define "_tmppath `mktemp -d /var/tmp/displayportXXXXXX`"  displaylink.spec --target=i386
/bin/sh: rpmbuild: command not found
make[1]: *** [Makefile:132: i386/displaylink-1.6.4-1.rawhide.20200303T134631Z.i386.rpm] Error 127
make[1]: Leaving directory '/root/displaylink-rpm'
make: *** [Makefile:78: rawhide] Error 2
```

BigZ1981, 2020-03-10:
Just an added note, I'm on GNOME w/Xorg.  I've been told that the driver doesn't work in the GNOME DE, even with Wayland disabled.  Do I need to switch to another DE like xcfe or cinnamon?  I would prefer not to, but I need my other monitor, so if I must, then I must.

alhalama, 2020-03-10:
I'm running Gnome with Wayland and have two external monitors so it can work.  

It appears that evdi isn't built for your kernel: 
`Mar 09 13:48:30 Revan modprobe[10147]: modprobe: FATAL: Module evdi not found in directory /lib/modules/5.5.7-200.fc31.x86_64`

If you installed before, maybe there is some cleanup needed.  You may want to try the following 

1. Uninstall the RPM. 
2. Then clear out anything left over under '/var/lib/dkms/evdi'. 
3. Reboot into the latest kernel. 
4. Then install the RPM again.

BigZ1981, 2020-03-10:
I'll try during my lunch break & post results.  Can't reboot just yet. lol

```
[root@Revan:~]# uname -r
5.5.7-200.fc31.x86_64

[root@Revan:~]# dnf list available kernel
Last metadata expiration check: 0:17:23 ago on Tue 10 Mar 2020 11:11:52 AM CDT.
Available Packages
kernel.x86_64                                                 5.5.8-200.fc31                                                  updates
```

BigZ1981, 2020-03-10:
Same issue: `Module evdi not found in directory /lib/modules/5.5.8-200.fc31.x86_64`

```
[root@Revan:~]# ls -al /lib/modules/5.5.8-200.fc31.x86_64
total 20468
drwxr-xr-x.  6 root root     4096 Mar 10 13:09 .
drwxr-xr-x.  5 root root     4096 Mar 10 13:08 ..
-rw-r--r--.  1 root root      249 Mar  5 15:42 bls.conf
lrwxrwxrwx.  1 root root       38 Mar  5 15:42 build -> /usr/src/kernels/5.5.8-200.fc31.x86_64
-rw-r--r--.  1 root root   217172 Mar  5 15:42 config
drwxr-xr-x.  5 root root     4096 Mar 10 13:09 extra
drwxr-xr-x. 13 root root     4096 Mar 10 13:08 kernel
-rw-r--r--.  1 root root  1133752 Mar 10 13:09 modules.alias
-rw-r--r--.  1 root root  1108437 Mar 10 13:09 modules.alias.bin
-rw-r--r--.  1 root root     1688 Mar  5 15:42 modules.block
-rw-r--r--.  1 root root     8503 Mar  5 15:42 modules.builtin
-rw-r--r--.  1 root root    10752 Mar 10 13:09 modules.builtin.bin
-rw-r--r--.  1 root root    62600 Mar  5 15:42 modules.builtin.modinfo
-rw-r--r--.  1 root root   417386 Mar 10 13:09 modules.dep
-rw-r--r--.  1 root root   576843 Mar 10 13:09 modules.dep.bin
-rw-r--r--.  1 root root      403 Mar 10 13:09 modules.devname
-rw-r--r--.  1 root root      165 Mar  5 15:42 modules.drm
-rw-r--r--.  1 root root       59 Mar  5 15:42 modules.modesetting
-rw-r--r--.  1 root root     2719 Mar  5 15:42 modules.networking
-rw-r--r--.  1 root root   140886 Mar  5 15:42 modules.order
-rw-r--r--.  1 root root      703 Mar 10 13:09 modules.softdep
-rw-r--r--.  1 root root   471599 Mar 10 13:09 modules.symbols
-rw-r--r--.  1 root root   577322 Mar 10 13:09 modules.symbols.bin
lrwxrwxrwx.  1 root root        5 Mar  5 15:42 source -> build
-rw-r--r--.  1 root root   269319 Mar  5 15:42 symvers.gz
-rw-------.  1 root root  5040726 Mar  5 15:42 System.map
drwxr-xr-x.  2 root root     4096 Mar  5 15:41 updates
drwxr-xr-x.  2 root root     4096 Mar 10 13:08 vdso
-rwxr-xr-x.  1 root root 10838728 Mar  5 15:42 vmlinuz
-rw-r--r--.  1 root root      166 Mar  5 15:40 .vmlinuz.hmac
```

I do, however, see edvi in dkms now:
```
[root@Revan:~]# ls -al /var/lib/dkms/evdi/1.6.4/
total 12
drwxr-xr-x. 3 root root 4096 Mar 10 14:08 .
drwxr-xr-x. 3 root root 4096 Mar 10 14:08 ..
drwxr-xr-x. 2 root root 4096 Mar 10 14:08 build
lrwxrwxrwx. 1 root root   19 Mar 10 14:08 source -> /usr/src/evdi-1.6.4
```

Is there a way to build the evdi module into the kernel using the git release for evdi?

yug0slav, 2020-03-14:
Anybody was able to solve this? 
fedora-31-displaylink-1.6.4-2.x86_64.rpm  definatly not working on fedora 31 / kernel 5.5.8-200.fc31.x86_64

$ sudo dnf install ./fedora-31-displaylink-1.6.4-2.x86_64.rpm 
Last metadata expiration check: 1:48:25 ago on Sat 14 Mar 2020 12:22:59 PM CDT.
Dependencies resolved.
==============================================================================================================================================================================================================================================
 Package                                              Architecture                                    Version                                                                     Repository                                             Size
==============================================================================================================================================================================================================================================
Installing:
 displaylink                                          x86_64                                          1.6.4-2                                                                     @commandline                                          3.6 M
Installing dependencies:
 dkms                                                 noarch                                          2.8.1-4.20200214git5ca628c.fc31                                             updates                                                76 k

Transaction Summary
==============================================================================================================================================================================================================================================
Install  2 Packages

Total size: 3.7 M
Total download size: 76 k
Installed size: 10 M
Is this ok [y/N]: y
Downloading Packages:
dkms-2.8.1-4.20200214git5ca628c.fc31.noarch.rpm                                                                                                                                                               249 kB/s |  76 kB     00:00    
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Total                                                                                                                                                                                                         133 kB/s |  76 kB     00:00     
Running transaction check
Transaction check succeeded.
Running transaction test
Transaction test succeeded.
Running transaction
  Preparing        :                                                                                                                                                                                                                      1/1 
  Installing       : dkms-2.8.1-4.20200214git5ca628c.fc31.noarch                                                                                                                                                                          1/2 
  Running scriptlet: dkms-2.8.1-4.20200214git5ca628c.fc31.noarch                                                                                                                                                                          1/2 
Created symlink /etc/systemd/system/multi-user.target.wants/dkms.service → /usr/lib/systemd/system/dkms.service.

  Installing       : displaylink-1.6.4-2.x86_64                                                                                                                                                                                           2/2 
  Running scriptlet: displaylink-1.6.4-2.x86_64                                                                                                                                                                                           2/2 
Job for displaylink.service failed because the control process exited with error code.
See "systemctl status displaylink.service" and "journalctl -xe" for details.
warning: %post(displaylink-1.6.4-2.x86_64) scriptlet failed, exit status 1

Error in POSTIN scriptlet in rpm package displaylink
warning: %triggerin(displaylink-1.6.4-2.x86_64) scriptlet failed, exit status 10

Error in <unknown> scriptlet in rpm package displaylink
  Verifying        : dkms-2.8.1-4.20200214git5ca628c.fc31.noarch                                                                                                                                                                          1/2 
  Verifying        : displaylink-1.6.4-2.x86_64                                                                                                                                                                                           2/2 

Installed:
  displaylink-1.6.4-2.x86_64                                                                                    dkms-2.8.1-4.20200214git5ca628c.fc31.noarch                                                                                   

Complete!

---
$ systemctl status displaylink.service 
● displaylink.service - DisplayLink Manager Service
   Loaded: loaded (/usr/lib/systemd/system/displaylink.service; static; vendor preset: disabled)
   Active: activating (auto-restart) (Result: exit-code) since Sat 2020-03-14 14:12:12 CDT; 1s ago
  Process: 9029 ExecStartPre=/sbin/modprobe evdi (code=exited, status=1/FAILURE)
      CPU: 7ms

---
$ sudo modprobe evdi
modprobe: FATAL: Module evdi not found in directory /lib/modules/5.5.8-200.fc31.x86_64

$ rpm -qa | grep kernel-headers
kernel-headers-5.5.8-200.fc31.x86_64

SergeyKons, 2020-03-17:
@yug0slav I solved this with this instruction
https://github.com/displaylink-rpm/displaylink-rpm/issues/119#issuecomment-592207142

beshad, 2020-04-10:
i have a similar problem on a later kernel:
`$ modprobe evdi
modprobe: FATAL: Module evdi not found in directory /lib/modules/5.5.15-200.fc31.x86_64`
`$ rpm -qa | grep kernel-headers
kernel-headers-5.5.15-200.fc31.x86_64`

and when i follow the instructions from #119  i get this error:

```
.........................
Saving to: ‘DisplayLink USB Graphics Software for Ubuntu 5.2.14.zip’
2020-04-10 14:34:58 (427 KB/s) - ‘DisplayLink USB Graphics Software for Ubuntu 5.2.14.zip’ saved [12768730/12768730]

rpmbuild -bb --define "_topdir `pwd`" --define "_sourcedir `pwd`" --define "_rpmdir `pwd`" --define "_specdir `pwd`" --define "_srcrpmdir `pwd`" --define "_buildrootdir `mktemp -d /var/tmp/displayportXXXXXX`" --define "_builddir `mktemp -d /var/tmp/displayportXXXXXX`" --define "_release 1.rawhide.20200303T134631Z" --define "_daemon_version 5.2.14" --define "_version 1.6.4" --define "_tmppath `mktemp -d /var/tmp/displayportXXXXXX`"  displaylink.spec --target=i386
/bin/sh: rpmbuild: command not found
make[1]: *** [Makefile:132: i386/displaylink-1.6.4-1.rawhide.20200303T134631Z.i386.rpm] Error 127
make[1]: Leaving directory '/home/behshad/Downloads/displaylink-rpm'
make: *** [Makefile:78: rawhide] Error 2
```

and it stops there, any suggestions?

beshad, 2020-04-10:
> i have a similar problem on a later kernel:
> `$ modprobe evdi modprobe: FATAL: Module evdi not found in directory /lib/modules/5.5.15-200.fc31.x86_64`
> `$ rpm -qa | grep kernel-headers [...]

actually problem above was due to some dependencies which i installed and successfully did the #119 , however second USB screen is still not working:

```
$ systemctl status displaylink.service
displaylink.service - DisplayLink Manager Service
   Loaded: loaded (/usr/lib/systemd/system/displaylink.service; static; vendor preset: disabled)
   Active: activating (auto-restart) (Result: exit-code) since Fri 2020-04-10 14:53:26 NZST; 1s ago
  Process: 4594 ExecStartPre=/sbin/modprobe evdi (code=exited, status=1/FAILURE)
      CPU: 7ms
```

```
$ systemctl status dlm.service
dlm.service - dlm control daemon
   Loaded: loaded (/usr/lib/systemd/system/dlm.service; enabled; vendor preset: disabled)
   Active: inactive (dead)

Apr 10 14:51:45 localhost.localdomain systemd[1]: Dependency failed for dlm control daemon.
Apr 10 14:51:45 localhost.localdomain systemd[1]: dlm.service: Job dlm.service/start failed with result 'de>
```
and finally
```
$ xrandr --listproviders
Providers: number : 1
Provider 0: id: 0x44 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 2 associated providers: 0 name:modesetting
```

elguero, 2020-04-22:
Just in case anyone missed it, `make rawhide` should be working now.

PR #131 (2020-05-08): Fixes related to evdi devel branch changes
The changes are:
  * Updates related to the new evdi driver version 1.7.0.
  * Update to the released Displaylink driver 5.3.1.
  * The minimum kernel supported in evdi 1.7.0 is now 4.15. Adjusted the spec
    accordingly.
  * Remove Travis build for Fedora 27 since the kernel is no longer supported.
  * Add Fedora 32 build to Travis.
  * Fix support for DL-6xxx devices. The firmware image was not being copied from
    the DisplayLink driver package.
  * Adjust how we use dkms inside the rpm to follow recommended way outlined in the
    documentation. [... cut]
commit 4a9f425b (2020-05-08): Fixes related to evdi devel branch changes

The changes are:
  * Updates related to the new evdi driver version 1.7.0.
  * Update to the released Displaylink driver 5.3.1.
  * The minimum kernel supported in evdi 1.7.0 is now 4.15. Adjusted the spec
    accordingly.
  * Remove Travis build for Fedora 27 since the kernel is no longer supported.
  * Fix support for DL-6xxx devices. The firmware image was not being copied from
    the DisplayLink driver package.
  * Adjust how we use dkms inside the rpm to follow recommended way outlined in the
    documentation.

Resolves #112, resolves #121, re [... cut]
closed 2020-05-08 by elguero