displaylink-rpm/displaylink-rpm #122: CentOS8 rpm

sbluhm, 2020-03-10:
Hello,

I just noticed that you include the CentOS 7 rpm which I usually installed manually.

Can you please also include a CentOS 8 rpm? It is actually a lot easier than C7 as all requirements are fulfilled on BaseOS already.

When I get round, I might provide a pull request.

elguero, 2020-03-10:
For reference: https://github.com/DisplayLink/evdi/issues/193

sbluhm, 2020-03-12:
Update: Currently, evdi only works on RHEL 8.0. The RHEL 8.1 kernel has backports from newer kernels included, that break evdi. So I potentially see these package options:
1. limit the kernel to "Required: kernel < 4.18.0-147.0.0".
2. Patch to fix the RHEL 8.1 issues (first reaction of evdi was to not fix it upstream).
3. Work with a different version of evdi (older or future)

shoguevara, 2020-07-23:
Hi! I'm currently using the rpm with CentOS Linux release 8.2.2004 (Core) and I can confirm it's working fine.
Although I had to switch to 5.7.9-1.el8.elrepo.x86_64 kernel, so I could install build dependencies.

BTW Thanks a lot! Saved me a lot of time!

sbluhm, 2020-07-25:
Update2: I created a patch to work (compile) with RHEL8.1+ which I still need to test on an actual DisplayLink device. I am trying to get into EVDI master but I am not sure if they accept it.

In the meantime, I would say this patch can be added here and kernel requirement enforced (not sure if RHEL 8.0 should really be included though but wouldn't be too much effort). 

@elguero , should I go ahead and put some effort into RHEL8 support here or is this currently not in scope?

elguero, 2020-07-26:
Excellent!  Thanks for your help and time on this.

I would say the best place to start would be to get your patches merged into the evdi project.  This repository here is just managing the creation of the RPM to install evdi and the displaylink binary.

After getting those patches merged upstream, then we can work on the changes for the RPM creation process in this project to support RHEL8/CentOS8.  Then you could submit a PR for those changes.

sbluhm, 2020-07-26:
Actually, I have just now done the changes but still not tested with actual hardware (will do so next week). I don't expect the evdi merge soon.

My approach:
1. Adapted Makefile for an additional EL8.1 target.
2. Added "el81" option to spec file which patches the sources and removed kernel-ml for RHEL8.
Tested and installed the el8.rpm file on RHEL8.2.

Once evdi contains my changes, I would revert changes to the Makefile and spec file.
**If you are fine with that, I would create a PR after my final test.**

Just for completeness, can anybody please check my changes from [https://github.com/sbluhm/displaylink-rpm](https://github.com/sbluhm/displaylink-rpm) ? Testing it on more than one system is always helpful.

sbluhm, 2020-07-31:
Test succeded. Find the PR: #146

closed 2021-12-29 by elguero