displaylink-rpm/displaylink-rpm #124: Display shutdown on notebook lid close

SergeyKons, 2020-03-14:
HI i've using Fedora 31 with kernel v.5.5
When i close laptop lid my screens connected to DisplayLInk hub (Dell D3100) disabling. I see image but can't interact with it

Notebook: HP Envy x360 13-ar0011ur

elguero, 2020-04-16:
@SergeyKons Probably similar to this https://github.com/DisplayLink/evdi/issues/192

elguero, 2020-05-14:
@SergeyKons I have not been able reproduce.  I do not have that hub though and I am on F32.  I am on the latest release and was able to successfully close my laptop and keep working with the monitors.

Do you still have this issue?

elguero, 2020-06-17:
Maybe related?

https://github.com/DisplayLink/evdi/issues/181#issuecomment-643955576

SergeyKons, 2020-07-28:
Hi, Currently I switched from fedora to win10 so this isn't relevant for me. I think we can close it. 

I think that issues that you attached to were relevant and now the problem is fixed.

closed 2020-07-28 by SergeyKons