displaylink-rpm/displaylink-rpm #125: Copr for Rebuilds

rapgro, 2020-03-25:
Can you use Copr for rebuilds? It supports at least Fedora and CentOS (EPEL) as well as Mageia.
See also #76 .

https://docs.pagure.org/copr.copr/user_documentation.html#what-i-can-build-in-copr
https://developer.fedoraproject.org/deployment/copr/copr-cli.html

rapgro, 2020-03-25:
OpenBuildService is an alternative.

https://openbuildservice.org/help/manuals/obs-user-guide/par.working.html

pvalena, 2021-04-19:
I've investigated COPR (would've done it myself) few weeks ago, but I found out you need to have a opensource license for that (not shipping any proprietary code)... basically same as in Fedora.

https://docs.pagure.org/copr.copr/user_documentation.html#what-i-can-build-in-copr

elguero, 2021-04-19:
Thanks for investigating and confirming.  I had a feeling that would be the case.

rapgro, 2021-04-20:
Why not evdi at least and a script to additionally download the proprietary part? That's allowed IMHO. I am going to find some time to create and suggest a spec file, maybe also for package review.

pvalena, 2021-04-20:
> Why not evdi at least and a script to additionally download the proprietary part? That's allowed IMHO. I am going to find some time to work on the spec file, maybe also for package review.

Yes, you're right, it could be done in a similar way (not sure about "during dnf install"), but don't you have to compile it also? It would require user to have preinstalled GCC/llvm toolchain, and all the devel libs... I'm not familiar with any restrictions about this in Fedora repos, but in any case, RPMFusion may be the right way to go.

> Besides there are some community packages built in OpenBuildService for openSUSE.

Maybe you could copy / compare the spec files, in that case?

rapgro, 2021-04-22:
@pvalena It's about DisplayLinkManager as the central component and officially shipped only as binary blob for Ubuntu, that's what a script should downlaod and properly install after dnf install. The kernel is tainted with evdi module, built from dkms having always trouble with major changes to kernel API as seen recently, see discussion in other issues here. But I can agree, rpmfusion seems the better repository to use therefore, besides dkms should be replaced with akmod.

elguero, 2021-04-26:
Adding reference to PR: https://github.com/displaylink-rpm/displaylink-rpm/pull/97