displaylink-rpm/displaylink-rpm #127: Cannot build on Fedora 31 (kernel 5.5.17)

cdelledonne, 2020-04-21:
I could not install the pre-built RPM, so I decided to try and build it from scratch using the Makefile (by running `make rpm`). But I'm facing two problems.

### Environment
* Fedora 31
* kernel 5.5.17-200.fc31.x86_64

### Error 1: EVDI library name

First, the `.spec` file specifies a name for the `.so` EVDI library file which is not compatible with what is produced by https://github.com/DisplayLink/evdi/blob/v1.6.4/library/Makefile. Specifically, the upstream Makefile produces a library named `libevdi.so`
```make
libevdi.so: evdi_lib.o
	$(CC) $^ -o $@ -lc -lgcc -shared -Wl,-soname,libevdi.so.0
```
but the `.spec` file seems to look for `libevdi.so.%{version}`, which is the name produced by the Makefile in the upstream `devel` branch. Can we find a way for the `.spec` to be compatible with both EVDI v1.6.4 and its development version?

### Error 2: insecure RPATH detected by check-rpaths

With a workaround for error 1, RPM returns an error when running `check-rpaths`:
```
ERROR   0004: file '/usr/libexec/displaylink/DisplayLinkManager' contains an insecure rpath '.' in [.]
error: Bad exit status from /var/tmp/displayportr4n2FQ/rpm-tmp.7rUgOT (%install)
```

I couldn't figure out what is triggering this. Any clue?

elguero, 2020-04-22:
@cdelledonne We are waiting for the evdi team to release a new version with support for kernels 5.4 and 5.5.  The evdi devel branch has the compatability fixes in it.

For now, we are recommending that if you need to run on these kernels, clone the repository here and run `make rawhide` to build a compatible version for your kernel.  Of course, all the disclaimers apply about running an unreleased / development version of the code.

I hope this helps.

elguero, 2020-04-22:
Duplicate issue: https://github.com/displaylink-rpm/displaylink-rpm/issues/121

For reference: https://github.com/displaylink-rpm/displaylink-rpm/issues/119#issuecomment-607264996

closed 2020-04-22 by elguero