displaylink-rpm/displaylink-rpm #128: Latest EVDI changes break install

ItsDom, 2020-05-02:
I wasn't able to install on Fedora 32, even with `make rawhide` I get the following error:

`cannot stat 'evdi-1.6.4/library/libevdi.so.1.6.4': No such file or directory`

Manually modifying the makefile to checkout commit 01e246f from EVDI repo as per 
https://github.com/DisplayLink/evdi/issues/199#issuecomment-621812547 seems to be a work around.

I think this is also why the tests have failed on the last few commits in this repo too.

elguero, 2020-05-03:
@ItsDom Yes, we know.  The evdi devel branch has been a moving target.  Just when we think we have something for the community, it changes.  We are waiting for a new release of EVDI to make an official release here.

wernermaes, 2020-05-05:
So upgrading to Fedora 32 break evdi? Good to know

closed 2020-05-08 by elguero

bowerth, 2020-05-08:
Hello and many thanks for the continued support. I am receiving the following error:

```
wget -O v1.7.0.tar.gz \
	https://github.com/DisplayLink/evdi/archive/v1.7.0.tar.gz
--2020-05-08 21:48:13--  https://github.com/DisplayLink/evdi/archive/v1.7.0.tar.gz
Resolving github.com (github.com)... 140.82.118.4
Connecting to github.com (github.com)|140.82.118.4|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/DisplayLink/evdi/tar.gz/v1.7.0 [following]
--2020-05-08 21:48:14--  https://codeload.github.com/DisplayLink/evdi/tar.gz/v1.7.0
Resolving codeload.github.com (codeload.github.com)... 140.82.114.9
Connecting to codeload.github.com (codeload.github.com)|140.82.114.9|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2020-05-08 21:48:14 ERROR 404: Not Found.

make: *** [Makefile:115: v1.7.0.tar.gz] Error 8
```

elguero, 2020-05-08:
@bowerth You need to do `make rawhide` for now.  The evdi library 1.7.0 has not been released yet.

bowerth, 2020-05-08:
OK, thanks - how can I deactivate QA_CHECK_RPATHS ?

elguero, 2020-05-09:
@bowerth I am not sure how you are building your RPM.  Those variables are all set through the Makefile and passed in to the displaylink.rpm.  Something doesn't seem right.

bowerth, 2020-05-09:
I agree. I am running 'make rawhide' as suggested and the resulting src.rpm contains e.g. a spec file where the respective variables have not been expanded (_version etc.)

PR #131 (2020-05-08): Fixes related to evdi devel branch changes
The changes are:
  * Updates related to the new evdi driver version 1.7.0.
  * Update to the released Displaylink driver 5.3.1.
  * The minimum kernel supported in evdi 1.7.0 is now 4.15. Adjusted the spec
    accordingly.
  * Remove Travis build for Fedora 27 since the kernel is no longer supported.
  * Add Fedora 32 build to Travis.
  * Fix support for DL-6xxx devices. The firmware image was not being copied from
    the DisplayLink driver package.
  * Adjust how we use dkms inside the rpm to follow recommended way outlined in the
    documentation. [... cut]
commit 4a9f425b (2020-05-08): Fixes related to evdi devel branch changes

The changes are:
  * Updates related to the new evdi driver version 1.7.0.
  * Update to the released Displaylink driver 5.3.1.
  * The minimum kernel supported in evdi 1.7.0 is now 4.15. Adjusted the spec
    accordingly.
  * Remove Travis build for Fedora 27 since the kernel is no longer supported.
  * Fix support for DL-6xxx devices. The firmware image was not being copied from
    the DisplayLink driver package.
  * Adjust how we use dkms inside the rpm to follow recommended way outlined in the
    documentation.

Resolves #112, resolves #121, re [... cut]