displaylink-rpm/displaylink-rpm #129: Freeze after screenunlock for some seconds

msandres13, 2020-05-03:
Using display link dock i-tec CADUAL4KDOCKPD with external monitor attached to display port. Logitec wireless dongle for Keyboard and Mouse on front USB3.0 Type A and USB to the Monitor (iiyama ProLite XUB2792UHSU-B1)

Fedora 31 5.6.8-200.fc31.x86_64
displaylink-rpm git version
commit c3e711faf2f16f8969789b20f82a9787cd4b10d7 (HEAD -> master, origin/master, origin/HEAD)
Author: Michael L. Young <elgueromexicano@gmail.com>
Date:   Tue Apr 21 15:48:49 2020 -0400

    Fix wrong date in .spec

How to reproduce:
1. let system idle for about an hour
2. after screen lock happened may be 30 minutes idle
3. login again
System shows normal screen after unlock, mouse moves only sporadic, about 10 second freezes, keyboard the same. This procedure works many times but not always.

dmesg log:
[11842.198878] usb 1-3.4.4: USB disconnect, device number 20
[11842.222785] usb 4-1.4.4: USB disconnect, device number 15
[11842.817384] usb 4-1.4.4: new SuperSpeed Gen 1 USB device number 16 using xhci_hcd
[11842.848783] usb 4-1.4.4: New USB device found, idVendor=0bda, idProduct=0412, bcdDevice= 1.43
[11842.848788] usb 4-1.4.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[11842.848791] usb 4-1.4.4: Product: 4-Port USB 3.0 Hub
[11842.848794] usb 4-1.4.4: Manufacturer: Generic
[11842.860363] hub 4-1.4.4:1.0: USB hub found
[11842.863116] hub 4-1.4.4:1.0: 2 ports detected
[11843.064398] usb 1-3.4.4: new high-speed USB device number 21 using xhci_hcd
[11843.180204] usb 1-3.4.4: New USB device found, idVendor=0bda, idProduct=5412, bcdDevice= 1.43
[11843.180210] usb 1-3.4.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[11843.180213] usb 1-3.4.4: Product: 4-Port USB 2.0 Hub
[11843.180216] usb 1-3.4.4: Manufacturer: Generic
[11843.182238] hub 1-3.4.4:1.0: USB hub found
[11843.183371] hub 1-3.4.4:1.0: 2 ports detected
[11846.203558] logitech-hidpp-device 0003:046D:101B.000C: multiplier = 8
[13818.780137] evdi: [D] evdi_painter_dpms_notify:559 (dev=1) Notifying dpms mode: 3
[14868.836298] evdi: [D] evdi_painter_dpms_notify:559 (dev=1) Notifying dpms mode: 0
[14870.484759] usb 4-1.4.4: USB disconnect, device number 16
[14870.692833] usb 1-3.4.4: USB disconnect, device number 21
[14875.976620] usb 1-3.4: reset high-speed USB device number 15 using xhci_hcd
[14876.347531] usb 4-1.4: Disable of device-initiated U1 failed.
[14876.354983] usb 4-1.4: Disable of device-initiated U2 failed.
[14876.381334] usb 1-3.4.4: new high-speed USB device number 22 using xhci_hcd
[14876.497224] usb 1-3.4.4: New USB device found, idVendor=0bda, idProduct=5412, bcdDevice= 1.43
[14876.497230] usb 1-3.4.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[14876.497234] usb 1-3.4.4: Product: 4-Port USB 2.0 Hub
[14876.497237] usb 1-3.4.4: Manufacturer: Generic
[14876.499526] hub 1-3.4.4:1.0: USB hub found
[14876.500617] hub 1-3.4.4:1.0: 2 ports detected
[14881.668565] usb 1-3.4: reset high-speed USB device number 15 using xhci_hcd
[14881.973349] xhci_hcd 0000:3a:00.0: Timeout while waiting for setup device command
[14882.069568] usb 1-3.4.4: reset high-speed USB device number 22 using xhci_hcd
[14882.187308] usb 4-1.4: Device not responding to setup address.
[14882.396328] usb 4-1.4: device not accepting address 12, error -71
[14883.493539] logitech-hidpp-device 0003:046D:101B.000C: multiplier = 8
[14887.556354] usb 1-3.4: reset high-speed USB device number 15 using xhci_hcd
[14887.957499] usb 1-3.4.4: reset high-speed USB device number 22 using xhci_hcd
[14888.116260] xhci_hcd 0000:3a:00.0: Timeout while waiting for setup device command
[14888.331153] usb 4-1.4: Device not responding to setup address.
[14888.540249] usb 4-1.4: device not accepting address 12, error -71
[14893.700455] usb 1-3.4: reset high-speed USB device number 15 using xhci_hcd
[14894.101456] usb 1-3.4.4: reset high-speed USB device number 22 using xhci_hcd
[14894.260311] xhci_hcd 0000:3a:00.0: Timeout while waiting for setup device command
[14894.474957] usb 4-1.4: Device not responding to setup address.
[14894.684192] usb 4-1.4: device not accepting address 12, error -71
[14899.844407] usb 1-3.4: reset high-speed USB device number 15 using xhci_hcd
[14900.245396] usb 1-3.4.4: reset high-speed USB device number 22 using xhci_hcd
[14900.404439] xhci_hcd 0000:3a:00.0: Timeout while waiting for setup device command
[14900.619257] usb 4-1.4: Device not responding to setup address.
[14900.828127] usb 4-1.4: device not accepting address 12, error -71
[14900.829142] usb 4-1.4: USB disconnect, device number 12
[14905.988361] usb 1-3.4: reset high-speed USB device number 15 using xhci_hcd
[14906.389352] usb 1-3.4.4: reset high-speed USB device number 22 using xhci_hcd
[14906.548250] xhci_hcd 0000:3a:00.0: Timeout while waiting for setup device command
[14906.763103] usb 4-1.4: Device not responding to setup address.
[14906.972070] usb 4-1.4: device not accepting address 17, error -71
[14912.132297] usb 1-3.4: reset high-speed USB device number 15 using xhci_hcd
[14912.533287] usb 1-3.4.4: reset high-speed USB device number 22 using xhci_hcd
[14912.692168] xhci_hcd 0000:3a:00.0: Timeout while waiting for setup device command
[14912.906863] usb 4-1.4: Device not responding to setup address.
[14913.116024] usb 4-1.4: device not accepting address 18, error -71
[14913.116470] usb 4-1-port4: attempt power cycle
[14918.532240] usb 1-3.4: reset high-speed USB device number 15 using xhci_hcd
[14918.836089] xhci_hcd 0000:3a:00.0: Timeout while waiting for setup device command
[14918.934087] usb 1-3.4.4: reset high-speed USB device number 22 using xhci_hcd
[14919.050648] usb 4-1.4: Device not responding to setup address.
[14919.251938] usb 4-1.4: device not accepting address 19, error -71
[14924.420200] usb 1-3.4: reset high-speed USB device number 15 using xhci_hcd
[14924.821189] usb 1-3.4.4: reset high-speed USB device number 22 using xhci_hcd
[14924.980265] xhci_hcd 0000:3a:00.0: Timeout while waiting for setup device command
[14925.194898] usb 4-1.4: Device not responding to setup address.
[14925.403912] usb 4-1.4: device not accepting address 20, error -71
[14925.404345] usb 4-1-port4: unable to enumerate USB device

lsusb:
Bus 004 Device 011: ID 2109:0210 VIA Labs, Inc. 
Bus 004 Device 010: ID 17e9:6000 DisplayLink USB3.1 Hub
Bus 004 Device 009: ID 05e3:0620 Genesys Logic, Inc. USB3.1 Hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 04f2:b685 Chicony Electronics Co., Ltd Chicony USB2.0 Camera
Bus 001 Device 003: ID 06cb:00a8 Synaptics, Inc. 
Bus 001 Device 022: ID 0bda:5412 Realtek Semiconductor Corp. 
Bus 001 Device 015: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 023: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 017: ID 065f:f60f Good Way Technology Co., Ltd & GWC technology Inc. USB2.1 Hub
Bus 001 Device 014: ID 2109:2210 VIA Labs, Inc. 
Bus 001 Device 013: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 006: ID 8087:0025 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

It looks like the unresponsive USB hub is inside the laptop. Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub.

I did report the problem here since this happens since I use the display link driver. Without the dock all devices directly attached to the laptop the problem does not occur.

Since I can reporduce it I can supply more data. Just let me know what is needed.

Thanks for your help.

Michael Andres

msandres13, 2020-05-03:
I forgot to add, I used make rawhide to build.

elguero, 2020-05-08:
@msandres13 Can you pull in the latest changes, `make clean-all` and then `make rawhide`?  Hopefully this will help resolve this issue.

msandres13, 2020-05-08:
Thanks for your help.
I pulled the change and after some trouble it works in general. I will check the next few days if problem is gone and report.
First I had problems since I did just a git pull in the directory where I built the last version, the make clean, even git clean -fd.
Build went smooth, after rpm install external monitor went black/out. after reboot the same.
Did a clean checkout in a new directory make rawhide, installed, reboot. Monitor worked again.
So it looks like make clean does not clean up everything.

msandres13, 2020-05-13:
Today after a resume from a suspend I had a hang again, this time the display was ok but the transceiver for keyboard and mouse on the front port did not work.
Attanging dmesg, event happened starting with [134848.809643] multiple unplug/plug did not recover the problem, had to reboot.

[demsg-usb-fail.txt](https://github.com/displaylink-rpm/displaylink-rpm/files/4623737/demsg-usb-fail.txt)

Any body an idea what is going on?

Thanks for your help.

Regards
Michael

elguero, 2020-05-15:
@msandres13 Just throwing this out there. Maybe it is a similar situation as with the Dell D600?  I see the same messages in the dmesg that you attached.

https://github.com/displaylink-rpm/displaylink-rpm#dell-d6000

elguero, 2020-05-15:
Did a little bit more digging. I am not sure that my prior suggestion would solve this since that seems to do with disconnects while using the machine.

For your issue, there is a lot of different ideas floating around it seems.  One of the things that was mentioned was making sure your BIOS is up to date and firmware on the dock is up to date.

I am thinking that since your displays continued to work, this is not a displaylink driver issue.  In looking at the dmesg, it seems to be a USB driver / system issue.  Maybe you should reach out to the manufacture of the hub you are using.

Ville1ero, 2020-07-23:
I was able to make DisplayLink work on Fedora 32 KDE Plasma.

You can see some settings and screenshots on this link:
https://bit.ly/3fXxdb1

closed 2020-10-26 by elguero