displaylink-rpm/displaylink-rpm #130: displaylink service fails to start on fc31 kernel 5.6.8-200

Gal-Zaidman, 2020-05-07:
Hi I tried installing the latest fedora rpm v5.2.14-2, the installation was successful
but the service failed to start:

```
SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=displaylink comm=">
systemd[1]: Failed to start DisplayLink Manager Service.
systemd[1]: displaylink.service: Failed with result 'exit-code'.
systemd[1]: displaylink.service: Control process exited, code=exited, status=1/FAILURE
modprobe[13482]: modprobe: FATAL: Module evdi not found in directory /lib/modules/5.6.8-200.fc31.x86_64
```
My kernel is:

```
uname -r
5.6.8-200.fc31.x86_64
```

I'm tried removing /var/lib/dkms/evdi/ and build it again
dkms add -m evdi -v 1.6.4
dkms build -m evdi -v 1.6.4

and the build failed with:
```
scripts/sorttable.c:34:10: fatal error: tools/be_byteshift.h: No such file or directory
   34 | #include <tools/be_byteshift.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.host:107: scripts/sorttable] Error 1
make[2]: *** Waiting for unfinished jobs....
make -C /usr/src/kernels/5.6.8-200.fc31.x86_64/tools/build CFLAGS= LDFLAGS= /usr/src/kernels/5.6.8-200.fc31.x86_64/tools/objtool/fixdep
scripts/selinux/genheaders/genheaders.c:18:10: fatal error: classmap.h: No such file or directory
   18 | #include "classmap.h"
      |          ^~~~~~~~~~~~
compilation terminated.
make[4]: *** [scripts/Makefile.host:107: scripts/selinux/genheaders/genheaders] Error 1
make[3]: *** [scripts/Makefile.build:505: scripts/selinux/genheaders] Error 2
make[3]: *** Waiting for unfinished jobs....
/usr/src/kernels/5.6.8-200.fc31.x86_64/tools/build/Makefile.build:37: /usr/src/kernels/5.6.8-200.fc31.x86_64/tools/build/Build.include: No such file or directory
make[5]: *** No rule to make target '/usr/src/kernels/5.6.8-200.fc31.x86_64/tools/build/Build.include'.  Stop.
make[4]: *** [Makefile:43: /usr/src/kernels/5.6.8-200.fc31.x86_64/tools/objtool/fixdep-in.o] Error 2
make[3]: *** [/usr/src/kernels/5.6.8-200.fc31.x86_64/tools/build/Makefile.include:5: fixdep] Error 2
make -f ./scripts/Makefile.headersinst obj=arch/x86/include/uapi
make[2]: *** [Makefile:68: objtool] Error 2
make[1]: *** [Makefile:1786: tools/objtool] Error 2
scripts/selinux/mdp/mdp.c:36:10: fatal error: classmap.h: No such file or directory
   36 | #include "classmap.h"
      |          ^~~~~~~~~~~~
compilation terminated.
```

I saw the following bug on kernel-headers:
https://bugzilla.redhat.com/show_bug.cgi?id=1821946

Is there a solution or a workaround for installing the fc31 rpm ?

elguero, 2020-05-08:
@Gal-Zaidman Support for that kernel is not in the last release.  We are waiting for evdi to release a new version of their library in order to prepare a new release here.

In the meantime, you can try cloning this repository and doing a `make rawhide` build which will build an rpm based on the latest changes from the evdi devel branch and the latest released DisplayLink drivers.

Gal-Zaidman, 2020-05-08:
> @Gal-Zaidman Support for that kernel is not in the last release. We are waiting for evdi to release a new version of their library in order to prepare a new release here.
> 

Make sense, is there a way to track which versions are supported (or at least working) ? I want to downgrade my kernel version to install the drivers, but I prefer to upgrade to a version I know works.

> In the meantime, you can try cloning this repository and doing a `make rawhide` build which will build an rpm based on the latest changes from the evdi devel branch and the latest released DisplayLink drivers.

I will hit the same issue, since evdi can't be built with my kernel

elguero, 2020-05-08:
> Make sense, is there a way to track which versions are supported (or at least working) ? I want to downgrade my kernel version to install the drivers, but I prefer to upgrade to a version I know works.

I think everything was working up until kernel 5.4 came out.  That is when things started to be broken.  We created an interim release, v5.2.14-3-rc1, which fixed 5.4.  Then 5.5 came out we decided to just wait until things settled down before making an official release.
 
> > In the meantime, you can try cloning this repository and doing a `make rawhide` build which will build an rpm based on the latest changes from the evdi devel branch and the latest released DisplayLink drivers.
> 
> I will hit the same issue, since evdi can't be built with my kernel

You shouldn't have a problem.  Running `make rawhide` downloads the latest evdi devel branch which has all the fixes and supports the latest kernels.  We are waiting for them to tag a release so that we have a target to build the next release against.

I am running Fedora 32 right now with it for little more than a week.  It has been running solid for me.

Gal-Zaidman, 2020-05-09:
@elguero thanks a lot, I used make rawhide and installed the driver.
I'm hitting a 2 different bugs now:
- if my laptop lid is closed there is a very big lag on the screens, when my lid is open then everything is ok.
- I see that DisplayLinkManager consumes a lot of CPU.

elguero, 2020-05-09:
@Gal-Zaidman

I just tried your setup of closing my laptop screen. I am not experiencing any lag on my 2 DisplayLink connected screens with the laptop screen closed.  I did experience lag with my laptop open on Fedora 31 with the older evdi 1.6.4 library and DisplayLink 5.2.14.  I am now on Fedora 32 which has a newer version of Gnome the latest evdi and DisplayLink drivers.  I am not sure if that is making a difference.

In regards to CPU usage, when I am moving the mouse around I see my CPU usage go up to around 48%.  It fluctuates.  When I am not moving the mouse around, the CPU usage is around 3% - 6%.  There might still be some optimizations needed in the evdi library but that is something the evdi folks would need to look into.

I have to say that this latest evdi library and DisplayLink driver is working fairly smooth with F32.

(cross referencing with https://github.com/DisplayLink/evdi/issues/181)

Gal-Zaidman, 2020-05-10:
> @Gal-Zaidman
> 
> I just tried your setup of closing my laptop screen. I am not experiencing any lag on my 2 DisplayLink connected screens with the laptop screen closed. I did experience lag with my laptop open on Fedora 31 with the older evdi 1.6.4 library and DisplayLink 5.2.14. I am now on Fedora 32 which has a newer version of Gnome the latest evdi and DisplayLink drivers. I am not sure if that is making a difference.
> 
> In regards to CPU usage, when I am moving the mouse around I see my CPU usage go up to around 48+%. It fluctuates. When I am not moving the mouse around, the CPU usage is around 3% - 6%. There might still be some optimizations needed in the evdi library but that is something the evdi folks would need to look into.
> 
> I have to say that this latest evdi library and DisplayLink driver is working fairly smooth with F32.
> 
> (cross referencing with [DisplayLink/evdi#181](https://github.com/DisplayLink/evdi/issues/181))

OK I will try to play around with it and see what is the package combination that solves the lag, thanks a lot for the fast reply

elguero, 2020-05-14:
@Gal-Zaidman Since the main issue has been resolved, is it okay to close?

Gal-Zaidman, 2020-05-15:
Well, it's up to you, the issue still exists, there is a workaround of building with make rawhide.
I'm good either way.

elguero, 2020-05-15:
@Gal-Zaidman There was a release this week and it shouldn't be necessary to use make rawhide.

https://github.com/displaylink-rpm/displaylink-rpm/releases/tag/v5.3.1

closed 2020-05-15 by elguero