displaylink-rpm/displaylink-rpm #135: AMD Processor config?

mattdioso, 2020-05-15:
I have a ThinkPad T495 that has an AMD Ryzen Pro Processor on it that I recently updated to 32 and installed the displaylink RPM

It's able to display on both screens but I"m experiencing a pretty bad lag on both of them. I do have Wayland disabled in `/etc/gdm/custom.conf`

Is there a specific config for different types of processors?

I noticed in 20-display.conf it's set to intel.

elguero, 2020-05-15:
@mattdioso I think that this is for an Intel graphics driver, nothing to do with the processor.

Since you have a Radeon GPU, perhaps it needs to be changed.  I am not sure since I do not have that hardware to play with.

Someone over on the EVDI project mentioned something about under Wayland, things work but under X11 they are experiencing lag.

https://github.com/DisplayLink/evdi/issues/200#issuecomment-627219274

elguero, 2020-06-17:
Maybe related?

https://github.com/DisplayLink/evdi/issues/181#issuecomment-643955576

closed 2020-10-26 by elguero