displaylink-rpm/displaylink-rpm #136: Login session crashes when window is moved to DisplayLink device

marlinw-um, 2020-05-26:
On Fedora 32 with kernel 5.6.7-300.fc32.x86_64 and the MATE Desktop, I find that my entire login session crashes when I move a browser window to a DisplayLink device. I have not yet determined if this is a general problem with any application window or any desktop environment.

Thank you.

-- Marlin

elguero, 2020-05-27:
Please see discussion on EVDI project: https://github.com/DisplayLink/evdi/issues/197

In comment https://github.com/DisplayLink/evdi/issues/197#issuecomment-633927952, the user has opened a BZ report describing a crash when moving windows to the external display.

I would follow those threads to see if there are any fixes that come about.

muppet3000, 2020-05-30:
+1 from me

For the benefit of others coming across this. I'm also seeing the same issue. Will follow the mentioned thread to see if a fix is presented and hopefully subsequently included in a release here.
I'm running FC32 with 2 external monitors, as well as my laptop screen.

To work-around this issue I have reverted back to the RC candidate build (https://github.com/displaylink-rpm/displaylink-rpm/releases/download/v5.2.14-3-rc1/fedora-31-displaylink-1.6.4-2.x86_64.rpm) and the older (FC31) kernel (5.4.19-200.fc31.x86_64) which is the last one I had that worked with that RPM and things have been "fixed".

I've used "versionlock" (shell code below) to ensure that any future installs of the kernel will not drop  this version of the kernel from my "most recent 5".

```
sudo dnf install python3-dnf-plugins-extras-versionlock

sudo dnf versionlock add kernel-core-5.4.19-200.fc31.x86_64 kernel-devel-5.4.19-200.fc31.x86_64 kernel-modules-5.4.19-200.fc31.x86_64
```
Once there is a fix that works and you want to remove the "versionlock" you can use the following:
```
sudo dnf versionlock delete kernel-core-5.4.19-200.fc31.x86_64 kernel-devel-5.4.19-200.fc31.x86_64 kernel-modules-5.4.19-200.fc31.x86_64
```

Hopefully this helps someone if they stumble across this.

grant-traynor, 2020-06-17:
I've been using spectrwm for a few days now with no issues. I was having issues with lightweight window managers still crashing, but gone-shell's been fine. Looks as though that might be resolved now too. Performance seems to be back on par with 5.2.3. Upgraded to the linux 5.7 series and things are looking pretty solid:

```
linux 5.7.2.arch1-1
evdi-git 1.7.0.r71.gdc595db-1
displaylink 5.3.1.34-2
xorg-server 1.20.8-2
spectrwm 3.3.0-1
```

Ville1ero, 2020-07-23:
I was able to make DisplayLink work on Fedora 32 KDE Plasma.

You can see some settings and screenshots on this link:
https://bit.ly/3fXxdb1

closed 2020-10-26 by elguero