displaylink-rpm/displaylink-rpm #138: 'Failed to start DisplayLink Manager Service'

marzukia, 2020-06-14:
Hi all, I installed displaylink on a fresh F32 install kernel 5.6.6-300.fc32.x86_64 and I'm getting issues around evdi.

```
Jun 15 10:02:34 marzuki-lenovo modprobe[2865]: modprobe: FATAL: Module evdi not found in directory /lib/modules/5.6.6-300.fc32.x86_64
Jun 15 10:02:34 marzuki-lenovo systemd[1]: Failed to start DisplayLink Manager Service.
```

This occurs when `systemctl start displaylink` is run. 

I was looking at previous discussion regarding this issue occurring on previous Fedora builds but I'm seeing people say "don't do this" around those methods.

elguero, 2020-06-15:
@marzukia Can you provide the log file "/var/log/displaylink/displaylink.log"?

marzukia, 2020-06-16:
Hi, it's as follows

```
Creating symlink /var/lib/dkms/evdi/1.7.0/source ->
                 /usr/src/evdi-1.7.0

DKMS: add completed.
Error! echo
Your kernel headers for kernel 5.6.6-300.fc32.x86_64 cannot be found at
/lib/modules/5.6.6-300.fc32.x86_64/build or /lib/modules/5.6.6-300.fc32.x86_64/source.
You can use the --kernelsourcedir option to tell DKMS where it's located.
Error! echo
Your kernel headers for kernel 5.6.6-300.fc32.x86_64 cannot be found at
/lib/modules/5.6.6-300.fc32.x86_64/build or /lib/modules/5.6.6-300.fc32.x86_64/source.
You can use the --kernelsourcedir option to tell DKMS where it's located.

------------------------------
Deleting module version: 1.7.0
completely from the DKMS tree.
------------------------------
Done.
```

elguero, 2020-06-17:
@marzukia Okay.  The kernel headers are missing.

Can you try doing "sudo dnf install kernel-headers" and then re-install the displaylink rpm?

SimpleBinary, 2020-06-21:
@elguero I'm having this problem too, I tried installing kernel-headers and reinstalling, but it didn't seem to work. If I run `systemctl status displaylink.service` it returns:

```
● displaylink.service - DisplayLink Manager Service
     Loaded: loaded (/usr/lib/systemd/system/displaylink.service; static; vendor preset: enabled)
     Active: inactive (dead) (Result: exit-code) since Sun 2020-06-21 15:00:55 AEST; 5min ago
        CPU: 7ms
```

And if I run `modprobe evdi`:

```
modprobe: ERROR: could not insert 'evdi': Exec format error
```

elguero, 2020-06-22:
@SimpleBinary Can you post your `/var/log/displaylink/displaylink.log`?

SimpleBinary, 2020-06-22:
@elguero The output of `/var/log/displaylink/displaylink.log` is:

```
Creating symlink /var/lib/dkms/evdi/1.7.0/source ->
                 /usr/src/evdi-1.7.0

DKMS: add completed.
Error! echo
Your kernel headers for kernel 5.6.6-300.fc32.x86_64 cannot be found at
/lib/modules/5.6.6-300.fc32.x86_64/build or /lib/modules/5.6.6-300.fc32.x86_64/source.
You can use the --kernelsourcedir option to tell DKMS where it's located.
Error! echo
Your kernel headers for kernel 5.6.6-300.fc32.x86_64 cannot be found at
/lib/modules/5.6.6-300.fc32.x86_64/build or /lib/modules/5.6.6-300.fc32.x86_64/source.
You can use the --kernelsourcedir option to tell DKMS where it's located.

-------- Uninstall Beginning --------
Module:  evdi
Version: 1.7.0
Kernel:  5.6.6-300.fc32.x86_64 (x86_64)
-------------------------------------

Status: Before uninstall, this module version was ACTIVE on this kernel.
```

elguero, 2020-06-22:
@SimpleBinary Can you post the result of running `uname -r`?  Also, please run `sudo rpm -qa | grep kernel`.

SimpleBinary, 2020-06-22:
@elguero `uname -r`:
```
5.6.6-300.fc32.x86_64
```
and `sudo rpm -qa | grep kernel`:

```
kernel-devel-5.6.19-300.fc32.x86_64
kernel-modules-5.6.6-300.fc32.x86_64
kernel-5.6.6-300.fc32.x86_64
libreport-plugin-kerneloops-2.12.0-3.fc32.x86_64
kernel-core-5.6.6-300.fc32.x86_64
kernel-headers-5.6.18-300.fc32.x86_64
abrt-addon-kerneloops-2.14.0-2.fc32.x86_64
kernel-modules-extra-5.6.6-300.fc32.x86_64
```

elguero, 2020-06-22:
Is there a reason you are locked at kernel-5.6.6?  Your kernel-devel package is at 5.6.19 and kernel-headers package is at 5.6.18. The latest kernel for Fedora 32 is 5.6.19.

If you are not able to use the latest kernel that Fedora 32 is up to, you need to try to match the kernel-devel and kernel-headers to the version of the kernel you are running.  I think this is the issue.

SimpleBinary, 2020-06-22:
@elguero After updating, everything is working perfectly. Thanks for your help.

elguero, 2020-06-22:
@SimpleBinary Great!

elguero, 2020-06-24:
@marzukia Were you able to get this working by installing the headers and being on the latest kernel?

marzukia, 2020-06-26:
> @marzukia Were you able to get this working by installing the headers and being on the latest kernel?

Apologies - didn't follow up on this sooner. My headers are already installed, I tried again with reinstalled headers with no luck

elguero, 2020-06-26:
@marzukia What kernel are you running?

Please include the following info:

`uname -a`
Include your latest `/var/log/displaylink/displaylink.log`
`rpm -qa | grep kernel`

closed 2020-07-08 by marzukia

elguero, 2020-07-09:
@marzukia No worries.  Out of your list, pretty sure it was secure boot causing your issues.

I would recommend using Wayland and turn SELinux back on.

Glad you sorted it out.