displaylink-rpm/displaylink-rpm #139: On a fresh install dependencnies aren't set accordingly

gbraad, 2020-06-21:
When install the displaylink=rpm with yum it will not satisfy development packages needs"

```
usr/sbin/dkms: line 98: make: command not found
```

elguero, 2020-06-22:
@gbraad Can you post any logs? (`/var/log/displaylink/displaylink.log`)

Then try running `sudo dnf install make`.  Please report back.  We may need to make adjustments to .spec file to check that tools needed by dkms are installed.

gbraad, 2020-06-22:
I already installed make, but using groupinstall "Development Tools".

Will check tomorrow if just 'make' satisfies

elguero, 2020-06-22:
I can confirm that `make` needs to be specified as being required to install the rpm since dkms needs to use it to build the evdi module.

Thank you for your help.

commit adf996b4 (2020-06-22): Add the requirement to have 'make' installed

If 'make' is not installed on the system, then dkms is unable to
build the evdi module.

Resolves #139
closed 2020-06-22 by elguero