displaylink-rpm/displaylink-rpm #141: Failed to start DisplayLink Manager Service on Fedora 32

alexandrulesi, 2020-06-23:
I've tried to get displaylink running on my machine and can't seem to get it to start. Installation of `fedora-32-displaylink-1.7.0-1.x86_64.rpm` worked without any issues.

Running `systemctl start displaylink` gets me:

```
Job for displaylink.service failed because the control process exited with error code.
See "systemctl status displaylink.service" and "journalctl -xe" for details.
```

Then calling `systemctl status displaylink.service` shows:

```
● displaylink.service - DisplayLink Manager Service
     Loaded: loaded (/usr/lib/systemd/system/displaylink.service; static; vendor preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Tue 2020-06-23 22:52:38 CEST; 2s ago
    Process: 40164 ExecStartPre=/sbin/modprobe evdi (code=exited, status=1/FAILURE)
        CPU: 4ms
```

And running `modprobe evdi` gives:

```
modprobe: ERROR: could not insert 'evdi': Operation not permitted
```

Not sure what to do from here. Any suggestions?

elguero, 2020-06-24:
@alexandrulesi Are you using "sudo"?  `sudo systemctl start displaylink`

How did you install the RPM?  Did you have any prior version running on that machine?

Please include logs and system info:
`/var/log/displaylink/displaylink.log`
`uname -a`

FranceBB, 2020-06-26:
Hi there, Alex.
I'm running Fedora 32 as well on my laptop and I can confirm that the module does not start with the very latest version of the kernel, however it does work (with several issues) with kernel 5.3.8-300.

elguero, 2020-06-26:
@FranceBB Kernel 5.3?  I thought that Fedora 32 was released with kernel 5.6?

In order to provide any help, I need folks to provide at the minimum the following information:

Logs: `/var/log/displaylink/displaylink.log`.
`uname -a`
`rpm -qa | grep kernel`

I running the latest Fedora 32 as well with no issues.

The logs and the steps of how you installed the RPM.  For instance, fresh install of F32 and then before running `dnf update` installed the rpm; Fresh install of F32, ran `dnf update`, rebooted, then installed the RPM;  Upgrade from F31; Upgrading from a prior version of the displaylink rpms provided here. etc.

alexandrulesi, 2020-06-27:
Hi @elguero,

here are the logs: 

```
Creating symlink /var/lib/dkms/evdi/1.7.0/source ->
                 /usr/src/evdi-1.7.0

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
make -j12 KERNELRELEASE=5.6.19-300.fc32.x86_64 all INCLUDEDIR=/lib/modules/5.6.19-300.fc32.x86_64/build/include KVERSION=5.6.19-300.fc32.x86_64 DKMS_BUILD=1....
cleaning build area....

DKMS: build completed.

evdi.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.6.19-300.fc32.x86_64/extra/
Adding any weak-modules

depmod.....

DKMS: install completed.
```

`uname -a` 
`Linux localhost.localdomain 5.6.19-300.fc32.x86_64 #1 SMP Wed Jun 17 16:10:48 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux`

`rpm -qa | grep kernel`
```
kernel-modules-extra-5.3.7-301.fc31.x86_64
kernel-devel-5.6.11-200.fc31.x86_64
libreport-plugin-kerneloops-2.13.1-4.fc32.x86_64
kernel-core-5.3.7-301.fc31.x86_64
kernel-devel-5.6.19-200.fc31.x86_64
kernel-5.3.7-301.fc31.x86_64
kernel-modules-5.6.19-300.fc32.x86_64
kernel-modules-5.6.19-200.fc31.x86_64
kernel-core-5.6.19-300.fc32.x86_64
texlive-l3kernel-svn51745-19.fc32.noarch
kernel-modules-extra-5.6.19-200.fc31.x86_64
kernel-modules-extra-5.6.19-300.fc32.x86_64
kernel-headers-5.6.18-300.fc32.x86_64
kernel-5.6.19-300.fc32.x86_64
kernel-core-5.6.19-200.fc31.x86_64
abrt-addon-kerneloops-2.14.2-2.fc32.x86_64
kernel-modules-5.3.7-301.fc31.x86_64
kernel-devel-5.6.19-300.fc32.x86_64
kernel-5.6.19-200.fc31.x86_64
```
It's the first version I'm installing. I got it from the latest releases on Github. Tried using sudo to run it as well.

aia832003, 2020-06-27:
Not sure if this is related, but was getting similar message and not supporting my monitors on new laptop. I manually did a make and make install on the evdi library from https://github.com/DisplayLink/evdi/releases/tag/v1.7.0 and it immediately recognized my displays.

elguero, 2020-06-27:
@alexandrulesi Hmm.  It looks like dkms was able to build the evdi library and install the module  So, now to figure out why the Displaylink manager is not starting.

Are there any clues in `dmesg`?

When you run `lsmod | grep evdi`, what is the output?  If the evdi module is loaded, you should get some output.
Maybe `dmesg | grep evdi`?

@aia832003 The RPM from this repo is using the evdi library from that same tag.  I think we are doing the same thing as the Displaylink `make install` target but through the RPM.  Without logs and seeing what happened, it is hard to say what was missed by the RPM.

elguero, 2020-06-27:
Duplicate of #138

alexandrulesi, 2020-06-27:
@elguero 

Here we go. There's a message on repeat in `dmesg`:

```
Lockdown: modprobe: unsigned module loading is restricted; see man kernel_lockdown.7
```

No output for the other two commands

elguero, 2020-06-29:
@alexandrulesi With secure boot turned on, as you have discovered you cannot load unsigned modules.  There are instructions out there on how to sign the evdi module.  This appears to fall on the end user to do right now.  In doing some research, it would seem that dkms has been adding some tools to help automate this better.  It does not appear to have made its way into F32 yet.

https://github.com/dell/dkms/search?q=sign&type=Commits

It would seem you have 2 choices right now.  You can go through the process of signing the module or you can turn off secure boot.

alexandrulesi, 2020-07-13:
Thank you for the clarification! I'll think about it.

Ville1ero, 2020-07-23:
I was able to make DisplayLink work on Fedora 32 KDE Plasma.

You can see some settings and screenshots on this link:
https://bit.ly/3fXxdb1

sourcecodere, 2021-03-04:
This is how I make displaylink-rpm work with secure boot enabled on Fedora 33. May be
helpful if you struggle:

- Create a self signed signing key:
```
# openssl req -new -x509 -newkey rsa:2048 -keyout MOK.priv -outform DER -out \
MOK.der -nodes -days 36500 -subj "/CN=Displaylink/"
```

- Register key to secure boot:
```
# sudo mokutil --import MOK.der
```
- Reboot to enroll the MOK (Machine Owner Key)

- Then for every kernel upgrade the unxz version of the module must be signed:
```
# sudo modinfo -n evdi
/lib/modules/5.10.19-200.fc33.x86_64/extra/evdi.ko.xz
# sudo unxz $(modinfo -n evdi)
# sudo /usr/src/kernels/$(uname -r)/scripts/sign-file sha256 ./MOK.priv \
./MOK.der /lib/modules/$(uname -r)/extra/evdi.ko
# xz -f /lib/modules/$(uname -r)/extra/evdi.ko
```
Now the DisplayLink DP and HDMI port should just work:)

Thank you @elguero for maintaining this repo! Great work!

elguero, 2021-03-05:
@sourcecodere Nice write up!  That is the way to do it.  I looked into this to see if there was a way to make this seamless but there doesn't seem to be a way to do that through RPM.  It is going to require the end user to use MOK, as you very nicely outlined.

There does appear to be a way that DKMS can automate signing the module when there is a kernel upgrade.  I haven't played with it at all.  The framework is there though to setup a script to be run.

Thank you for adding the note.  Maybe we should add this to the README.md file or add a file to the repo with this helpful documenation?  Feel free to send a PR if you have to time to do so.

sourcecodere, 2021-03-09:
@elguero thank you. I have not seen any god solutions on building RPM and signing them for secure boot either. I guess MOK is the way to go. Made a PR you can look into. I think that having s receipt for MOK signing in the repo would be great!

elguero, 2021-03-09:
@sourcecodere Thank you for the contribution!

closed 2021-03-09 by elguero

magic-muschrum, 2024-09-18:
Hi, I just wanted to thank everybody for the contribution and the information here, as it fixed the very same problem on fedora 40 for me. Much appreciated. Added my own signed signing key. @sourcecodere, I can only agree with @elguero, thanks for the nice write-up!

PR #159 (2021-03-09): Added secure boot instructions to README.md
fixes #141