displaylink-rpm/displaylink-rpm #142: 1fps on Fedora 32 kernel 5.3.8-300

FranceBB, 2020-06-26:
Hi there,
I have an Asus N552VX Laptop (i7 6700HQ, NVIDIA GTX 950M, 32GB RAM DDR4) running Fedora 32 kernel 5.3.8-300 and connected to a Dell D3100 Dock Station with two external HP monitors FULL HD 60Hz.
The issue I have is that although I managed to install the drivers and run them correctly, as soon as I close my laptop lid, I get 1fps on both monitors. If I keep my laptop lid open, I have three monitors and the refresh rate is indeed 60Hz, however I get all sort of graphical glitches with parts of the screen superimposing old parts that haven't been refreshed and it quickly becomes unusable. The driver I installed is displaylink-1.7.0-1..x86_64.

elguero, 2020-06-26:
@FranceBB Why are you running kernel 5.3.8-300 on F32?

elguero, 2020-06-26:
Please see this thread as well: https://github.com/DisplayLink/evdi/issues/181#issuecomment-643955576

FranceBB, 2020-06-29:
I'm using it for compatibility reasons as new kernels break a program I use, so I gotta stick with that. Anyway, I saw the thread and I saw why it's reverting to 1fps: when the laptop lid is closed, it has no screen to sync to, therefore the function returns NULL and the two Display Link monitors run at 1fps, which is the intended behavior, although it's far from being ideal...

elguero, 2020-06-29:
It looks like Displaylink is working on getting this fixed.

Is it okay to close out this issue and you can continue to follow on Displaylink's repo?

FranceBB, 2020-06-30:
Yep, sure, I'm gonna keep a close eye on the Displaylink's repository. 
I'm gonna close the issue.

closed 2020-06-30 by FranceBB