displaylink-rpm/displaylink-rpm #143: fedora 32, not seeing screen attached to displaylink

ericfaure007, 2020-07-06:
Hi,

I don't see the screen attached to the displaylink.
The evdi  module is compiled, but isn't loaded automatically, and systemd displaylink is not started neither.
I can load evdi module and start  systemd displaylink, but I still can't see the screen.

Fedora 32
running Xorg

dkms status
evdi, 1.7.0, 5.7.6-201.fc32.x86_64, x86_64: installed
evdi, 1.7.0, 5.7.7-200.fc32.x86_64, x86_64: installed

when loading evdi via modprobe evdi command:
evdi: [I] Initialising logging on level 5

elguero, 2020-07-06:
@ericfaure007 Do you have secure boot enabled?

Please share logs and system info: 

`/var/log/displaylink/displaylink.log`
`systemctl status displaylink`
`lsmod | grep evdi`
`uname -a`

What chipset from DisplayLink is being used?

Thanks,

ericfaure007, 2020-07-07:
cat /var/log/displaylink/displaylink.log 

Creating symlink /var/lib/dkms/evdi/1.7.0/source ->
                 /usr/src/evdi-1.7.0

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
make -j4 KERNELRELEASE=5.7.6-201.fc32.x86_64 all INCLUDEDIR=/lib/modules/5.7.6-201.fc32.x86_64/build/include KVERSION=5.7.6-201.fc32.x86_64 DKMS_BUILD=1.....
cleaning build area....

DKMS: build completed.

evdi.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.7.6-201.fc32.x86_64/extra/
Adding any weak-modules

depmod......

DKMS: install completed.

systemctl status displaylink
● displaylink.service - DisplayLink Manager Service
     Loaded: loaded (/usr/lib/systemd/system/displaylink.service; static; vendor preset: enabled)
     Active: active (running) since Mon 2020-07-06 15:21:08 CEST; 19h ago
    Process: 4636 ExecStartPre=/sbin/modprobe evdi (code=exited, status=0/SUCCESS)
   Main PID: 4637 (DisplayLinkMana)
      Tasks: 13 (limit: 14191)
     Memory: 8.8M
     CGroup: /system.slice/displaylink.service
             └─4637 /usr/libexec/displaylink/DisplayLinkManager

Jul 06 15:21:08 star.alpegagroup.com systemd[1]: Starting DisplayLink Manager Service...
Jul 06 15:21:08 star.alpegagroup.com systemd[1]: Started DisplayLink Manager Service.

lsmod | grep evdi
evdi                   65536  0
drm_kms_helper        249856  2 evdi,i915
drm                   618496  10 drm_kms_helper,evdi,i915

uname -a
Linux mycomputer 5.7.7-200.fc32.x86_64 #1 SMP Wed Jul 1 19:53:01 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

rpm -qa | grep kernel
kernel-core-5.7.6-201.fc32.x86_64
kernel-headers-5.7.7-200.fc32.x86_64
abrt-addon-kerneloops-2.14.2-2.fc32.x86_64
kernel-modules-extra-5.7.7-200.fc32.x86_64
kernel-5.7.6-201.fc32.x86_64
kernel-modules-extra-5.6.19-300.fc32.x86_64
kernel-devel-5.7.7-200.fc32.x86_64
kernel-modules-5.7.7-200.fc32.x86_64
kernel-core-5.6.19-300.fc32.x86_64
kernel-devel-5.6.19-300.fc32.x86_64
kernel-modules-extra-5.7.6-201.fc32.x86_64
kernel-5.6.19-300.fc32.x86_64
kernel-devel-5.7.6-201.fc32.x86_64
kernel-5.7.7-200.fc32.x86_64
libreport-plugin-kerneloops-2.13.1-4.fc32.x86_64
kernel-modules-5.7.6-201.fc32.x86_64
kernel-core-5.7.7-200.fc32.x86_64
kernel-modules-5.6.19-300.fc32.x86_64

Where Can I get the chipset info?

I'm going to test with wayland, but the displaylink is at office and I will go back there only next monday.

elguero, 2020-07-07:
Everything looks right.  I am trying to figure out if you have a supported device.

Could you try running `usb-devices`?  Find the devices that are DisplayLink.

For instance, here is from my system:
```
T:  Bus=04 Lev=03 Prnt=03 Port=01 Cnt=01 Dev#=  4 Spd=5000 MxCh= 0
D:  Ver= 3.20 Cls=ef(misc ) Sub=02 Prot=01 MxPS= 9 #Cfgs=  1
P:  Vendor=17e9 ProdID=4301 Rev=34.16
S:  Manufacturer=DisplayLink
S:  Product=Plugable USB3-HDMI-DVI
S:  SerialNumber=PLUG[redacted]
C:  #Ifs= 5 Cfg#= 1 Atr=a0 MxPwr=504mA
I:  If#=0x0 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=00 Prot=03 Driver=usbfs
I:  If#=0x1 Alt= 0 #EPs= 0 Cls=fe(app. ) Sub=01 Prot=01 Driver=usbfs
I:  If#=0x2 Alt= 0 #EPs= 1 Cls=01(audio) Sub=01 Prot=20 Driver=snd-usb-audio
I:  If#=0x3 Alt= 0 #EPs= 0 Cls=01(audio) Sub=02 Prot=20 Driver=snd-usb-audio
I:  If#=0x4 Alt= 0 #EPs= 0 Cls=01(audio) Sub=02 Prot=20 Driver=snd-usb-audio

T:  Bus=04 Lev=03 Prnt=03 Port=02 Cnt=02 Dev#=  5 Spd=5000 MxCh= 0
D:  Ver= 3.20 Cls=ef(misc ) Sub=02 Prot=01 MxPS= 9 #Cfgs=  1
P:  Vendor=17e9 ProdID=4301 Rev=34.63
S:  Manufacturer=DisplayLink
S:  Product=Plugable USB3-HDMI-DVI
S:  SerialNumber=PLUG[redacted]
C:  #Ifs= 5 Cfg#= 1 Atr=a0 MxPwr=504mA
I:  If#=0x0 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=00 Prot=03 Driver=usbfs
I:  If#=0x1 Alt= 0 #EPs= 0 Cls=fe(app. ) Sub=01 Prot=01 Driver=usbfs
I:  If#=0x2 Alt= 0 #EPs= 1 Cls=01(audio) Sub=01 Prot=20 Driver=snd-usb-audio
I:  If#=0x3 Alt= 0 #EPs= 0 Cls=01(audio) Sub=02 Prot=20 Driver=snd-usb-audio
I:  If#=0x4 Alt= 0 #EPs= 0 Cls=01(audio) Sub=02 Prot=20 Driver=snd-usb-audio
```

ericfaure007, 2020-07-08:
Secure boot is disabled.

As soon as I go to the office, next monday, I will give you the ouput.

Ville1ero, 2020-07-23:
I was able to make DisplayLink work on Fedora 32 KDE Plasma.

You can see some settings and screenshots on this link:
https://bit.ly/3fXxdb1

ericfaure007, 2020-07-24:
I didn't have the opportunity to go the office yet and I will go on holidays.

ericfaure007, 2020-10-06:
I'm at the office now, It's ok now, it works, physical connections were wrong.
Thanks a lot for your help :-)

closed 2020-10-06 by ericfaure007