displaylink-rpm/displaylink-rpm #144: F31 - Display sluggish when native display is turned off

JetStreamVelocity, 2020-07-19:
I have noticed the display turns sluggish when I turn off the native display and use only the 2 displays I have connected to the dock. 

I have a laptop and a minipc running F31 where I see this behaviour (and also a laptop running F32). As soon as I close the laptop lid or disconnect the external display on the minipc, I see this behaviour. It takes about 3 seconds from a mouse gesture for a response to come up. Would be nice if the native display could be turned off.

As soon as the laptop display is turned on or external display on the minipc, the display is responsive again.

Wondering if anyone has encountered this issue.

Please let me know if I could help with more info / logs.

My dock: `Dell D3100`

Display: `Xorg`

uname -a:
`Linux nucv210 5.6.16-200.fc31.x86_64 #1 SMP Thu Jun 4 18:07:41 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux`

> ● displaylink.service - DisplayLink Manager Service
   Loaded: loaded (/usr/lib/systemd/system/displaylink.service; static; vendor preset: enabled)
   Active: active (running) since Sun 2020-07-19 01:14:34 BST; 32min ago
  Process: 1257 ExecStartPre=/sbin/modprobe evdi (code=exited, status=0/SUCCESS)
 Main PID: 1266 (DisplayLinkMana)
    Tasks: 43 (limit: 38368)
   Memory: 96.7M
      CPU: 11min 970ms
   CGroup: /system.slice/displaylink.service
           └─1266 /usr/libexec/displaylink/DisplayLinkManager
>
> Jul 19 01:14:34 nucv210 systemd[1]: Starting DisplayLink Manager Service...
Jul 19 01:14:34 nucv210 systemd[1]: Started DisplayLink Manager Service.

rpm -qa | grep kernel
> kernel-modules-5.6.15-200.fc31.x86_64
kernel-modules-5.7.8-100.fc31.x86_64
kernel-devel-5.7.8-100.fc31.x86_64
kernel-core-5.6.16-200.fc31.x86_64
kernel-devel-5.6.15-200.fc31.x86_64
libreport-plugin-kerneloops-2.13.1-3.fc31.x86_64
kernel-modules-extra-5.6.16-200.fc31.x86_64
kernel-headers-5.7.7-100.fc31.x86_64
kernel-core-5.6.15-200.fc31.x86_64
kernel-modules-5.6.16-200.fc31.x86_64
kernel-modules-extra-5.6.15-200.fc31.x86_64
kernel-5.6.15-200.fc31.x86_64
kernel-core-5.7.8-100.fc31.x86_64
kernel-modules-extra-5.7.8-100.fc31.x86_64
abrt-addon-kerneloops-2.14.2-2.fc31.x86_64
kernel-devel-5.6.16-200.fc31.x86_64

laurapoggio-sptools, 2020-07-20:
I have the same behaviour.

My docking station is Lenovo ThinkPad Hybrid Usb C

Display (I think) `Xorg`, I use xfce

`Linux_laptop 5.7.8-100.fc31.x86_64 #1 SMP Thu Jul 9 15:16:52 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux` with Fedora31

Is this happening also with F32? Because I could think to upgrade if this works on F32.

On the positive, since using kernel 5.7.x, the random crashes of the xdisplay stopped.

JetStreamVelocity, 2020-07-20:
> Is this happening also with F32? Because I could think to upgrade if this works on F32.

Yes, seeing this with F32 as well.

elguero, 2020-07-20:
Please have a look here: https://github.com/DisplayLink/evdi/issues/181

JetStreamVelocity, 2020-07-20:
Thanks @elguero I'll look forward to a `displaylink-rpm` release after a fix gets pushed to `evdi`. 

I'll leave it to you how you want to track i.e. if you want to mark this issue as closed.

elguero, 2020-07-20:
@JetStreamVelocity Well, I believe it is more of an issue with Xserver, rather than an evdi issue.  I would follow the other thread as to when the PR from the Display Link / evdi team is accepted in Xserver.  Then it is a matter of the Xserver fix making its way into Fedora 32 through an update.

closed 2020-07-20 by elguero