displaylink-rpm/displaylink-rpm #145: Sluggish performance on Fedora32-Gnome

Arthedian, 2020-07-28:
Hello,
I have 3 connected displays to my notebook (Fedora 32-Gnome). All three is Full HD. One of them is connected through HDMi directly to my laptop and two are connected through one displaylink device through USB 3.0. I recently upgraded from Fedora 31 where it works way better than now. Now it performs very slow, on my HDMI displaylink is everything ok, but od Displaylink displays is very slow (few frames per second) and on some occasion it is showing artefacts on it, mostly if some special dialogue pop up and rest of the screen is dimmed. What information should I add to help find the problem?

elguero, 2020-07-29:
Seems like this thread might be describing what you are experiencing?

https://github.com/DisplayLink/evdi/issues/186

Arthedian, 2020-07-30:
Not really, because they are talking about delay. I only have few frames per seconds so it is not running fluently at all.

elguero, 2020-07-30:
There is this one too, maybe:

https://github.com/DisplayLink/evdi/issues/181

Arthedian, 2020-08-21:
I wasnt sure at first, but even with main screen on, it wont help

nicobao, 2020-08-24:
> There is this one too, maybe:
> 
> [DisplayLink/evdi#181](https://github.com/DisplayLink/evdi/issues/181)

I run Fedora 32 and the link you mentioned corresponds exactly to what I experience. It works fine when the laptop screen is on (doesn't need to be "main"). But whenever I turn the laptop screen off - things go wild. The mouse speed is OK but the clicks events latency (windows, buttons...) is terrible.

The DisplayLink team released a patch for Ubuntu 20.4: https://www.displaylink.org/forum/showthread.php?t=67148
It would be awesome if you could make one for Fedora 32 as well.
Other people need the patch for other distros as well: https://www.displaylink.org/forum/member.php?u=20834
Not sure about the plan here for release a stable version that will work.
Thanks for your work on this.

elguero, 2020-08-24:
@nicobao The efforts by the DisplayLink team will benefit the Fedora distro once it is included in a future release by the XServer community and that release is incorporated into Fedora.  It appears that it is still being actively worked on and the final solution has not been attained yet.  This is an issue with XServer, not the DisplayLink drivers - from what I understand.

The DisplayLink team produced a build of XServer (not DisplayLink drivers) for Ubuntu users in the meantime.  It is a development build with potential problems.

My thoughts are that this is out of the scope of this project here which is just to help make available DisplayLink drivers for those using the Fedora distro.

Someone in this community would need to create a development build of XServer that has the patch and keep it up to date as the patches are updated.  This would not go into this repository though.

While waiting, the quickest workaround is to enable your laptop screen.

nicobao, 2020-08-24:
Thank you for your quick and comprehensive reply!
Keep up the good work!

Cheers,
Nicolas

nicobao, 2020-08-24:
> I wasnt sure at first, but even with main screen on, it wont help

 Are you using the latest EVDI version?
(https://github.com/DisplayLink/evdi/releases)
I didn't install those at first, and I experienced small lag even with the laptop screen on.

elguero, 2020-08-24:
@nicobao The latest EVDI release, 1.8.0, does not work with the current DisplayLink drivers.  We are waiting for a new release.

elguero, 2020-10-26:
It looks like DisplayLink/evdi closed the issue on their side and it looks like upstream has merged the patch as well.

Closing since this thread has been inactive for 3 months now.

closed 2020-10-26 by elguero

nicobao, 2020-10-26:
> @nicobao The latest EVDI release, 1.8.0, does not work with the current DisplayLink drivers. We are waiting for a new release.

I had to `git clone` the evdi repo and use make/make install to get a workable environment as their latest release didn't work.