displaylink-rpm/displaylink-rpm #147: Evdi module is not built automatically when installing new rpm

wernermaes, 2020-09-14:
Hello

I've upgraded the kernel package from kernel 5.7.14-200 to 5.8.7-200 but the evdi module is not being built for the updatet kernel.

 modprobe evdi
modprobe: FATAL: Module evdi not found in directory /lib/modules/5.8.7-200.fc32.x86_64/

The one for kernel 5.7.14-200 exist (have to test it though)
var/lib/dkms/evdi/1.7.0/5.7.14-200.fc32.x86_64/x86_64/module/evdi.ko.xz

Can I somehow force this build?

Kind regards

Werner

wernermaes, 2020-09-14:
Hello

I did a force reinstall of the displaylink rpm and that has resolved the issue.
So that's ok but how can I make sure that the module is being built after a kernel upgrade?

Werner

elguero, 2020-09-14:
DKMS has been handling kernel upgrades and it has been working fairly well.

I noticed this too recently on a single kernel upgrade.  I didn't pay too much attention to it and I am not sure which kernel. I figured it was just a one time thing. I am not sure if there is something specific to this kernel.  The easy fix was to re-install as you mentioned.

We will have to see what happens with the next kernel update.  Then we can dig into further and see if DKMS changed something.

The logs would be helpful to see:
`/var/log/displaylink/displaylink.log`
`/var/lib/dkms/evdi/kernel-[version]/log/make.log`

wernermaes, 2020-09-15:
Hello 

I noticed this on these this kernel:
5.8.6.201 ==> there is no /var/lib/dkms/evdi/kernel-5.8.6.201.fc23-x86_64 folder.
Also there is nothing in /var/log/displaylink/displaylink.log that refers to that kernel.

5.8.7-200 did not work either, I reinstalled displaylink rpm.

elguero, 2020-09-15:
Not sure what to tell you.  Open to suggestions from the community.

I just did a `sudo dnf update` this morning and the process installed `5.8.8-200.fc32.x86_64`.  The evdi module was removed from the kernel being erased and built for the new kernel being installed.  The messages stated that everything was built successfully.  I rebooted my machine and can confirm that everything is working as expected.

wernermaes, 2020-09-15:
Strange
Maybe it's because I keep up to 10 kernels (installonly_limit=10 in dnf.conf).
But I doubt that

bartoszmajsak, 2020-09-20:
I was on 5.8.7 and everything worked fine, after upgrade to 5.8.9 it stopped working, causing X/Gnome to not start properly. I also tried this https://fedora.pkgs.org/32/cheese-x86_64/displaylink-1.8.0-1.fc32.x86_64.rpm.html (with EVDI 1.8.0) but it is giving me the same result.

elguero, 2020-09-20:
For support, please include the following such as:

`/var/log/displaylink/displaylink.log`
`/var/lib/dkms/evdi/kernel-[version]/log/make.log`
`rpm -qa | grep kernel`
`uname -a`
`dmesg | grep evdi`
`dmesg | grep Displaylink`

In regards to using that Fedora package you found, unless I am mistaken, I did not think that the evdi library 1.8 was working with the current official Ubuntu driver.  That is why we are have not released anything yet with 1.8.0 in it.  Maybe I am wrong but we have not been able to get 1.8 to work with the currently released Ubuntu release.  I have seen similar comments in the official EVDI repo.

bartoszmajsak, 2020-09-21:
> In regards to using that Fedora package you found, unless I am mistaken, I did not think that the evdi library 1.8 was working with the current official Ubuntu driver. That is why we are have not [...]

@elguero Thanks for your explanation. But primarily for this great module :) I tried all possible things I found :) Do you want me to open a separated issue with my exact details?

wernermaes, 2020-09-21:
> > 
> I was on 5.8.7 and everything worked fine, after upgrade to 5.8.9 it stopped working, causing X/Gnome to not start properly. 

I'll suppose I will delay upgrading to  5.8.9 :)

elguero, 2020-09-22:
@wernermaes I have been running 5.8.9 without any issues.

@bartoszmajsak If you feel that your issue is unrelated to this one, sure, you can open up a separate issue with details.

I just updated to 5.8.10 without any issues this morning.  I am using the same released package as found on this site on Fedora 32 using Wayland.

```
  Running scriptlet: kernel-modules-5.8.7-200.fc32.x86_64                                                                                                                                                                             116/140 
  Running scriptlet: kernel-core-5.8.7-200.fc32.x86_64                                                                                                                                                                                117/140 
dkms: removing: evdi 1.7.0 (5.8.7-200.fc32.x86_64) (x86_64)

-------- Uninstall Beginning --------
Module:  evdi
Version: 1.7.0
Kernel:  5.8.7-200.fc32.x86_64 (x86_64)
-------------------------------------

Status: Before uninstall, this module version was ACTIVE on this kernel.
Removing any linked weak-modules

evdi.ko.xz:
 - Uninstallation
   - Deleting from: /lib/modules/5.8.7-200.fc32.x86_64/extra/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

depmod....

DKMS: uninstall completed.

  Erasing          : kernel-core-5.8.7-200.fc32.x86_64                                                                                                                                                                                117/140 
warning: file /lib/modules/5.8.7-200.fc32.x86_64/updates: remove failed: No such file or directory
```

```
 Running scriptlet: kernel-core-5.8.10-200.fc32.x86_64                                                                                                                                                                               140/140 
dkms: running auto installation service for kernel 5.8.10-200.fc32.x86_64

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
make -j8 KERNELRELEASE=5.8.10-200.fc32.x86_64 all INCLUDEDIR=/lib/modules/5.8.10-200.fc32.x86_64/build/include KVERSION=5.8.10-200.fc32.x86_64 DKMS_BUILD=1.....
cleaning build area....

DKMS: build completed.

evdi.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.8.10-200.fc32.x86_64/extra/
Adding any weak-modules

depmod.........

DKMS: install completed.
 Done. 
dkms: running auto installation service for kernel 5.8.10-200.fc32.x86_64
 Done. 
```

One thing you all might try is uninstalling the package, making sure nothing is left behind, rebooting your system and then installing the released package.

Check that the following is not left on your system before installing the package again:
`/var/lib/dkms/evdi`
`/usr/libexec/displaylink`
`/usr/src/evdi-*`

`/etc/X11/xorg.conf.d/20-displaylink.conf`
`/etc/logrotate.d/displaylink`
`/usr/lib/systemd/system/displaylink.service`
`/usr/lib/systemd/system-preset/95-displaylink.preset`
`/usr/lib/systemd/system-sleep/displaylink.sh`
`/etc/udev/rules.d/99-displaylink.rules`/usr/share/doc/displaylink
`/usr/share/doc/displaylink/LICENSE`

```/usr/lib/modules/`uname -r`/weak-updates/evdi.ko.xz```
```/usr/lib/modules/`uname -r`/extra/evdi.ko.xz```

Please paste any logs or error messages you get when installing the RPM package.

wernermaes, 2020-09-24:
@elguero I will let you know if my kernel will cause any problems.