displaylink-rpm/displaylink-rpm #149: Can't see monitor with Dell 3100 Docking Station (Fedora 33)

FranceBB, 2020-10-23:
I installed the very latest version and although I can see audio devices and ethernet as well as attached USB devices, I can't really see the monitor.
I'm running Fedora 33, kernel 5.8.15-301.fc33.x86_64

elguero, 2020-10-26:
Wow. You are early.  I didn't think Fedora 33 was going to be released until tomorrow.  I don't think anyone will have had a chance to test this out yet.

Thanks for the info.  We will make sure to test on Fedora 33 once we have a chance to update to the release and test.

Something that would be helpful with any troubleshooting would be the following.

Did you upgrade to Fedora 33?  Clean install?

Logs?  Dmesg, journalctl, /var/log/displaylink/displaylink.log, /var/lib/dkms/evdi/1.7.0/5.8.15-301.fc33.x86_64/x86_64/log/make.log

elguero, 2020-10-27:
I do not have a Dell D3100 and am unable to speak about that specific device.

I just upgraded from Fedora 32 to 33.  Everything upgraded smoothly and appears to be still working.

Logs are crucial to trying to offer any advice on how to resolve.  Once you have some logs uploaded, then maybe others in the community can chime in with some help.  If anyone is using a Dell D3100, that would be helpful as well since I am limited in being able to support that device.

Thanks.

FranceBB, 2020-10-29:
Well, what can I say, I was early xD

Anyway, those are the logs from /var/log/displaylink/displaylink.log
[displaylink.log](https://github.com/displaylink-rpm/displaylink-rpm/files/5456672/displaylink.log)
And those are the logs from /var/lib/dkms/evdi/1.8.0/5.8.15-301.fc33.x86_64/x86_64/log
[make.log](https://github.com/displaylink-rpm/displaylink-rpm/files/5456674/make.log)

Unfortunately, I don't own the Dell D3100, but every time I get to work, there are all those things like "smart workplace" in which you can basically sit wherever you like with your laptop, connect the USB 3.0 and you're ready to go. Sadly, though, it ain't really true with Fedora. Still, let me know if you see something unusual from the logs. I also tried the 1.8.0 version of the displaylink-rpm since the 1.7.0 you provided didn't work, but to no avail. Reading on the web, I noticed that a lot of people were complaining about the Display Port support and indeed the HP EliteDisplay E243m is connected through the Display Port and not via HDMI to the Dell D3100 docking station.

elguero, 2020-10-29:
There is a reason why we have not created a build with evdi 1.8.0.  That release does not work with the DisplayLink drivers.  We are waiting for a new release from DisplayLink (https://www.displaylink.com/downloads/ubuntu).

In looking at your displaylink.log, there is some interesting things.  You may need to clean some things out.  I see in the logs DKMS built things for Fedora 32 and then later built something for Fedora 31, then jumped to Fedora 33 with evdi 1.8.0.

My recommendation would be to do a `sudo rpm -qa | grep displaylink` and remove all packages you have installed, `sudo rpm -evh [packages]`.

Then you need to clear out all evdi stuff.  Where you built evdi 1.8.0 manually, do a `make uninstall` to remove that package.

Then see what is left over and remove:

`/var/lib/dkms/evdi`
`/usr/libexec/displaylink`
`/usr/src/evdi-*`

`/etc/X11/xorg.conf.d/20-displaylink.conf`
`/etc/logrotate.d/displaylink`
`/usr/lib/systemd/system/displaylink.service`
`/usr/lib/systemd/system-preset/95-displaylink.preset`
`/usr/lib/systemd/system-sleep/displaylink.sh`
`/etc/udev/rules.d/99-displaylink.rules`/usr/share/doc/displaylink
`/usr/share/doc/displaylink/LICENSE`

```/usr/lib/modules/`uname -r`/weak-updates/evdi.ko.xz```
```/usr/lib/modules/`uname -r`/extra/evdi.ko.xz```

That should clean everything out.

Reboot your machine, selecting the Fedora 33 kernel installed.

Install the RPM release from this repo.  Hopefully that gets you back on the right path.  DKMS should take care of any kernel updates you get later on.

FranceBB, 2020-10-30:
It took me a while to remove the packages, check everything and clean everything up, but I eventually did it.
I rebooted, I re-installed the RPM and this time it targeted the right kernel and the right evdi version (1.7.0).
I rebooted again et voila it was working again! :+1: 
Thank you a lot.
The only thing is that when my laptop lid is closed, it gets the refresh rate of the monitor of the laptop which is basically 1Hz as it's disabled, so everything is incredibly slow, but this was the very same behavior I was experiencing with Fedora 32 before upgrading to 33 so I guess there's nothing I can do about it.

closed 2020-10-30 by FranceBB

elguero, 2020-10-30:
Excellent!  Glad to hear that it is working now.  My experiences in the past was that I would have to clean things out once and a while, especially between major kernel releases.

In the last release, the RPM spec was updated to let DKMS do its thing.  Changes were made to conform with the recommendations fo the DKMS project.  My experience has been that it has handled kernel upgrades, even between major upgrades, very smoothly, as long as artifacts from prior RPM releases had been cleaned up first.