displaylink-rpm/displaylink-rpm #152: Update evdi to 1.8 for support 5.6.xxx 5.7.xxx 5.8.xxx and 5.9.xxx Kernel line

chiddekel, 2020-11-03:
Update evdi to 1.8 for support 5.6.xxx 5.7.xxx 5.8.xxx and 5.9.xxx Kernel line.

elguero, 2020-11-03:
@chiddekel I think you are at the wrong repo.  The DisplayLink folks over at this repo, https://github.com/DisplayLink/evdi, are responsible for maintaining evdi.  I believe they have already patched the code and it is waiting for a release.

FYI: evdi 1.8 currently does not work with the latest DisplayLink drivers.  Waiting for that release.

closed 2020-11-03 by elguero

elguero, 2020-11-03:
@chiddekel I just re-read your sentence.  You are asking us to update the RPM to evdi 1.8.

The main reason why we have not updated with evdi 1.8 is that it simply does not work with the currently released DisplayLink drivers, v5.3.1. (https://www.displaylink.com/downloads/ubuntu)  There was an API change in evdi 1.8.  Once there is a release from DisplayLink, then we will work on updating to evdi 1.8.

Second, evdi 1.8 has not been released with the patch to support kernel 5.9.

I am not sure why you think we need to update to 1.8 in order to have support in the kernels you listed. The current RPM package built on this site works with kernels 5.6, 5.7 and 5.8.  Fedora has not updated to 5.9 yet.  They just had a test day last week.  It is probably 2 - 3 weeks out.

Sorry for not understanding at the beginning what you were looking for.