displaylink-rpm/displaylink-rpm #156: evdi not found

Arthedian, 2020-12-16:
Hello,
I downloaded latest version of rpm (v5.3.1-1) for fedora 33. I Installed it with command

> rpm -i fedora-33-displaylink-1.7.2-1.x86_64.rpm

. Finally i restarted the computer and look into the log of the service via

>  journalctl -u displaylink.service 

 and it is saying, that evdi module is not found. Do I need to installed it separately? Sadly I dont remember it from previous version which I had installed.

> 
> Dec 16 10:00:34 localhost.localdomain systemd[1]: Starting DisplayLink Manager Service...
> Dec 16 10:00:34 localhost.localdomain modprobe[17149]: modprobe: FATAL: Module evdi not found in directory /lib/modules/5.9.11-200.fc33.x86_64
> Dec 16 10:00:34 localhost.localdomain systemd[1]: displaylink.service: Control process exited, code=exited, status=1/FAILURE
> Dec 16 10:00:34 localhost.localdomain systemd[1]: displaylink.service: Failed with result 'exit-code'.
> Dec 16 10:00:34 localhost.localdomain systemd[1]: Failed to start DisplayLink Manager Service.
> Dec 16 10:00:40 localhost.localdomain systemd[1]: displaylink.service: Scheduled restart job, restart counter is at 7.
> Dec 16 10:00:40 localhost.localdomain systemd[1]: Stopped DisplayLink Manager Service.
>

UnleashedMV, 2020-12-16:
Hello 
i install the Version of "fedora-33-displaylink-1.7.2-1.x86_64.prm" with the "sudo dnf install" command. 

Cannot start displaylink-service 
I get this errors stack. 

`Nov 27 17:58:45 localhost.localdomain systemd[1]: displaylink.service: Scheduled restart job, restart counter is at 34616.
Nov 27 17:58:45 localhost.localdomain systemd[1]: Stopped DisplayLink Manager Service.
Nov 27 17:58:45 localhost.localdomain systemd[1]: Starting DisplayLink Manager Service...
Nov 27 17:58:45 localhost.localdomain modprobe[302886]: modprobe: FATAL: Module evdi not found in directory /lib/modules/5.9.9-200.fc33.x86_64
Nov 27 17:58:45 localhost.localdomain systemd[1]: displaylink.service: Control process exited, code=exited, status=1/FAILURE
Nov 27 17:58:45 localhost.localdomain systemd[1]: displaylink.service: Failed with result 'exit-code'.
Nov 27 17:58:45 localhost.localdomain systemd[1]: Failed to start DisplayLink Manager Service.
`

elguero, 2020-12-16:
Can you please provide the following:

* Kernel version `(uname -a)`

* `/var/log/displaylink/displaylink.log`
* `/var/lib/dkms/evdi/kernel-[version]/log/make.log`

Thanks

elguero, 2020-12-16:
@UnleashedMV 

I see a something is not in sync on your system:

`Nov 27 17:58:45 localhost.localdomain modprobe[302886]: modprobe: FATAL: Module evdi not found in directory /lib/modules/5.9.9-200.fc33.x86_64`

`ls /lib/modules -> 5.9.11-200.fc33.x86_64 5.9.12-200.fc33.x86_64 5.9.13-200.fc33.x86_64`

Your kernel versions do not match.

I would remove the RPM.

Check that nothing was left behind:
`/var/lib/dkms/evdi`
`/usr/libexec/displaylink`
`/usr/src/evdi-*`

`/etc/X11/xorg.conf.d/20-displaylink.conf`
`/etc/logrotate.d/displaylink`
`/usr/lib/systemd/system/displaylink.service`
`/usr/lib/systemd/system-preset/95-displaylink.preset`
`/usr/lib/systemd/system-sleep/displaylink.sh`
`/etc/udev/rules.d/99-displaylink.rules/usr/share/doc/displaylink`
`/usr/share/doc/displaylink/LICENSE`

`/usr/lib/modules/`uname -r`/weak-updates/evdi.ko.xz`
`/usr/lib/modules/`uname -r`/extra/evdi.ko.xz`

Make sure you have the latest kernel-* packages `dnf upgrade`.  Reboot.  Then install the RPM again.

Include any logs if it is still not working.

calebccff, 2020-12-23:
I had this error too, resolved it by updating to the latest 5.9 kernel, installing `kernel-headers` and then uninstalling / reinstalling the displaylink package.

FelipePepe, 2020-12-26:
Hello, I have fedora 33 and the drivers only found with kernel 5.8.200

elguero, 2020-12-26:
@FelipePepe Can you explain a bit more?  Not sure what you are asking.

Thanks

FelipePepe, 2020-12-30:
Hi Michael, I'm sorry, I read the comments that saying nor found the driver and just I want to say that me found with kernel updated by fedora 5.8.200

closed 2021-02-02 by elguero

Arthedian, 2021-02-07:
- Kernel version (uname -a)

> Linux localhost.localdomain 5.10.12-200.fc33.x86_64 #1 SMP Mon Feb 1 02:40:52 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

- /var/log/displaylink/displaylink.log

> Creating symlink /var/lib/dkms/evdi/1.7.2/source ->
>                  /usr/src/evdi-1.7.2
> 
> DKMS: add completed.
> 
> Kernel preparation unnecessary for this kernel.  Skipping...
> 
> Building module:
> cleaning build area....
> make -j4 KERNELRELEASE=5.9.11-200.fc33.x86_64 all INCLUDEDIR=/lib/modules/5.9.11-200.fc33.x86_64/build/include KVERSION=5.9.11-200.fc33.x86_64 DKMS_BUILD=1.....
> cleaning build area....
> 
> DKMS: build completed.
> 
> evdi.ko.xz:
> Running module version sanity check.
>  - Original module
>    - No original module exists within this kernel
>  - Installation
>    - Installing to /lib/modules/5.9.11-200.fc33.x86_64/extra/
> Adding any weak-modules
> 
> depmod.......
> 
> DKMS: install completed.
> 
> -------- Uninstall Beginning --------
> Module:  evdi
> Version: 1.7.2
> Kernel:  5.9.11-200.fc33.x86_64 (x86_64)
> -------------------------------------
> 
> Status: Before uninstall, this module version was ACTIVE on this kernel.
> Removing any linked weak-modules
> 
> evdi.ko.xz:
>  - Uninstallation
>    - Deleting from: /lib/modules/5.9.11-200.fc33.x86_64/extra/
>  - Original module
>    - No original module was found for this module on this kernel.
>    - Use the dkms install command to reinstall any previous module version.
> 
> depmod......
> 
> DKMS: uninstall completed.
> 
> ------------------------------
> Deleting module version: 1.7.2
> completely from the DKMS tree.
> ------------------------------
> Done.
> 
> Creating symlink /var/lib/dkms/evdi/1.7.2/source ->
>                  /usr/src/evdi-1.7.2
> 
> DKMS: add completed.
> 
> Kernel preparation unnecessary for this kernel.  Skipping...
> 
> Building module:
> cleaning build area....
> make -j4 KERNELRELEASE=5.10.12-200.fc33.x86_64 all INCLUDEDIR=/lib/modules/5.10.12-200.fc33.x86_64/build/include KVERSION=5.10.12-200.fc33.x86_64 DKMS_BUILD=1.....
> cleaning build area....
> 
[... 40 lines cut ...]
>   gcc -Wp,-MMD,/var/lib/dkms/evdi/1.7.2/build/.evdi_gem.o.d -nostdinc -isystem /usr/lib/gcc/x86_64-redhat-linux/10/include -I./arch/x86/include -I./arch/x86/include/generated  -I./include -I./arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/kconfig.h -include ./include/linux/compiler_types.h -D__KERNEL__ -fmacro-prefix-map=./= -Wall -Wundef -Werror=strict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -fshort-wchar -fno-PIE -Werror=implicit-function-declaration -Werror=implicit-int -Werror=return-type -Wno-format-security -Wno-address-of-packed-member -std=gnu89 -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mtune=generic -mno-red-zone -mcmodel=kernel -Wno-sign-compare -fno-asynchronous-unwind-tables -mindirect-branch=thunk-extern -mindirect-branch-register -fno-jump-tables -fno-delete-null-pointer-checks -Wno-frame-address -Wno-format-truncation -Wno-format-overflow -Wno-address-of-packed-member -O2 -fno-allow-store-data-races -Wframe-larger-than=2048 -fstack-protector -Wno-unused-but-set-variable -Wimplicit-fallthrough -Wno-unused-const-variable -g -pg -mrecord-mcount -mfentry -DCC_USING_FENTRY -Wdeclaration-after-statement -Wvla -Wno-pointer-sign -Wno-stringop-truncation -Wno-zero-length-bounds -Wno-array-bounds -Wno-stringop-overflow -Wno-restrict -Wno-maybe-uninitialized -fno-strict-overflow -fno-stack-check -fconserve-stack -Werror=date-time -Werror=incompatible-pointer-types -Werror=designated-init -fcf-protection=none -Wno-packed-not-aligned -Iinclude/drm  -DMODULE  -DKBUILD_BASENAME='"evdi_gem"' -DKBUILD_MODNAME='"evdi"' -c -o /var/lib/dkms/evdi/1.7.2/build/evdi_gem.o /var/lib/dkms/evdi/1.7.2/build/evdi_gem.c
>    ./tools/objtool/objtool orc generate  --module --no-fp --retpoline --uaccess /var/lib/dkms/evdi/1.7.2/build/evdi_modeset.o
>   gcc -Wp,-MMD,/var/lib/dkms/evdi/1.7.2/build/.evdi_painter.o.d -nostdinc -isystem /usr/lib/gcc/x86_64-redhat-linux/10/include -I./arch/x86/include -I./arch/x86/include/generated  -I./include -I./arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/kconfig.h -include ./include/linux/compiler_types.h -D__KERNEL__ -fmacro-prefix-map=./= -Wall -Wundef -Werror=strict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -fshort-wchar -fno-PIE -Werror=implicit-function-declaration -Werror=implicit-int -Werror=return-type -Wno-format-security -Wno-address-of-packed-member -std=gnu89 -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mtune=generic -mno-red-zone -mcmodel=kernel -Wno-sign-compare -fno-asynchronous-unwind-tables -mindirect-branch=thunk-extern -mindirect-branch-register -fno-jump-tables -fno-delete-null-pointer-checks -Wno-frame-address -Wno-format-truncation -Wno-format-overflow -Wno-address-of-packed-member -O2 -fno-allow-store-data-races -Wframe-larger-than=2048 -fstack-protector -Wno-unused-but-set-variable -Wimplicit-fallthrough -Wno-unused-const-variable -g -pg -mrecord-mcount -mfentry -DCC_USING_FENTRY -Wdeclaration-after-statement -Wvla -Wno-pointer-sign -Wno-stringop-truncation - [... cut ...] r-types -Werror=designated-init -fcf-protection=none -Wno-packed-not-aligned  -DMODULE  -DKBUILD_BASENAME='"evdi.mod"' -DKBUILD_MODNAME='"evdi"' -c -o /var/lib/dkms/evdi/1.7.2/build/evdi.mod.o /var/lib/dkms/evdi/1.7.2/build/evdi.mod.c
>   ld -r -m elf_x86_64 --build-id=sha1  -T scripts/module.lds -o /var/lib/dkms/evdi/1.7.2/build/evdi.ko /var/lib/dkms/evdi/1.7.2/build/evdi.o /var/lib/dkms/evdi/1.7.2/build/evdi.mod.o;  true
> make[1]: Leaving directory '/usr/src/kernels/5.10.12-200.fc33.x86_64'
>

derekmurawsky, 2021-06-29:
FYI - I followed these instructions and everything worked. I think I upgraded my system before installing this rpm, but didn't reboot so things were out of sync. I uninstalled the rpm, upgraded everything, rebooted, reinstalled the RPM, and everything worked perfectly.
Thanks for this package!!!