displaylink-rpm/displaylink-rpm #158: D6000 on fedora 33 -> labeled 33 version does not work, the 32 does

baverhey, 2021-01-14:
[make.log](https://github.com/displaylink-rpm/displaylink-rpm/files/5816900/make.log)
[displaylink.log](https://github.com/displaylink-rpm/displaylink-rpm/files/5816902/displaylink.log)

**Please note:** The only working combination is *DisplayLink 5.3.1* and *evdi v1.7.2*
If you are trying to use evdi 1.8.0, please do not open an issue. It will not work.

  * Are you using the latest DisplayLink driver?
    
  * Are you using the latest EVDI modu
[make.log](https://github.com/displaylink-rpm/displaylink-rpm/files/5816890/make.log)
[displaylink.log](https://github.com/displaylink-rpm/displaylink-rpm/files/5816891/displaylink.log)

le?
    
  * If you are using a DisplayLink device, have you checked 'troubleshooting'
    on DisplayLink's website?
    
  * Is this issue related to evdi/kernel?
    
  * Is this issue related to DisplayLinkManager?
    
  * If you are having an issue upgrading, have you tried to do a clean install of the driver?
    

  * Fedora or CentOS
  Fedora 33 state of 14-12021
  * Version
  * Kernel version `(uname -a)`
  Linux  5.10.6-200.fc33.x86_64 #1 SMP Mon Jan 11 20:05:21 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

  * `/var/log/displaylink/displaylink.log`
  * `/var/lib/dkms/evdi/kernel-[version]/log/make.log`
[make.log](https://github.com/displaylink-rpm/displaylink-rpm/files/5816905/make.log)
[displaylink.log](https://github.com/displaylink-rpm/displaylink-rpm/files/5816906/displaylink.log)

install fedora 33
install release vfedora-33-displaylink-1.7.2-1.x86_64.rpm --> does not work

install previus release for fedora 32 fedora-32-displaylink-1.7.2-1.x86_64.rpm --> does work

elguero, 2021-01-14:
@baverhey  Did you upgrade from Fedora 32 to Fedora 33?  Did you try cleaning prior DisplayLink installs first?

I see from logs that this system is not a clean install of Fedora 33.  I would try cleaning out any previous versions you may have installed (1.6.2 and the not supported 1.9.0 evdi drivers).

```
Do the following for a clean driver install:
           Find the current package installed - `sudo rpm -qa | grep displaylink`
           Remove what is returned: `sudo rpm -evh [package]`

           Clear out anything left over:
            /var/lib/dkms/evdi
            /usr/libexec/displaylink
            /usr/src/evdi-*

            /etc/X11/xorg.conf.d/20-displaylink.conf
            /etc/logrotate.d/displaylink
            /usr/lib/systemd/system/displaylink.service
            /usr/lib/systemd/system-preset/95-displaylink.preset
            /usr/lib/systemd/system-sleep/displaylink.sh
            /etc/udev/rules.d/99-displaylink.rules
            /usr/share/doc/displaylink
            /usr/share/doc/displaylink/LICENSE

            /usr/lib/modules/`uname -r`/weak-updates/evdi.ko.xz
            /usr/lib/modules/`uname -r`/extra/evdi.ko.xz

          Reboot
          Reinstall the RPM package 
```

baverhey, 2021-01-16:
Thank you for the nice instructions.
My previous build was not stable.

I executed your instructions and now everything seems to work.

Especially  /var/lib/dkms/evdi  contained a lot of old stuff.

Keep you posted

baverhey, 2021-01-16:
An update the screens flicker off and on, (2 times in the last 5 minutes)
i'm on a dell Latitude 5590

baverhey, 2021-01-16:
Now the 2 external screens are frozen and focus and command is back to the laptop screen.
so worked for 20'
only opens a few 3 chrome windows,no other load, i have 32GB Ram so that is probably not it.
[make.log](https://github.com/displaylink-rpm/displaylink-rpm/files/5824354/make.log)
[displaylink.log](https://github.com/displaylink-rpm/displaylink-rpm/files/5824355/displaylink.log)

elguero, 2021-01-16:
@baverhey If you haven't tried yet, look at the Readme in this repository. It talks about what to do with D6000 and it sort of sounds like what is happening to you.

You can give that a try.  It sounds like the driver has been installed properly now.

baverhey, 2021-01-18:
Seems to do the trick.
I read the readme but it did not click at the time.

thx for your assistance!

closed 2021-01-18 by baverhey