displaylink-rpm/displaylink-rpm #161: Building EVDI fails

qoheniac, 2021-03-22:
Hi, I'm on Fedora 33 with kernel 5.11.7 and installing the RPM package for DisplayLink 1.7.2 fails during the building procedure for EVDI 1.7.2. The log files seem to suggest I should go to `/usr/src/kernels/5.11.7-200.fc33.x86_64` and run `make oldconfig && make prepare`, but as I have no idea what those commands do, I opened this issue looking for help. Thank you in advance!

```
$ cat /var/log/displaylink/displaylink.log
DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
make -j4 KERNELRELEASE=5.11.7-200.fc33.x86_64 all INCLUDEDIR=/lib/modules/5.11.7-200.fc33.x86_64/build/include KVERSION=5.11.7-200.fc33.x86_64 DKMS_BUILD=1....(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.11.7-200.fc33.x86_64 (x86_64)
Consult /var/lib/dkms/evdi/1.7.2/build/make.log for more information.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
make -j4 KERNELRELEASE=5.11.7-200.fc33.x86_64 all INCLUDEDIR=/lib/modules/5.11.7-200.fc33.x86_64/build/include KVERSION=5.11.7-200.fc33.x86_64 DKMS_BUILD=1....(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.11.7-200.fc33.x86_64 (x86_64)
Consult /var/lib/dkms/evdi/1.7.2/build/make.log for more information.

------------------------------
Deleting module version: 1.7.2
completely from the DKMS tree.
------------------------------
Done.

Creating symlink /var/lib/dkms/evdi/1.7.2/source ->
                 /usr/src/evdi-1.7.2

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
make -j4 KERNELRELEASE=5.11.7-200.fc33.x86_64 all INCLUDEDIR=/lib/modules/5.11.7-200.fc33.x86_64/build/include KVERSION=5.11.7-200.fc33.x86_64 DKMS_BUILD=1....(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.11.7-200.fc33.x86_64 (x86_64)
Consult /var/lib/dkms/evdi/1.7.2/build/make.log for more information.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
make -j4 KERNELRELEASE=5.11.7-200.fc33.x86_64 all INCLUDEDIR=/lib/modules/5.11.7-200.fc33.x86_64/build/include KVERSION=5.11.7-200.fc33.x86_64 DKMS_BUILD=1....(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.11.7-200.fc33.x86_64 (x86_64)
Consult /var/lib/dkms/evdi/1.7.2/build/make.log for more information.
```

```
$ cat /var/lib/dkms/evdi/1.7.2/build/make.log
DKMS make.log for evdi-1.7.2 for kernel 5.11.7-200.fc33.x86_64 (x86_64)
Mo 22. Mär 22:29:46 CET 2021
make KBUILD_VERBOSE=1 M=/var/lib/dkms/evdi/1.7.2/build SUBDIRS=/var/lib/dkms/evdi/1.7.2/build SRCROOT=/var/lib/dkms/evdi/1.7.2/build CONFIG_MODULE_SIG= -C /lib/modules/5.11.7-200.fc33.x86_64/build modules
make[1]: Entering directory '/usr/src/kernels/5.11.7-200.fc33.x86_64'
test -e include/generated/autoconf.h -a -e include/config/auto.conf || (		\
echo >&2;							\
echo >&2 "  ERROR: Kernel configuration is invalid.";		\
echo >&2 "         include/generated/autoconf.h or include/config/auto.conf are missing.";\
echo >&2 "         Run 'make oldconfig && make prepare' on kernel src to fix it.";	\
echo >&2 ;							\
/bin/false)
make -f ./scripts/Makefile.build obj=/var/lib/dkms/evdi/1.7.2/build \
single-build= \
need-builtin=1 need-modorder=1
  gcc -Wp,-MMD,/var/lib/dkms/evdi/1.7.2/build/.evdi_drv.o.d -nostdinc -isystem /usr/lib/gcc/x86_64-redhat-linux/10/include -I./arch/x86/include -I./arch/x86/include/generated  -I./include -I./arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/kconfig.h -include ./include/linux/compiler_types.h -D__KERNEL__ -fmacro-prefix-map=./= -Wall -Wundef -Werror=strict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -fshort-wchar -fno-PIE -Werror=implicit-function-declaration -Werror=implicit-int -Wno-address-of-packed-member -Werror=return-type -Wno-format-security -std=gnu89 -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -fcf-protection=none -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mtune=generic -mno-red-zone -mcmodel=kernel -Wno-sign-compare -fno-asynchronous-unwind-tables -mindirect-branch=thunk-extern -mindirect-branch-register -fno-jump-tables -fno-delete-null-pointer-checks -Wno-frame-address -Wno-format-truncation -Wno-format-overflow -Wno-address-of-packed-member -O2 -fno-allow-store-data-races -Wframe-larger-than=2048 -fstack-protector -Wno-unused-but-set-variable -Wimplicit-fallthrough -Wno-unused-const-variable -g -pg -mrecord-mcount -mfentry -DCC_USING_FENTRY -Wdeclaration-after-statement -Wvla -Wno-pointer-sign -Wno-stringop-truncation -Wno-zero-length-bounds -Wno-array-bounds -Wno-stringop-overflow -Wno-restrict -Wno-maybe-uninitialized -fno-strict-overflow -fno-stack-check -fconserve-stack -Werror=date-time -Werror=incompatible-pointer-types -Werror=designated-init -Wno-packed-not-aligned -Iinclude/drm  -DMODULE  -DKBUILD_BASENAME='"evdi_drv"' -DKBUILD_MODNAME='"evdi"' -c -o /var/lib/dkms/evdi/1.7.2/build/evdi_drv.o /var/lib/dkms/evdi/1.7.2/build/evdi_drv.c
  gcc -Wp,-MMD,/var/lib/dkms/evdi/1.7.2/build/.evdi_modeset.o.d -nostdinc -isystem /usr/lib/gcc/x86_64-redhat-linux/10/include -I./arch/x86/include -I./arch/x86/include/generated  -I./include -I./arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/kconfig.h -include ./include/linux/compiler_types.h -D__KERNEL__ -fmacro-prefix-map=./= -Wall -Wundef -Werror=strict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -fshort-wchar -fno-PIE -Werror=implicit-function-declaration -Werror=implicit-int -Wno-address-of-packed-member -Werror=return-type -Wno-format-security -std=gnu89 -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -fcf-protection=none -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mtune=generic -mno-red-zone -mcmodel=kernel -Wno-sign-compare -fno-asynchronous-unwind-tables -mindirect-branch=thunk-extern -mindirect-branch-register -fno-jump-tables -fno-delete-null-pointer-checks -Wno-frame-address -Wno-format-truncation -Wno-format-overflow -Wno-address-of-packed-member -O2 -fno-allow-store-data-races -Wframe-larger-than=2048 -fstack-protector -Wno-unused-but-set-variable -Wimplicit-fallthrough -Wno-unused-const-variable -g -pg -mrecord-mcount -mfentry -DCC_USING_FENTRY -Wdeclaration-after-statement -Wvla -Wno-pointer-sign -Wno-stringop-truncation -Wno-zero-length-bounds -Wno-array-bounds -Wno-stringop-overflow -Wno-restrict -Wno-maybe-uninitialized -fno-strict-overflow -fno-stack-check -fconserve-stack -Werror=date-time -Werror=incompatible-pointer-types -Werror=designated-init -Wno-packed-not-aligned -Iinclude/drm  -DMODULE  -DKBUILD_BASENAME='"evdi_modeset"' -DKBUILD_MODNAME='"evdi"' -c -o /var/lib/dkms/evdi/1.7.2/build/evdi_modeset.o /var/lib/dkms/evdi/1.7.2/build/evdi_modeset.c
  gcc -Wp,-MMD,/var/lib/dkms/evdi/1.7.2/build/.evdi_connector.o.d -nostdinc -isystem /usr/lib/gcc/x86_64-redhat-linux/10/include -I./arch/x86/include -I./arch/x86/include/generated  -I./include -I./arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/kconfig.h -include ./include/linux/compiler_types.h -D__KERNEL__ -fmacro-prefix-map=./= -Wall -Wundef -Werror=strict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -fshort-wchar -fno-PIE -Werror=implicit-function-declaration -Werror=implicit-int -Wno-address-of-packed-member -Werror=return-type -Wno-format-security -std=gnu89 -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -fcf-protection=none -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mtune=generic -mno-red-zone -mcmodel=kernel -Wno-sign-compare -fno-asynchronous-unwind-tables -mindirect-branch=thunk-extern -mindirect-branch-register -fno-jump-tables -fno-delete-null-pointer-checks -Wno-frame-address -Wno-format-truncation -Wno-format-overflow -Wno-address-of-packed-member -O2 -fno-allow-store-data-races -Wframe-larger-than=2048 -fstack-protector -Wno-unused-but-set-variable -Wimplicit-fallthrough -Wno-unused-const-variable -g -pg -mrecord-mcount -mfentry -DCC_USING_FENTRY -Wdeclaration-after-statement -Wvla -Wno-pointer-sign -Wno-stringop-truncation -W [... cut ...]
   ./tools/objtool/objtool orc generate  --module --no-fp --retpoline --uaccess /var/lib/dkms/evdi/1.7.2/build/evdi_encoder.o
   ./tools/objtool/objtool orc generate  --module --no-fp --retpoline --uaccess /var/lib/dkms/evdi/1.7.2/build/evdi_connector.o
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:279: /var/lib/dkms/evdi/1.7.2/build/evdi_modeset.o] Error 1
make[1]: *** [Makefile:1807: /var/lib/dkms/evdi/1.7.2/build] Error 2
make[1]: Leaving directory '/usr/src/kernels/5.11.7-200.fc33.x86_64'
make: *** [Makefile:28: all] Error 2
```

elguero, 2021-03-22:
Please see this comment from DisplayLink: https://github.com/DisplayLink/evdi/pull/256#issuecomment-787848715

There currently is no official support for kernel 5.11

qoheniac, 2021-03-22:
Thank you, for your quick reply. I'll subscribe to that issue and wait.

Chosco, 2021-04-05:
I have exactly the same problem: After upgrading to fedora 33, the monitor on my docking station no longer works because my fedora has a 5.11.11 kernel.

Thank you for now for the great software.
I am sorry that I cannot help to implement kernel version 5.11.

elguero, 2021-04-06:
Here is an update from Displaylink on the issue with kernel 5.11+.

https://github.com/DisplayLink/evdi/issues/275

bnavigator, 2021-04-06:
FYI: DisplayLink 2.4.0 has been released, which also bundles the second to last commit of evdi on the devel branch (1.9.1 plus 4 commits). It works with Kernel 5.11.

jteib, 2021-04-08:
> FYI: DisplayLink 2.4.0 has been released, which also bundles the second to last commit of evdi on the devel branch (1.9.1 plus 4 commits). It works with Kernel 5.11.

Shouldn't we see a new release then?  Looks like the latest release is still v5.3.1-1

elguero, 2021-04-08:
@jteib Yes, it is being worked on.

jteib, 2021-04-08:
Got it.  Didn't mean to sound pushy or impatient, just curious.  Thanks for the great work!!

gabops, 2021-04-15:
Hello there,

My dockstation with two displays was not working after moving to Fedora 33. I tried to use the latest rpm package for it that this repo provides with no luck. Fortunately, I've been able to get everything working on Fedora 33 with kernel "**5.11.13-200**" (the latest installed today after a dnf update) by installing the driver from the official package (5.4 version that supports kernels 5.11 and 5.12). I did the following:

1. Install dkms
```bash
$ dnf install dkms
```
2. Download the driver at https://www.synaptics.com/products/displaylink-graphics/downloads/ubuntu
3. Extract the content of the file by doing:
```bash
$ chmod u+x displaylink-driver-5.4.0-55.153.run
$ ./displaylink-driver-5.4.0-55.153.run --noexec --keep
```
4. Move into the directory created as a result of the extraction:
```bash
$ cd displaylink-driver-5.4.0-55.153
```
5. Now open the script file called "displaylink-installer.sh" and go to the function defined on line 529 called "chedk_libdrm" and do the following:
```bash
check_libdrm()
{
  #hash apt 2>/dev/null || return
  #apt list -qq --installed libdrm-dev 2>/dev/null | grep -q libdrm-dev
  return 0
}
```
6. Finally, run the script **as root** with:
```bash
$ sudo ./displaylink-installer.sh
```

Everything is working flawlessly and I am quite happy with the performance. Ironically, despite the official driver supports only Ubuntu 20.04, the driver works far much better on Fedora in terms of FPS and cpu consumed by the displaylink process itself. 

Hope you find this helpful.

Regards.

elguero, 2021-04-15:
Thank you for sharing.

This is what we are waiting on from DisplayLink to see if it saves us time from having to change the current process we have in place.

https://github.com/DisplayLink/evdi/issues/275#issuecomment-815955502

jteib, 2021-04-16:
It seems displaylink-installer.sh has a few more functions aside from check_libdrm() that are Ubuntu specific, and look for the services init directories.  I was not able to get it to install on Fedora 33 by simply editing the displaylink-installer.sh as shown above.

elguero, 2021-04-16:
Correct.  That is why this project is specific to Fedora and does not use the installer at all.

brego81, 2021-04-16:
Thanks @gabops for the solution, it works for me despite I see an error in the installation process:

```
$  sudo ./displaylink-installer.sh
DisplayLink Linux Software 5.4.0-55.153 install script called: 
Distribution discovered: "Fedora release 33 (Thirty Three)"

Installing

[ Installing EVDI ]
[[ Installing EVDI DKMS module ]]

Creating symlink /var/lib/dkms/evdi/1.9.1/source ->
                 /usr/src/evdi-1.9.1

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j8 KERNELRELEASE=5.11.13-200.fc33.x86_64 all INCLUDEDIR=/lib/modules/5.11.13-200.fc33.x86_64/build/include KVERSION=5.11.13-200.fc33.x86_64 DKMS_BUILD=1.....
cleaning build area...

DKMS: build completed.

evdi.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.11.13-200.fc33.x86_64/extra/
Adding any weak-modules

depmod.....

DKMS: install completed.
[[ Installing module configuration files ]]
[[ Installing EVDI library ]]
cc -I../module -std=gnu99 -fPIC -D_FILE_OFFSET_BITS=64    -c -o evdi_lib.o evdi_lib.c
evdi_lib.c:6:10: fatal error: libdrm/drm.h: No such file or directory
    6 | #include <libdrm/drm.h>
      |          ^~~~~~~~~~~~~~
compilation terminated.
make: *** [<builtin>: evdi_lib.o] Error 1
ERROR: Failed to build evdi library.
```

jteib, 2021-04-17:
How are you getting this Ubuntu script to work with Fedora by only following @gabops steps above?

gabops, 2021-04-17:
Hello,

I've tried again on a virtual machine over a fresh Fedora 33 installation and I was able to install evdi again. The only difference is that I had to install "libdrm-devel" package. I am pretty sure I overlooked the entry in my bash history when writing the instructions for my original message. I edited it. Sorry for that.

jteib, 2021-04-17:
@gabops no worries, I figured out my issue.  In my case, I am running a Fedora spin with xfce desktop.  The lsb_release that the detect_distro() function uses was not present.  After installing redhat-lsb-core, the script executed and finished successfully!  

After that I created a new MOK key, signed for this kernel and my extra monitor is back!

Thanks for your help!

elguero, 2021-04-22:
Since we are not getting a response back from DisplayLInk as to whether they plan to make an official release of the EVDI library and kernel module that works with kernels 5.11 / 5.12, I have come up with an approach to solve this for now.

I submitted a PR a couple of days ago to get feedback (https://github.com/displaylink-rpm/displaylink-rpm/pull/163).  My thoughts are to use the EVDI tarball that is included in the DisplayLink 5.4 drivers package.  This way we can release new RPM builds that work with kernels 5.11 / 5.12.  I have also adjusted the build to allow for us to pull from the EVDI repo here on GitHub by using `make unbundled`, which can be used more for testing and development purposes.

What is everyone's thoughts on this approach?

Should we stick to the way we are doing things and hope that DisplayLink will eventually tag a proper working release on their repo?

Or, should we just get a release out that we know is working so that everyone can get their work setups back?

Thank you

davidschenz, 2021-04-23:
Howdy - I built the branch this afternoon and it's worked fine so far on Fedora 33 with kernel 5.11.15-200. It's silly that DisplayLink won't tag the release in the branch, but until they do that, using the tarball from the drivers package makes the most sense.

Also - thank you for maintaining this project. It certainly makes my life better being able to use the extra DisplayLink screen with Fedora.

Phaestion, 2021-04-23:
Hi everyone! Yeah, I agree with the approach of using the tarball to get a release out. I look forward to the release!

closed 2021-04-23 by elguero