displaylink-rpm/displaylink-rpm #164: Travis builds fail for PRs

ffgiff, 2021-04-21:
Hi,

The Travis job for my PR #97 failed here: https://travis-ci.org/github/displaylink-rpm/displaylink-rpm/builds/767919525 due to non-existent docker credentials. The commit from the PR is fine in Travis when run using my own credentials: https://travis-ci.com/github/ffgiff/displaylink-rpm/builds/223800598.

Thanks!

elguero, 2021-04-22:
Thanks for reporting. It looks like Travis doesn't provide the environmental variables for security reasons when pull requests are external.

I have attempted to resolve this with the following commit to the `.travis` file.
https://github.com/displaylink-rpm/displaylink-rpm/commit/d7bad976ae5c454db0add721e06385e64d2314ef

ffgiff, 2021-04-22:
> Thanks for reporting. It looks like Travis doesn't provide the environmental variables for security reasons when pull requests are external.
> 
Yes, I can see what would happen if you allowed just anybody to edit .travis.yml in a PR.

> I have attempted to resolve this with the following commit to the `.travis` file.
> [d7bad97](https://github.com/displaylink-rpm/displaylink-rpm/commit/d7bad976ae5c454db0add721e06385e64d2314ef)

Thanks :+1:

closed 2021-04-23 by elguero