displaylink-rpm/displaylink-rpm #165: modprobe: ERROR: could not insert 'evdi': Key was rejected by service on Fedora 33 Kernel 5.11.15-200.fc33.x86_64

sebastiano1972, 2021-04-23:
Hi,

Thank you for your hard work. 

I just downloaded and installed the latest rpm but after installing it and rebooting the laptop with the docking station attached, I am getting the following error and the attached external monitors remain blank.

Maybe I've done something wrong but I thought it would be useful to report the problem.

Many thanks.

Sebastiano

My system is:

DisplayLink 5.4* and *evdi v1.9.1* on Fedora 33 Kernel 5.11.15-200.fc33.x86_64 on Lenovo L14 Ryzen pro 4750U attached to Lenovo docking station USB C type 40AF with Display Link Chipset.

The jurnalctl output is:

Apr 23 19:22:48 Obsidian systemd[1]: Stopped DisplayLink Manager Service.
Apr 23 19:22:48 Obsidian audit[1]: SERVICE_STOP pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=>
Apr 23 19:22:48 Obsidian audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit>
Apr 23 19:22:48 Obsidian systemd[1]: displaylink.service: Scheduled restart job, restart counter is at 5.
Apr 23 19:22:46 Obsidian audit[1]: SERVICE_STOP pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=>
Apr 23 19:22:46 Obsidian systemd[1]: NetworkManager-dispatcher.service: Succeeded.
Apr 23 19:22:43 Obsidian systemd[1]: Failed to start DisplayLink Manager Service.
Apr 23 19:22:43 Obsidian systemd[1]: displaylink.service: Failed with result 'exit-code'.
Apr 23 19:22:43 Obsidian systemd[1]: displaylink.service: Control process exited, code=exited, status=1/FAILURE
Apr 23 19:22:43 Obsidian audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit>
Apr 23 19:22:43 Obsidian kernel: Loading of unsigned module is rejected
Apr 23 19:22:43 Obsidian modprobe[2566]: modprobe: ERROR: could not insert 'evdi': Key was rejected by service
Apr 23 19:22:43 Obsidian systemd[1]: Starting DisplayLink Manager Service...

[displaylink.log](https://github.com/displaylink-rpm/displaylink-rpm/files/6367479/displaylink.log)

elguero, 2021-04-23:
Do you have secure boot enabled?

If you do, check out the README: https://github.com/displaylink-rpm/displaylink-rpm#secure-boot-on-fedora

sebastiano1972, 2021-04-23:
My bad, sorry. I just disabled the secure boot but still no joy.
If I connect and disconnect the docking station only the audio and the nic seems to be recognised correctly.

Below the journalctl when I connect the docking station to the laptop.

Apr 23 20:09:13 Obsidian mtp-probe[2652]: bus: 4, device: 13 was not an MTP device
Apr 23 20:09:13 Obsidian mtp-probe[2652]: checking bus 4, device 13: "/sys/devices/pci0000:00/0000:00:08.1/0000:06:00.4/usb4>
Apr 23 20:09:13 Obsidian mtp-probe[2649]: bus: 4, device: 13 was not an MTP device
Apr 23 20:09:13 Obsidian mtp-probe[2649]: checking bus 4, device 13: "/sys/devices/pci0000:00/0000:00:08.1/0000:06:00.4/usb4>
Apr 23 20:09:13 Obsidian kernel: usb 4-1.4.2.1: Manufacturer: Cypress Semiconductor
Apr 23 20:09:13 Obsidian kernel: usb 4-1.4.2.1: Product: USB-Serial Bridge
Apr 23 20:09:13 Obsidian kernel: usb 4-1.4.2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Apr 23 20:09:13 Obsidian kernel: usb 4-1.4.2.1: New USB device found, idVendor=04b4, idProduct=5218, bcdDevice= 0.00
Apr 23 20:09:12 Obsidian kernel: usb 4-1.4.2.1: new full-speed USB device number 13 using xhci_hcd
Apr 23 20:09:12 Obsidian plasmashell[1797]: org.kde.plasma.pulseaudio: No object for name "alsa_input.usb-DisplayLink_ThinkP>
Apr 23 20:09:12 Obsidian rtkit-daemon[987]: Supervising 5 threads of 1 processes of 1 users.
Apr 23 20:09:12 Obsidian rtkit-daemon[987]: Successfully made thread 2644 of process 1835 (/usr/bin/pulseaudio) owned by '10>
Apr 23 20:09:12 Obsidian upowerd[999]: treating change event as add on /sys/devices/pci0000:00/0000:00:08.1/0000:06:00.4/usb>
Apr 23 20:09:12 Obsidian rtkit-daemon[987]: Supervising 4 threads of 1 processes of 1 users.
Apr 23 20:09:12 Obsidian kernel: hub 4-1.4.2:1.0: 4 ports detected
Apr 23 20:09:12 Obsidian kernel: hub 4-1.4.2:1.0: USB hub found
Apr 23 20:09:12 Obsidian rtkit-daemon[987]: Supervising 4 threads of 1 processes of 1 users.
Apr 23 20:09:12 Obsidian rtkit-daemon[987]: Successfully made thread 2642 of process 1835 (/usr/bin/pulseaudio) owned by '10>
Apr 23 20:09:12 Obsidian kernel: usb 4-1.4.2: Product: USB 2.0 Hub [MTT]
Apr 23 20:09:12 Obsidian kernel: usb 4-1.4.2: New USB device strings: Mfr=0, Product=1, SerialNumber=0
Apr 23 20:09:12 Obsidian kernel: usb 4-1.4.2: New USB device found, idVendor=1a40, idProduct=0101, bcdDevice= 1.00
Apr 23 20:09:12 Obsidian rtkit-daemon[987]: Supervising 3 threads of 1 processes of 1 users.
Apr 23 20:09:12 Obsidian kernel: usb 4-1.4.2: new high-speed USB device number 12 using xhci_hcd
Apr 23 20:09:11 Obsidian upowerd[999]: treating change event as add on /sys/devices/pci0000:00/0000:00:08.1/0000:06:00.4/usb>
Apr 23 20:09:11 Obsidian upowerd[999]: treating change event as add on /sys/devices/pci0000:00/0000:00:08.1/0000:06:00.4/usb>
Apr 23 20:09:11 Obsidian plasmashell[1797]: UdevQt: unhandled device action "move"
Apr 23 20:09:11 Obsidian kded5[1709]: plasma-nm: Unhandled active connection state change:  1
Apr 23 20:09:11 Obsidian kded5[1709]: plasma-nm: Unhandled active connection state change:  1
Apr 23 20:09:11 Obsidian avahi-daemon[964]: Registering new address record for fe80::fba6:132b:2ac7:81a2 on enp6s0f4u1u3c2.*.
Apr 23 20:09:11 Obsidian avahi-daemon[964]: New relevant interface enp6s0f4u1u3c2.IPv6 for mDNS.
Apr 23 20:09:11 Obsidian avahi-daemon[964]: Joining mDNS multicast group on interface enp6s0f4u1u3c2.IPv6 with address fe80:>
Apr 23 20:09:11 Obsidian NetworkManager[1110]: <info>  [1619204951.8625] dhcp4 (enp6s0f4u1u3c2): activation: beginning trans>
Apr 23 20:09:11 Obsidian NetworkManager[1110]: <info>  [1619204951.8621] device (enp6s0f4u1u3c2): state change: config -> ip>
Apr 23 20:09:11 Obsidian audit[968]: NETFILTER_CFG table=?:0;?:0 family=0 entries=11 op=nft_register_gen pid=968 subj=system>
Apr 23 20:09:11 Obsidian audit[968]: NETFILTER_CFG table=firewalld:3;filter_FORWARD_OUT_ZONES:102 family=1 entries=416 op=nf>
Apr 23 20:09:11 Obsidian audit[968]: NETFILTER_CFG table=firewalld:3;filter_FORWARD_IN_ZONES:100 family=1 entries=415 op=nft>
Apr 23 20:09:11 Obsidian audit[968]: NETFILTER_CFG table=firewalld:3;filter_INPUT_ZONES:91 family=1 entries=414 op=nft_regis>
Apr 23 20:09:11 Obsidian audit[968]: NETFILTER_CFG table=firewalld:5;nat_POSTROUTING_ZONES:29 family=10 entries=108 op=nft_r>
Apr 23 20:09:11 Obsidian audit[968]: NETFILTER_CFG table=firewalld:4;nat_POSTROUTING_ZONES:29 family=2 entries=108 op=nft_re>
Apr 23 20:09:11 Obsidian audit[968]: NETFILTER_CFG table=firewalld:5;nat_PREROUTING_ZONES:26 family=10 entries=107 op=nft_re>
Apr 23 20:09:11 Obsidian audit[968]: NETFILTER_CFG table=firewalld:4;nat_PREROUTING_ZONES:26 family=2 entries=107 op=nft_reg>
Apr 23 20:09:11 Obsidian audit[968]: NETFILTER_CFG table=firewalld:3;mangle_PREROUTING_ZONES:82 family=1 entries=413 op=nft_>
Apr 23 20:09:11 Obsidian audit[968]: NETFILTER_CFG table=firewalld:3;raw_PREROUTING_ZONES:79 family=1 entries=412 op=nft_reg>
Apr 23 20:09:11 Obsidian baloo_file[1804]: UdevQt: unhandled device action "move"
Apr 23 20:09:11 Obsidian org_kde_powerdevil[1866]: UdevQt: unhandled device action "move"
Apr 23 20:09:11 Obsidian upowerd[999]: treating change event as add on /sys/devices/pci0000:00/0000:00:08.1/0000:06:00.4/usb>
Apr 23 20:09:11 Obsidian NetworkManager[1110]: <info>  [1619204951.8470] device (enp6s0f4u1u3c2): state change: prepare -> c>
Apr 23 20:09:11 Obsidian NetworkManager[1110]: <info>  [1619204951.8466] device (enp6s0f4u1u3c2): state change: disconnected>
[... cut ...]
Apr 23 20:09:09 Obsidian kernel: hub 4-1:1.0: USB hub found
Apr 23 20:09:09 Obsidian kernel: usb 4-1: Manufacturer: VIA Labs, Inc.         
Apr 23 20:09:09 Obsidian kernel: usb 4-1: Product: USB2.0 Hub             
Apr 23 20:09:09 Obsidian kernel: usb 4-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Apr 23 20:09:09 Obsidian kernel: usb 4-1: New USB device found, idVendor=17ef, idProduct=1028, bcdDevice=55.43
Apr 23 20:09:09 Obsidian kernel: usb 4-1: new high-speed USB device number 9 using xhci_hcd

sebastiano1972, 2021-04-23:
I re-enabled the secure-boot. I will try with the suggested procedure.

sebastiano1972, 2021-04-25:
Ok, I removed the rpm and checked all the folders as suggested when I opened the issue. I re-enabled the secure boot, and I followed the instructions to sign the module. The good news is that then the external monitors now both work. The bad news are:

1) I can only use one external monitor (4K@60) and the laptop monitor or both the external monitors only (4K@60) but not the laptop monitor. maybe this is a hardware limit though.

2) when hooked to the docking startion the laptop becomes extremely laggy to the point it is almost un-usable (I cannot ytpe more than 1 char per 2 secs on the teminal, mouse movements are choppy and cannot wven browse the internet as it takes forever to open a page)

3) I can confirm th power usage of the laptop increases when it is attached to the docking station as per the attached
 powertop report. Disconnecting the laptop from the docking station solves the power consumption issue.

[Uploading report.txt…]()

elguero, 2021-04-26:
@sebastiano1972 Perhaps you are having the same issue that was reported to DisplayLink on their repo: https://github.com/DisplayLink/evdi/issues/271

sebastiano1972, 2021-04-26:
@elguero It seems a very similar issue indeed. Although only Wayland and X11 are mentioned and I use Kde instead (it should be Qt based?). But the symptoms are the same.

elguero, 2021-04-26:
@sebastiano1972 I thought that I saw someone reporting issues with KDE towards the bottom in one of the comments.

husmanahmed, 2021-04-28:
Not sure if this helps with the diagnostics for this issue. I have just installed: fedora-33-displaylink-1.9.1-1.x86_64.rpm
on my Thinkpad T460s, running Fedora 33. 

> uname -a
Linux thinkpad 5.11.14-200.fc33.x86_64

The monitor connected to the HDMI port of my dock works (even before installing the displaylink driver)
The monitor connected to the DVI port of the dock is not recognised.

I tried installing the RPM with the monitor connected. I tried a reboot (or two).
Is there anything else I can provide to help diagnose the issue?

elguero, 2021-05-07:
FYI: https://github.com/DisplayLink/evdi/pull/282

husmanahmed, 2021-05-10:
> FYI: [DisplayLink/evdi#282](https://github.com/DisplayLink/evdi/pull/282)

Thank you for replying. I'm not quite sure if this relates to me, I am on an intel i5 processor.

derekmurawsky, 2021-06-29:
FYI - I followed these instructions and everything worked. I think I upgraded my system before installing this rpm, but didn't reboot so things were out of sync. I uninstalled the rpm, upgraded everything, rebooted, reinstalled the RPM, and everything worked perfectly. 
Thanks for this package!!!

elguero, 2022-07-22:
A newer RPM has been released.  Can you provide some feedback on whether this solves your issue?  Thanks.

elguero, 2022-08-12:
Closing for now.

closed 2022-08-12 by elguero