displaylink-rpm/displaylink-rpm #167: Not be able to remove old version of displaz

Arthedian, 2021-05-01:
Hello.
I am using Fedora 34 (i updated today) with kernel 5.11.16-300.fc34.x86_64. I was having problem to get it running in the past, but I wanted to get it running again. I am trying to remove the old version before installing the new one. Sadly it throw me an error. What should I do?
```
sudo rpm -qa | grep displaylink
displaylink-1.7.2-1.x86_64
[]$ sudo rpm -evh displaylink-1.7.2-1.x86_64
Preparing...                          ################################# [100%]
Error! The module/version combo: evdi-1.7.2
is not located in the DKMS tree.
error: %preun(displaylink-1.7.2-1.x86_64) scriptlet failed, exit status 3
error: displaylink-1.7.2-1.x86_64: erase failed
```

elguero, 2021-05-03:
When you created the issue, there were instructions on how to do a clean driver install.  Here it is for you to try:

           Find the current package installed - `sudo rpm -qa | grep displaylink`
           Remove what is returned: `sudo rpm -evh [package]`

           Clear out anything left over:
            /var/lib/dkms/evdi
            /usr/libexec/displaylink
            /usr/src/evdi-*

            /etc/X11/xorg.conf.d/20-displaylink.conf
            /etc/logrotate.d/displaylink
            /usr/lib/systemd/system/displaylink.service
            /usr/lib/systemd/system-preset/95-displaylink.preset
            /usr/lib/systemd/system-sleep/displaylink.sh
            /etc/udev/rules.d/99-displaylink.rules
            /usr/share/doc/displaylink
            /usr/share/doc/displaylink/LICENSE

            /usr/lib/modules/`uname -r`/weak-updates/evdi.ko.xz
            /usr/lib/modules/`uname -r`/extra/evdi.ko.xz

          Reboot
          Reinstall the RPM package

elguero, 2021-05-03:
I just re-read your message...  try uninstalling without running the scripts: `sudo rpm -evh --noscripts displaylink-1.7.2-1.x86_64`

Then proceed with the instructions to manually clean things out that were left behind.

Arthedian, 2021-05-05:
Perfect, it worked. Thank you very much

closed 2021-05-05 by Arthedian