displaylink-rpm/displaylink-rpm #172: Fedora 34 - evdi 1.9.1 error when compiling

iranzo, 2021-07-24:
**Please note:** The only working combination is *DisplayLink 5.3.1* and *evdi v1.7.2* OR *DisplayLink 5.4* and *evdi v1.9.1*

  * Are you using the latest DisplayLink driver?
    
  * Are you using the latest EVDI module?
    
  * If you are using a DisplayLink device, have you checked 'troubleshooting'
    on DisplayLink's website?
    
  * Is this issue related to evdi/kernel?
    
  * Is this issue related to DisplayLinkManager?
    
  * If you are having an issue upgrading, have you tried to do a clean install of the driver?
    

Fedora 34 on 5.13.4-200.fc34.x86_64

  * `/var/log/displaylink/displaylink.log`
Creating symlink /var/lib/dkms/evdi/1.9.1/source ->
                 /usr/src/evdi-1.9.1

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
make -j8 KERNELRELEASE=5.12.13-300.fc34.x86_64 all INCLUDEDIR=/lib/modules/5.12.13-300.fc34.x86_64/build/include KVERSION=5.12.13-300.fc34.x86_64 DKMS_BUILD=1.....
cleaning build area....

DKMS: build completed.

evdi.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.12.13-300.fc34.x86_64/extra/

depmod......

DKMS: install completed.

  * `/var/lib/dkms/evdi/kernel-[version]/log/make.log`
https://paste.centos.org/view/raw/3219ebad

DKMS module doesn't compile:

make -C /lib/modules/5.13.4-200.fc34.x86_64/build M=$PWD
make[1]: Entering directory '/usr/src/kernels/5.13.4-200.fc34.x86_64'
  CC [M]  /usr/src/evdi-1.9.1/evdi_modeset.o
/usr/src/evdi-1.9.1/evdi_modeset.c:331:26: error: initialization of 'void (*)(struct drm_plane *, struct drm_atomic_state *)' from incompatible pointer type 'void (*)(struct drm_plane *, struct drm_plane_state *)' [-Werror=incompatible-pointer-types]
  331 |         .atomic_update = evdi_plane_atomic_update,
      |                          ^~~~~~~~~~~~~~~~~~~~~~~~
/usr/src/evdi-1.9.1/evdi_modeset.c:331:26: note: (near initialization for 'evdi_plane_helper_funcs.atomic_update')
/usr/src/evdi-1.9.1/evdi_modeset.c:332:23: error: 'drm_gem_fb_prepare_fb' undeclared here (not in a function); did you mean 'drm_gem_fb_create'?
  332 |         .prepare_fb = drm_gem_fb_prepare_fb
      |                       ^~~~~~~~~~~~~~~~~~~~~
      |                       drm_gem_fb_create
/usr/src/evdi-1.9.1/evdi_modeset.c:336:26: error: initialization of 'void (*)(struct drm_plane *, struct drm_atomic_state *)' from incompatible pointer type 'void (*)(struct drm_plane *, struct drm_plane_state *)' [-Werror=incompatible-pointer-types]
  336 |         .atomic_update = evdi_cursor_atomic_update,
      |                          ^~~~~~~~~~~~~~~~~~~~~~~~~
/usr/src/evdi-1.9.1/evdi_modeset.c:336:26: note: (near initialization for 'evdi_cursor_helper_funcs.atomic_update')
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:273: /usr/src/evdi-1.9.1/evdi_modeset.o] Error 1
make[1]: *** [Makefile:1853: /usr/src/evdi-1.9.1] Error 2
make[1]: Leaving directory '/usr/src/kernels/5.13.4-200.fc34.x86_64'
make: *** [Makefile:77: module] Error 2

davidschenz, 2021-07-24:
It looks like this issue in the upstream EVDI:

https://github.com/DisplayLink/evdi/issues/296

davidschenz, 2021-07-24:
As a brief update, I was able to download the devel branch of EVDI, unzipped it, and then ran "make" and "sudo make install" and my monitor popped right on. I'm guessing that means that displaylink-rpm needs a new build to pick up the upstream changes in EVDI?

iranzo, 2021-07-24:
I was able to compile it when using the devel branch, still other stuff on my case as it complained about the module (invalid format)

Would be great to have a repo we can use to pull updates from here when they arrive

elguero, 2021-07-24:
Correct.  [Upstream ](https://github.com/DisplayLink/evdi) has not released a new version of evdi.

If you clone this repo and do a `make rawhide`, it will build an rpm using the latest commit in the evdi repo.  There is no need for us to maintain a repo with evdi updates.

bobertlo, 2021-07-25:
> Correct. [Upstream ](https://github.com/DisplayLink/evdi) has not released a new version of evdi.
> 
> If you clone this repo and do a `make rawhide`, it will build an rpm using the latest commit in the evdi repo. There is no need for us to maintain a repo with evdi updates.

I have been having trouble with this. I run `make rawhide` and install the package and the dkms source code is the 1.9.1 release, while `/var/tmp/evdi-1.9.1` is checked out on the devel branch.

phily245, 2021-07-26:
What is the workaround command?

bobertlo, 2021-07-26:
> 
> 
> What is the workaround command?

I just copied the contents of `/var/tmp/evdi-1.9.1/module/` (or a different copy of the evdi `devel` git branch) into `/usr/src/evdi-1.9.1`. I am not near that machine now so double check the paths but I installed the `devel` source files directly over the installed source files so they had the patches to compile with 5.13 kernels.

elguero, 2021-07-26:
My fault. `make rawhide` uses the bundled version still.  Probably will change this.

DisplayLink changes have really messed up the flow of things here.

In the meantime, try the following:

`make devel`  <--- Should pull down and tarball up the latest development drivers
`make unbundled` <--- Should create an RPM not using what is in the DisplayLink driver package. Rather, should use the tarball devel version created with the first command.

Haven't had time to test this but I believe that should work.  When I have time, I will dig into how we can make this cleaner and simple for everyone.

Hopefully, DisplayLink / EVDI will release a new version soon.

sgoley, 2021-07-28:
I followed @bobertlo's instructions first on simply building the `devel` branch and then copying `/module/` to `/usr/src/evdi-1.9.1`. 
It's definitely a step but only 1 of 2 monitors is working solidly. The other one.... it's "on" but basically freeze framed from when I enable that within Settings -> Displays. It had some significant visual artifacts when on 60 Hz, but appears "correctly" but frozen on the 75 Hz setting.

Currently on Fedora 34 w kernel 5.13.4-200.fc34.x86_64.

Happy to provide logs, etc. as requested.

kraftzwerg, 2021-07-30:
The Problem occurs with the latest Kernel update.
for me the manual setup of evdi solved the problem:

`git clone https://github.com/DisplayLink/evdi.git
cd evdi
make
sudo make install
cd ..
wget https://github.com/displaylink-rpm/displaylink-rpm/releases/download/v5.4.0-1/fedora-34-displaylink-1.9.1-1.x86_64.rpm
sudo yum install fedora-34-displaylink-1.9.1-1.x86_64.rpm`

blackforestcode, 2021-08-01:
> The Problem occurs with the latest Kernel update.
> for me the manual setup of evdi solved the problem:
>
> git clone https://github.com/DisplayLink/evdi.git
> cd evdi
> make
> sudo make install
> cd ..
> wget [...]

Worked for me, thanks!

t2d, 2021-08-02:
> The Problem occurs with the latest Kernel update.
> for me the manual setup of evdi solved the problem:
>
> git clone https://github.com/DisplayLink/evdi.git
> cd evdi
> make
> sudo make install
> cd ..
> wget [...]

I had to copy `signing_key.pem` and `signing_key.x509` to `/usr/src/kernels/5.13.5-200.fc34.x86_64/certs` before that to make Secure Boot work.
Than `sudo make install` will sign the compiled module automatically. (Similar to [this question](https://superuser.com/questions/1214116/no-openssl-sign-file-signing-key-pem-leads-to-error-while-loading-kernel-modules))

phily245, 2021-08-04:
> I had to copy `signing_key.pem` and `signing_key.x509` to `/usr/src/kernels/5.13.5-200.fc34.x86_64/certs` before that to make Secure Boot work.
> Than `sudo make install` will sign the compiled module [...]

I had to copy the keys too.

I also have the added peril of following the secure boot instructions, but it looks like this is failing as I'm missing a file when signing the driver:

> [root@fedora displaylink-rpm]# modinfo -n evdi
> /lib/modules/5.13.6-200.fc34.x86_64/kernel/drivers/gpu/drm/evdi/evdi.ko
> [root@fedora displaylink-rpm]# sudo /usr/src/kernels/$(uname -r)/scripts/sign-file sha256 ./MOK.priv ./MOK.der /lib/modules/$(uname -r)/extra/evdi.ko
> At main.c:291:
> - SSL error:02001002:system library:fopen:No such file or directory: crypto/bio/bss_file.c:69
> - SSL error:2006D080:BIO routines:BIO_new_file:no such file: crypto/bio/bss_file.c:76
> sign-file: /lib/modules/5.13.6-200.fc34.x86_64/extra/evdi.ko: No such file or directory

phily245, 2021-08-04:
> I had to copy the keys too.
>
> I also have the added peril of following the secure boot instructions, but it looks like this is failing as I'm missing a file when signing the driver:

I managed to solve this.

I uninstalled both , ran `make clean` for evdi and ran the evdi/modules Makefile as `install_dkms`, so the happy path for me (using my existing MOK key):

> cd /usr/src/kernels/5.13.6-200.fc34.x86_64/certs
> openssl req -new -nodes -utf8 -sha512 -days 36500 -batch -x509 -config x509.genkey -outform DER -out signing_key.x509 -keyout signing_key.pem
> cd ~/displaylink
> git clone https://github.com/DisplayLink/evdi.git
> cd evdi
> make
> cd modules
> sudo make install_dkms
> cd ../..
> wget https://github.com/displaylink-rpm/displaylink-rpm/releases/download/v5.4.0-1/fedora-34-displaylink-1.9.1-1.x86_64.rpm
> sudo yum install fedora-34-displaylink-1.9.1-1.x86_64.rpm
> sudo modinfo -n evdi
> sudo unxz $(modinfo -n evdi)
> sudo /usr/src/kernels/$(uname -r)/scripts/sign-file sha256 ./MOK.priv \
> ./MOK.der /lib/modules/$(uname -r)/extra/evdi.ko
> xz -f /lib/modules/$(uname -r)/extra/evdi.ko

brego81, 2021-08-16:
Hi all, when running make i got this error:
``
CFLAGS="-Werror -Wextra -Wall -Wmissing-prototypes -Wstrict-prototypes -Wno-error=missing-field-initializers " make -C module 
make[1]: Entering directory '/home/matteo-bregonzio/evdi/module'
make -C /lib/modules/5.13.9-200.fc34.x86_64/build M=$PWD
make[2]: Entering directory '/usr/src/kernels/5.13.9-200.fc34.x86_64'
make[2]: Leaving directory '/usr/src/kernels/5.13.9-200.fc34.x86_64'
make[1]: Leaving directory '/home/matteo-bregonzio/evdi/module'
CFLAGS="-I../module -Werror -Wextra -Wall -Wmissing-prototypes -Wstrict-prototypes -Wno-error=missing-field-initializers " make -C library 
make[1]: Entering directory '/home/matteo-bregonzio/evdi/library'
cc -I../module -std=gnu99 -fPIC -D_FILE_OFFSET_BITS=64 -I../module -Werror -Wextra -Wall -Wmissing-prototypes -Wstrict-prototypes -Wno-error=missing-field-initializers    -c -o evdi_lib.o evdi_lib.c
evdi_lib.c:6:10: fatal error: libdrm/drm.h: No such file or directory
    6 | #include <libdrm/drm.h>
      |          ^~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [<builtin>: evdi_lib.o] Error 1
make[1]: Leaving directory '/home/matteo-bregonzio/evdi/library'
make: *** [Makefile:9: all] Error 2
``

brego81, 2021-08-17:
> Hi all, when running make on fedora 34 i got this error:
> 
> CFLAGS="-Werror -Wextra -Wall -Wmissing-prototypes -Wstrict-prototypes -Wno-error=missing-field-initializers " make -C module
> make[1]: Entering directory '/home/evdi/module'
> make -C /lib/modules/5.13.9-200.fc34.x86_64/build M=$PWD
> make[2]: Entering directory '/usr/src/kernels/5.13.9-200.fc34.x86_64'
> make[2]: Leaving directory '/usr/src/kernels/5.13.9-200.fc34.x86_64'
> make[1]: Leaving directory '/home/evdi/module'
> CFLAGS="-I../module -Werror -Wextra -Wall -Wmissing-prototypes -Wstrict-prototypes -Wno-error=missing-field-initializers " make -C library
> make[1]: Entering directory '/home/evdi/library'
> cc -I../module -std=gnu99 -fPIC -D_FILE_OFFSET_BITS=64 -I../module -Werror -Wextra -Wall -Wmissing-prototypes -Wstrict-prototypes -Wno-error=missing-field-initializers -c -o evdi_lib.o evdi_lib.c
> **evdi_lib.c:6:10: fatal error: libdrm/drm.h: No such file or directory 6 | #include <libdrm/drm.h> | ^~~~~~~~~~~~~~ compilation terminated.**
> make[1]: *** [: evdi_lib.o] Error 1
> make[1]: Leaving directory '/home/evdi/library'
> make: *** [Makefile:9: all] Error 2

I managed to get it working by doing:

- **sudo dnf install libdrm-devel.x86_64 libdrm.x86_64**
- git clone https://github.com/DisplayLink/evdi.git
- cd evdi
- make
- sudo make install
- cd ..
- wget https://github.com/displaylink-rpm/displaylink-rpm/releases/download/v5.4.0-1/fedora-34-displaylink-1.9.1-1.x86_64.rpm
- sudo dnf install fedora-34-displaylink-1.9.1-1.x86_64.rpm

kovacs-andras, 2021-09-05:
Many thanks @kraftzwerg  and @brego81 !
It also works on Fedora33 with:
`dnf install https://github.com/displaylink-rpm/displaylink-rpm/releases/download/v5.4.0-1/fedora-33-displaylink-1.9.1-1.x86_64.rpm
`
(you don't need to wget the package first, dnf/yum can both install on-the-fly)

avilesj, 2021-11-07:
@brego81 's solution also works under F35 (using 34's RPM). The experience is terrible since the extra monitor is very laggy.

dave-fl, 2021-11-20:
@phily245 I don't believe there is any need to generate the certificates twice.  I basically generated the certs under `/usr/src/kernels/5.14.18-300.fc35.x86_64/certs` and then imported using `mokutil`.  Seemed to work just fine.

dave-fl, 2021-11-20:
@phily245 I don't believe there is a need to generate the certificates twice. I generated the certs under /usr/src/kernels/5.14.18-300.fc35.x86_64/certs and then imported using mokutil. Seems to work fine.

davetwog, 2021-11-28:
Fedora 35 and Kernel 5.14.18-300.fc35.x86_64 not working. Is an update comming?

elguero, 2021-11-28:
DisplayLink is working on an update.

https://github.com/DisplayLink/evdi/issues/309#issuecomment-979831346

askb, 2021-12-03:
> > Hi all, when running make on fedora 34 i got this error:
> > CFLAGS="-Werror -Wextra -Wall -Wmissing-prototypes -Wstrict-prototypes -Wno-error=missing-field-initializers " make -C module
> > make[1]: Entering directory '/home/evdi/module'
> > make -C /lib/modules/5.13.9-200.fc34.x86_64/build M=$PWD
> > make[2]: Entering directory '/usr/src/kernels/5.13.9-200.fc34.x86_64'
> > make[2]: Leaving directory '/usr/src/kernels/5.13.9-200.fc34.x86_64'
> > make[1]: Leaving directory '/home/evdi/module'
> > CFLAGS="-I../module -Werror -Wextra -Wall -Wmissing-prototypes -Wstrict-prototypes -Wno-error=missing-field-initializers " make -C library
> > make[1]: Entering directory '/home/evdi/library'
> > cc -I../module -std=gnu99 -fPIC -D_FILE_OFFSET_BITS=64 -I../module -Werror -Wextra -Wall -Wmissing-prototypes -Wstrict-prototypes -Wno-error=missing-field-initializers -c -o evdi_lib.o evdi_lib.c
> > **evdi_lib.c:6:10: fatal error: libdrm/drm.h: No such file or directory 6 | #include <libdrm/drm.h> | ^~~~~~~~~~~~~~ compilation terminated.**
> > make[1]: *** [: evdi_lib.o] Error 1
> > make[1]: Leaving directory '/home/evdi/library'
> > make: *** [Makefile:9: all] Error 2
> 
> I managed to get it working by doing:
> 
> * **sudo dnf install libdrm-devel.x86_64 libdrm.x86_64 kernel-devel-$(uname -r)**
> * git clone https://github.com/DisplayLink/evdi.git
> * cd evdi
> * make
> * sudo make install
> * cd ..
> * wget https://github.com/displaylink-rpm/displaylink-rpm/releases/download/v5.4.0-1/fedora-34-displaylink-1.9.1-1.x86_64.rpm
> * sudo dnf install fedora-34-displaylink-1.9.1-1.x86_64.rpm

Thanks! Works with Fedora 35 (5.15.5-200.fc35.x86_64)

jglizardi, 2021-12-07:
> Thanks! Works with Fedora 35 (5.15.5-200.fc35.x86_64)

Works for me too!!!

Fedore 35 (5.15.6-200.fc35.x86_64)

elguero, 2021-12-29:
I am going to close this issue out since it should be resolved with the 5.4.1 release.

closed 2021-12-29 by elguero

CryptoKiddies, 2023-08-31:
> The Problem occurs with the latest Kernel update. for me the manual setup of evdi solved the problem:
>
> git clone https://github.com/DisplayLink/evdi.git cd evdi make sudo make install cd .. wget [...]

This is not working in Fedora 38. It also looks like evdi repo is configured for ubuntu. When I run make, I get:
```
[g@fedora evdi]$ make
CFLAGS="-isystem./include -isystem./include/uapi -Werror -Wextra -Wall -Wno-error=missing-field-initializers -Werror=sign-compare -Wmissing-prototypes -Wstrict-prototypes -Werror=discarded-qualifiers " make -C module 
make[1]: Entering directory '/home/g/projects/evdi/module'
make -C /lib/modules/6.4.12-200.fc38.x86_64/build M=$PWD
make[2]: Entering directory '/home/g/projects/evdi/module'
make[2]: *** /lib/modules/6.4.12-200.fc38.x86_64/build: No such file or directory.  Stop.
make[2]: Leaving directory '/home/g/projects/evdi/module'
make[1]: *** [Makefile:86: module] Error 2
make[1]: Leaving directory '/home/g/projects/evdi/module'
make: *** [Makefile:10: all] Error 2
```