displaylink-rpm/displaylink-rpm #173: `make rawhide` uses the bundled version evdi

bobertlo, 2021-08-07:
My fault. `make rawhide` uses the bundled version still.  Probably will change this.

DisplayLink changes have really messed up the flow of things here.

In the meantime, try the following:

`make devel`  <--- Should pull down and tarball up the latest development drivers
`make unbundled` <--- Should create an RPM not using what is in the DisplayLink driver package. Rather, should use the tarball devel version created with the first command.

Haven't had time to test this but I believe that should work.  When I have time, I will dig into how we can make this cleaner and simple for everyone.

Hopefully, DisplayLink / EVDI will release a new version soon.

_Originally posted by @elguero in https://github.com/displaylink-rpm/displaylink-rpm/issues/172#issuecomment-886772311_

bobertlo, 2021-08-07:
I am not sure if the documentation should be updated or the `rawhide` target should be changed. Does the target do something useful as is or should it be changed to match the documentation?

I would not mind looking into this but I am unsure what the best way to fix it would be.

ssaavedra, 2021-09-06:
Historically both things have been useful in different moments of the repo.

That's why the devel target exists separately from rawhide.

Initially, rawhide did what the current unbundled/devel targets do: it goes to EVDI_DEV_BRANCH and builds with that code.

But at some point in the future we needed to build displaylink-rpm for actual Fedora Rawhide and we repurposed the names. However, it seems we forgot to update the documentation about it.

Long story short, both things are useful, and we should probably discuss here what's _clearer_ before doing one over the other.

I personally believe that rawhide is easier to understand if it means "latest versions, like Fedora Rawhide" and devel or unbundled to mean an unbundled evdi from an EVDI_DEV_BRANCH Make variable. But please speak out, I just stepped in to dig up history 😅

elguero, 2021-09-17:
Thanks for the history @ssaavedra .

DisplayLink is not making it easy to count on a tagged release anymore for evdi.

I would like to see what they have to say about their plans to help us understand the best direction to go.

https://github.com/DisplayLink/evdi/issues/309

PR #184 (2021-12-28): Fix rawhide build, clarify using evdi github release package
- Fix rawhide build to use package from Github and not the one
  that is being bundled with the Displaylink driver package

- Change wording of 'unbundled' to refer to Github package to
  clarify which tarball will be used in the RPM build.

- Cleanups and add more documentation to README

Fixes #173
commit b2b03e1b (2021-12-28): Fix rawhide build, clarify using evdi github release package (#184)

- Fix rawhide build to use package from Github and not the one
  that is being bundled with the Displaylink driver package

- Change wording of 'unbundled' to refer to Github package to
  clarify which tarball will be used in the RPM build.

- Cleanups and add more documentation to README

Fixes #173
commit 9eec8543 (2021-12-28): Fix rawhide build, clarify using evdi github release package

- Fix rawhide build to use package from Github and not the one
  that is being bundled with the Displaylink driver package

- Change wording of 'unbundled' to refer to Github package to
  clarify which tarball will be used in the RPM build.

- Cleanups and add more documentation to README

Fixes #173
closed 2021-12-28 by elguero