displaylink-rpm/displaylink-rpm #174: extrem lag and slow overall performance

groovyman, 2021-08-10:
I am trying to run a new notebook with usb3.2 gen2 with 10Gb/s with a capable docking station usiing a DL-6000 chip.
The dockingstation comes with 3 display connectors but onyl 2 are working now.

Thank you , i was able to install the displaylink-rpm under Fedora F34.

The lack is a desaster. The displaylink process eats up more than 100% CPU time and seems to be part of the very slow overall output. Before i was able to run the same notebook with W10 on an old docking station with max 5Gb/s and i had absolutly no problem with 2 4K monitors and no CPU eating process.

Hardware:
. Lenovo 16p with AMD CPU

    Startech DK31C3HDPD
    and 3 4K Monitors

Just let you know: 2*4K Monitor did work with an old Docking station using a USB3 connection on Windows10.

I will try to figure out, what is going wrong.

groovyman, 2021-08-20:
I was able to install a W10 on my laptop and was surprised what could be, if ....
There was a little lag, and a low CPU consumption. Besides i was able to adress all 3 of my 4K Monitors without any problem.

Maybe there is something wrong in the way, how the code accesses the USB on the AMD-Driven notebook.

groovyman, 2021-08-29:
Dear @elguero 

in oder to figure out, why the DisplayManager eats up the CPU time i ran a strace on that process, and was surprised, that this process stayed in the execution of a select call. This means, that all burned CPU time is wasted in the kernel!
Did somebody figured out, what the hell the driver is doing there?
If not, i could perform a trace in order to figures out, whats wrong!

elguero, 2022-03-21:
@groovyman Can you try the latest release?  I am seeing a lot of reports over on the [Displaylink/evdi ](https://github.com/DisplayLink/evdi/issues) repo. Maybe you can help add some information over there if you are still seeing this issue.

elguero, 2022-07-22:
There is a new RPM package out.  Can you give that a try and provide some feedback?  Thanks.

elguero, 2022-08-12:
Closing for now.

closed 2022-08-12 by elguero