displaylink-rpm/displaylink-rpm #176: Fedora 35 and Kernel 5.14.16 issues

avilesj, 2021-11-07:
**Please note:** The only working combination is *DisplayLink 5.3.1* and *evdi v1.7.2* OR *DisplayLink 5.4* and *evdi v1.9.1*

  * Are you using the latest DisplayLink driver?
    Yes, 5.4
  * Are you using the latest EVDI module?
   Yes, 1.9.1
  * If you are using a DisplayLink device, have you checked 'troubleshooting'
    on DisplayLink's website?
    Yes
  * Is this issue related to evdi/kernel?
    Probably
  * Is this issue related to DisplayLinkManager?
    Probably not
  * If you are having an issue upgrading, have you tried to do a clean install of the driver?
    

  * Fedora or CentOS: Fedora
  * Version: 35
  * Kernel version `(uname -a)`:  Linux fedora 5.14.16-301.fc35.x86_64 #1 SMP Wed Nov 3 13:55:42 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

  * `/var/log/displaylink/displaylink.log`

```
Creating symlink /var/lib/dkms/evdi/1.9.1/source -> /usr/src/evdi-1.9.1
Error! Your kernel headers for kernel 5.14.10-300.fc35.x86_64 cannot be found at /lib/modules/5.14.10-300.fc35.x86_64/build or /lib/modules/5.14.10-300.fc35.x86_64/source.
You can use the --kernelsourcedir option to tell DKMS where it's located.
Error! Your kernel headers for kernel 5.14.10-300.fc35.x86_64 cannot be found at /lib/modules/5.14.10-300.fc35.x86_64/build or /lib/modules/5.14.10-300.fc35.x86_64/source.
You can use the --kernelsourcedir option to tell DKMS where it's located.
Deleting module evdi-1.9.1 completely from the DKMS tree.
Creating symlink /var/lib/dkms/evdi/1.9.1/source -> /usr/src/evdi-1.9.1

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
make -j8 KERNELRELEASE=5.14.16-301.fc35.x86_64 all INCLUDEDIR=/lib/modules/5.14.16-301.fc35.x86_64/build/include KVERSION=5.14.16-301.fc35.x86_64 DKMS_BUILD=1...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.14.16-301.fc35.x86_64 (x86_64)
Consult /var/lib/dkms/evdi/1.9.1/build/make.log for more information.

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
make -j8 KERNELRELEASE=5.14.16-301.fc35.x86_64 all INCLUDEDIR=/lib/modules/5.14.16-301.fc35.x86_64/build/include KVERSION=5.14.16-301.fc35.x86_64 DKMS_BUILD=1...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.14.16-301.fc35.x86_64 (x86_64)
Consult /var/lib/dkms/evdi/1.9.1/build/make.log for more information.
Deleting module evdi-1.9.1 completely from the DKMS tree.
Creating symlink /var/lib/dkms/evdi/1.9.1/source -> /usr/src/evdi-1.9.1

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
make -j8 KERNELRELEASE=5.14.16-301.fc35.x86_64 all INCLUDEDIR=/lib/modules/5.14.16-301.fc35.x86_64/build/include KVERSION=5.14.16-301.fc35.x86_64 DKMS_BUILD=1...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.14.16-301.fc35.x86_64 (x86_64)
Consult /var/lib/dkms/evdi/1.9.1/build/make.log for more information.

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
make -j8 KERNELRELEASE=5.14.16-301.fc35.x86_64 all INCLUDEDIR=/lib/modules/5.14.16-301.fc35.x86_64/build/include KVERSION=5.14.16-301.fc35.x86_64 DKMS_BUILD=1...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.14.16-301.fc35.x86_64 (x86_64)
Consult /var/lib/dkms/evdi/1.9.1/build/make.log for more information.
Deleting module evdi-1.9.1 completely from the DKMS tree.
Creating symlink /var/lib/dkms/evdi/1.9.1/source -> /usr/src/evdi-1.9.1

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
make -j8 KERNELRELEASE=5.14.16-301.fc35.x86_64 all INCLUDEDIR=/lib/modules/5.14.16-301.fc35.x86_64/build/include KVERSION=5.14.16-301.fc35.x86_64 DKMS_BUILD=1...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.14.16-301.fc35.x86_64 (x86_64)
Consult /var/lib/dkms/evdi/1.9.1/build/make.log for more information.

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
make -j8 KERNELRELEASE=5.14.16-301.fc35.x86_64 all INCLUDEDIR=/lib/modules/5.14.16-301.fc35.x86_64/build/include KVERSION=5.14.16-301.fc35.x86_64 DKMS_BUILD=1...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.14.16-301.fc35.x86_64 (x86_64)
Consult /var/lib/dkms/evdi/1.9.1/build/make.log for more information.

```
  * `/var/lib/dkms/evdi/kernel-[version]/log/make.log`

`evdi/kernel-[version]` does not exists

Past releases don't seem to work under F35 even manually compiling evdi. `make rawhide` is not working.

* Clean install Fedora 35
* Install Fedora 34's rpm from releases
* Manually compile `evdi`

I recently installed Fedora 35 and proceeded to use version 34's RPM just to see if it worked. It did not, so I compiled `evdi` 1.9.1 manually, and after `sudo make install`, the extra display lighted up.

The problem is, the display is extremely laggy and I should've probably compiled the repo. I tried to compile this repo with `make rawhide` but I'm getting  `ERROR   0004: file '/usr/libexec/displaylink/DisplayLinkManager' contains an insecure  '.' in [.]` which, looking around the issues, seems to be fixed by running `make rawhide`.. therefore `make rawhide` does not work.

There is nothing to check regarding hardware. I was running PopOS earlier this morning without issue.

ssaavedra, 2021-11-10:
Have you checked whether the last comments at https://github.com/displaylink-rpm/displaylink-rpm/issues/172 solve your issue? This might be a duplicate.

dave-fl, 2021-11-20:
@ssaavedra I can confirm that #172 steps work with F35.

cdonatom, 2021-12-02:
@elguero yesterday I committed some changes on the evdi repo to support Kernel 5.14 and 5.15, so the module should compile and be functional.

Meanwhile, trying out to generate the RPM packages, it won't work due to this issue

`+ /usr/lib/rpm/check-rpaths
*******************************************************************************
*
* WARNING: 'check-rpaths' detected a broken RPATH OR RUNPATH and will cause
*          'rpmbuild' to fail. To ignore these errors, you can set the
*          '$QA_RPATHS' environment variable which is a bitmask allowing the
*          values below. The current value of QA_RPATHS is 0x0000.
*
*    0x0001 ... standard RPATHs (e.g. /usr/lib); such RPATHs are a minor
*               issue but are introducing redundant searchpaths without
*               providing a benefit. They can also cause errors in multilib
*               environments.
*    0x0002 ... invalid RPATHs; these are RPATHs which are neither absolute
*               nor relative filenames and can therefore be a SECURITY risk
*    0x0004 ... insecure RPATHs; these are relative RPATHs which are a
*               SECURITY risk
*    0x0008 ... the special '$ORIGIN' RPATHs are appearing after other
*               RPATHs; this is just a minor issue but usually unwanted
*    0x0010 ... the RPATH is empty; there is no reason for such RPATHs
*               and they cause unneeded work while loading libraries
*    0x0020 ... an RPATH references '..' of an absolute path; this will break
*               the functionality when the path before '..' is a symlink
*          
*
* Examples:
* - to ignore standard and empty RPATHs, execute 'rpmbuild' like
*   $ QA_RPATHS=$(( 0x0001|0x0010 )) rpmbuild my-package.src.rpm
* - to check existing files, set $RPM_BUILD_ROOT and execute check-rpaths like
*   $ RPM_BUILD_ROOT=<top-dir> /usr/lib/rpm/check-rpaths
*  
*******************************************************************************
ERROR   0004: file '/usr/libexec/displaylink/DisplayLinkManager' contains an insecure  '.' in [.]
error: Bad exit status from /var/tmp/displayportUMVOfB/rpm-tmp.GtzQso (%install)

RPM build errors:
    Bad exit status from /var/tmp/displayportUMVOfB/rpm-tmp.GtzQso (%install)
make: *** [Makefile:147: i386/displaylink-1.9.1-2.i386.rpm] Error 1
`

Could you please update your scripts to properly generate the rpms? Thx!

elguero, 2021-12-03:
> ERROR   0004: file '/usr/libexec/displaylink/DisplayLinkManager' contains an insecure  '.' in [.]
> error: Bad exit status from /var/tmp/displayportUMVOfB/rpm-tmp.GtzQso (%install)

I have investigated this and see what the issue is. With the release of Fedora 35, the RPATH check is now being enforced. This binary being referenced is part of the DisplayLink binary package that is downloaded.

We will work on disabling this check for now and hopefully DisplayLink will remove RPATH in a future update.

closed 2022-03-21 by elguero