displaylink-rpm/displaylink-rpm #177: Only works if dock plugged in after boot

dave-fl, 2021-11-20:
**Please note:** The only working combination is *DisplayLink 5.3.1* and *evdi v1.7.2* OR *DisplayLink 5.4* and *evdi v1.9.1*

  * Are you using the latest DisplayLink driver?
    Yes, 5.4
    
  * Are you using the latest EVDI module?
    
    Yes, 1.9.1
  * If you are using a DisplayLink device, have you checked 'troubleshooting'
    on DisplayLink's website?
    Yes
    
  * Is this issue related to evdi/kernel?
    
    No idea
  * Is this issue related to DisplayLinkManager?
    No idea
    
  * If you are having an issue upgrading, have you tried to do a clean install of the driver?
    

  * Fedora
  * F35
  * Linux fedora 5.14.18-300.fc35.x86_64 #1 SMP Fri Nov 12 16:43:17 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

Starting from no power to on if the dock is plugged into the USBC port then the displays will not work.  Starting from no power to on if the dock is plugged in once the system has fully booted then things work fine.

ssaavedra, 2021-12-17:
This could be any number of things, including stuff related to the hardware platform. But I'd say the most likely suspects are either the udev rules or DisplayLinkManager acting funky.

If you start form no power with the dock cold-plugged, and do a systemctl restart DisplayLinkManager, does it work?

Does the output of `lsusb -vv` tell you anything different if the device is cold-plugged than if hot-plugged?

I mean this:
1. Power off, connect dock, power on, open terminal, type: `lsusb -vv > $HOME/lsusb-coldplug`
2. Power off, disconnect dock, power on, connect dock, wait for signal, open terminal, type: `lsusb -vv > $HOME/lsusb-hotplug`
3. Is there anything significantly different between the lsusb-coldplug and lsusb-hotplug files? Particularly w.r.t. the driver detected on the usb device, or whether it's been found.

dave-fl, 2021-12-18:
@ssaavedra After upgrading to `5.15.7-200.fc35.x86_64` and doing a pull for the latest evdi Kernel 5.15 fixes https://github.com/DisplayLink/evdi/commit/a41fb18a12d6d189e0fb01066e3c6a8e4e7d877c things seem to working just fine.

closed 2021-12-18 by dave-fl