displaylink-rpm/displaylink-rpm #178: Disable RPATH Check

elguero, 2021-12-06:
Reported by @cdonatom: https://github.com/displaylink-rpm/displaylink-rpm/issues/176#issuecomment-984426514

closed 2021-12-06 by elguero

cdonatom, 2021-12-06:
Thanks a lot!

hlovdal, 2022-01-01:
Well, the spec file change has no effect on my machine and the build still fails with the latest commit on a fresh checkout:

```shell
$ git clone https://github.com/displaylink-rpm/displaylink-rpm
Cloning into 'displaylink-rpm'...
remote: Enumerating objects: 611, done.
remote: Counting objects: 100% (168/168), done.
remote: Compressing objects: 100% (116/116), done.
remote: Total 611 (delta 98), reused 103 (delta 50), pack-reused 443
Receiving objects: 100% (611/611), 160.27 KiB | 2.39 MiB/s, done.
Resolving deltas: 100% (381/381), done.
$ cd displaylink-rpm
$ make
wget -O DisplayLink\ USB\ Graphics\ Software\ for\ Ubuntu\ 5.4.1-55.174.zip \
        "https://www.synaptics.com/sites/default/files/exe_files/2021-09/DisplayLink USB Graphics Software for Ubuntu5.4.1-EXE.zip"
--2022-01-01 00:54:00--  https://www.synaptics.com/sites/default/files/exe_files/2021-09/DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu5.4.1-EXE.zip
Resolving www.synaptics.com (www.synaptics.com)... 34.214.248.103, 52.41.97.53
Connecting to www.synaptics.com (www.synaptics.com)|34.214.248.103|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 14500327 (14M) [application/zip]
Saving to: 'DisplayLink USB Graphics Software for Ubuntu 5.4.1-55.174.zip'

DisplayLink USB Graphics Software for Ubu 100%[====================================================================================>]  13.83M  7.38MB/s    in 1.9s

2022-01-01 00:54:03 (7.38 MB/s) - 'DisplayLink USB Graphics Software for Ubuntu 5.4.1-55.174.zip' saved [14500327/14500327]

rpmbuild -bb --define "_topdir `pwd`" --define "_sourcedir `pwd`" --define "_rpmdir `pwd`" --define "_specdir `pwd`" --define "_srcrpmdir `pwd`" --define "_buildrootdir `mktemp -d /var/tmp/displayportXXXXXX`" --define "_builddir `mktemp -d /var/tmp/displayportXXXXXX`" --define "_release 3" --define "_daemon_version 5.4.1-55.174" --define "_version 1.9.1" --define "_tmppath `mktemp -d /var/tmp/displayportXXXXXX`"  displaylink.spec --target=i386
Building target platforms: i386
Building for target i386
setting SOURCE_DATE_EPOCH=1640736000
Executing(%prep): /bin/sh -e /var/tmp/displayportFo9LLr/rpm-tmp.udiKkt
+ umask 022
+ cd /var/tmp/displayport3O4Ycf
+ cd /var/tmp/displayport3O4Ycf
+ rm -rf displaylink-1.9.1
+ /usr/bin/mkdir -p displaylink-1.9.1
+ cd displaylink-1.9.1
+ /usr/bin/chmod -Rf a+rX,u+w,g-w,o-w .
+ cd /var/tmp/displayport3O4Ycf
+ cd displaylink-1.9.1
+ /usr/bin/unzip -qq '/download/src/github/displaylink-rpm/DisplayLink USB Graphics Software for Ubuntu 5.4.1-55.174.zip'
+ STATUS=0
+ '[' 0 -ne 0 ']'
+ /usr/bin/chmod -Rf a+rX,u+w,g-w,o-w .
+ chmod +x displaylink-driver-5.4.1-55.174.run
+ ./displaylink-driver-5.4.1-55.174.run --noexec --keep
Creating directory displaylink-driver-5.4.1-55.174
Verifying archive integrity...  100%   All good.
Uncompressing DisplayLink Linux Driver 5.4.1-55.174  100%
+ mkdir -p evdi-1.9.1
+ mv displaylink-driver-5.4.1-55.174/evdi.tar.gz evdi-1.9.1
+ cd evdi-1.9.1
+ gzip -dc evdi.tar.gz
+ tar -xvvf -
drwxrwxr-x root/root         0 2021-08-11 11:30 .github/
-rw-rw-r-- root/root      1310 2021-08-11 11:30 .github/ISSUE_TEMPLATE.md
-rw-rw-r-- root/root       656 2021-08-11 11:30 .github/PULL_REQUEST_TEMPLATE.md
-rw-rw-r-- root/root       212 2021-08-11 11:30 .gitignore
-rw-rw-r-- root/root       623 2021-08-11 11:30 .travis.yml
-rw-rw-r-- root/root       572 2021-08-11 11:30 Makefile
-rw-rw-r-- root/root      3177 2021-08-11 11:30 README.md
drwxrwxr-x root/root         0 2021-08-11 11:30 ci/
-rwxrwxr-x root/root      6825 2021-08-11 11:30 ci/build_against_kernel
-rwxrwxr-x root/root      3460 2021-08-11 11:30 ci/export_for_kernel
-rwxrwxr-x root/root       653 2021-08-11 11:30 ci/run_style_check
-rwxrwxr-x root/root      1168 2021-08-11 11:30 ci/set_version
drwxrwxr-x root/root         0 2021-08-11 11:30 docs/
-rw-rw-r-- root/root     16071 2021-08-11 11:30 docs/details.md
-rw-rw-r-- root/root      1061 2021-08-11 11:30 docs/index.md
-rw-rw-r-- root/root      7517 2021-08-11 11:30 docs/quickstart.md
[... 112 lines cut ...]
* - to ignore standard and empty RPATHs, execute 'rpmbuild' like
*   $ QA_RPATHS=$(( 0x0001|0x0010 )) rpmbuild my-package.src.rpm
* - to check existing files, set $RPM_BUILD_ROOT and execute check-rpaths like
*   $ RPM_BUILD_ROOT=<top-dir> /usr/lib/rpm/check-rpaths
*
*******************************************************************************
ERROR   0004: file '/usr/libexec/displaylink/DisplayLinkManager' contains an insecure  '.' in [.]
error: Bad exit status from /var/tmp/displayportFo9LLr/rpm-tmp.jXypLZ (%install)

RPM build errors:
    Bad exit status from /var/tmp/displayportFo9LLr/rpm-tmp.jXypLZ (%install)
make: *** [Makefile:152: i386/displaylink-1.9.1-3.i386.rpm] Error 1
$ uname -a
Linux ........  5.15.11-200.fc35.x86_64 #1 SMP Wed Dec 22 15:41:11 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
$ cat /etc/fedora-release
Fedora release 35 (Thirty Five)
$ git show HEAD | head -6
commit 41a7aa3edab30370ce693bbad427acee0b40192d
Author: Michael L. Young <elgueromexicano@gmail.com>
Date:   Wed Dec 29 23:36:23 2021 -0500

    Add a CI job to test the ability to run mock using the spec file (#191)

$
```

hlovdal, 2022-01-01:
Running `env QA_RPATHS=0x0004 make` builds `displaylink-1.9.1-3.src.rpm` successfully.

elguero, 2022-01-01:
Hi @hlovdal.  I wonder what the difference is.  The CI build check for Fedora 35 passed without any errors.

https://github.com/displaylink-rpm/displaylink-rpm/runs/4664101603?check_suite_focus=true#logs

elguero, 2022-01-01:
I checked into this some more. I reproduced it on my end on a Fedora 35 machine.

What I found is that I had a `.rpmmacros` file under my home directory which is calling the `check_rpaths` command directly.  That would be the reason the change made to the spec file is not having any effect due to your build environment.

I have tracked this down to running the `rpmdev-setuptree` command.  Have you run that command on your system?  It creates the `.rpmmacros` file which calls `check-rpaths` directly.

hlovdal, 2022-01-01:
Ah, good catch about the `.rpmmacros` file. The end of it contains

```rpmspec

#%__arch_install_post   /usr/lib/rpm/check-rpaths   /usr/lib/rpm/check-buildroot

%__arch_install_post \
    [ "%{buildarch}" = "noarch" ] || QA_CHECK_RPATHS=1 ; \
    case "${QA_CHECK_RPATHS:-}" in [1yY]*) /usr/lib/rpm/check-rpaths ;; esac \
    /usr/lib/rpm/check-buildroot
```

My `.rpmmacros` file was created manually and has a long history from several machine installations, to the point I was actually sort of expecting to find some old `.rpmmacros,v` RCS instances, although I only found one `.rpmmacros.old` file dated 2014 with the `%__arch_install_post ` commented version active.

I am not quite sure of my source for that version of `%__arch_install_post`, but yes it is extremely likely that the origin is [rpmdevtools](https://github.com/joequant/rpmdevtools/blob/39d36d677ee40b220d2c7cb84c60ef0067193bec/rpmdev-setuptree#L60-L63).

--------

So that clearly identifies the problem and cause. The question is then how to handle this, fail early like this?

```shell
if grep -q 'QA_CHECK_RPATHS=1' $HOME/.rpmmacros 2>/dev/null
then
    echo "Your $HOME/.rpmmacros file will explicitly call check-rpaths which will fail. You need to (temporarily) disable that. " 1>&2
    exit 1
fi
```

Or attempt to set `QA_RPATHS`?

elguero, 2022-01-01:
I don't think there is anything to fix here on our side. That is outside the scope of this project. We cannot control what others have put into their environment.

The goal of this project is to provide built RPMs for people to avoid them having to setup build environments.

The real fix to this issue would be to convince Displaylink to build their proprietary Displaylink Manager without `rpaths`.