displaylink-rpm/displaylink-rpm #182: Error when installing displaylink RPM on fedora35

agun76, 2021-12-08:
**Please note:** The only working combination is *DisplayLink 5.3.1* and *evdi v1.7.2* OR *DisplayLink 5.4* and *evdi v1.9.1*

  * Are you using the latest DisplayLink driver?
    
  * Are you using the latest EVDI module?
    
  * If you are using a DisplayLink device, have you checked 'troubleshooting'
    on DisplayLink's website?
    
  * Is this issue related to evdi/kernel?
    
  * Is this issue related to DisplayLinkManager?
    
  * If you are having an issue upgrading, have you tried to do a clean install of the driver?
    

  * Fedora or CentOS
  * Version
  * Kernel version `(uname -a)`

  * `/var/log/displaylink/displaylink.log`
  * `/var/lib/dkms/evdi/kernel-[version]/log/make.log`

System : Fedore35
Kernel : Linux fedora 5.15.6-200.fc35.x86_64 #1 SMP Wed Dec 1 13:41:10 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
Log files :
Displaylink.log :
Creating symlink /var/lib/dkms/evdi/1.9.1/source -> /usr/src/evdi-1.9.1

Building module:
cleaning build area...
make -j8 KERNELRELEASE=5.15.6-200.fc35.x86_64 all INCLUDEDIR=/lib/modules/5.15.6-200.fc35.x86_64/build/include KVERSION=5.15.6-200.fc35.x86_64 DKMS_BUILD=1...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.15.6-200.fc35.x86_64 (x86_64)
Consult /var/lib/dkms/evdi/1.9.1/build/make.log for more information.

Building module:
cleaning build area...
make -j8 KERNELRELEASE=5.15.6-200.fc35.x86_64 all INCLUDEDIR=/lib/modules/5.15.6-200.fc35.x86_64/build/include KVERSION=5.15.6-200.fc35.x86_64 DKMS_BUILD=1...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.15.6-200.fc35.x86_64 (x86_64)
Consult /var/lib/dkms/evdi/1.9.1/build/make.log for more information.
Deleting module evdi-1.9.1 completely from the DKMS tree.
Creating symlink /var/lib/dkms/evdi/1.9.1/source -> /usr/src/evdi-1.9.1

Building module:
cleaning build area...
make -j8 KERNELRELEASE=5.15.6-200.fc35.x86_64 all INCLUDEDIR=/lib/modules/5.15.6-200.fc35.x86_64/build/include KVERSION=5.15.6-200.fc35.x86_64 DKMS_BUILD=1...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.15.6-200.fc35.x86_64 (x86_64)
Consult /var/lib/dkms/evdi/1.9.1/build/make.log for more information.

Building module:
cleaning build area...
make -j8 KERNELRELEASE=5.15.6-200.fc35.x86_64 all INCLUDEDIR=/lib/modules/5.15.6-200.fc35.x86_64/build/include KVERSION=5.15.6-200.fc35.x86_64 DKMS_BUILD=1...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.15.6-200.fc35.x86_64 (x86_64)
Consult /var/lib/dkms/evdi/1.9.1/build/make.log for more information.
Deleting module evdi-1.9.1 completely from the DKMS tree.
Creating symlink /var/lib/dkms/evdi/1.9.1/source -> /usr/src/evdi-1.9.1

make.log:
[make.log](https://github.com/displaylink-rpm/displaylink-rpm/files/7675196/make.log)

I try to install displaylink for my docking (wavlink), but unfortunately amazon says it doesn't support linux. Found it on google but no luck when installing the driver.

appreciate for any help, many thanks

ssaavedra, 2021-12-17:
Problem is upstream: https://github.com/DisplayLink/evdi/issues/308

Your logfile complains because:
```
/var/lib/dkms/evdi/1.9.1/build/evdi_drm_drv.h:23:10: fatal error: drm/drm_irq.h: No such file or directory
   23 | #include <drm/drm_irq.h>
      |          ^~~~~~~~~~~~~~~
```

This is because in Kernels 5.14+ there is no `drm/drm_irq.h` file anymore. This requires upstream changes. Users in the linked thread had varying degrees of success by altering themselves the evdi version to be built.

agun76, 2021-12-17:
Problem solved after execute steps provided in : https://github.com/DisplayLink/evdi/issues/308#issuecomment-985428747 

many thanks, case closed

dave-fl, 2021-12-20:
There doesn't seem to be any reason to use the fork mentioned in the above comment.  The current DEV branch has changes from December that seem to work just fine with the new Kernel.

viraptor, 2021-12-26:
I'm trying to use the dev branch, but that doesn't seem to work well. I've created the package via `make rawhide`, getting `displaylink-1.9.1-2.rawhide.20211221T070835Z.x86_64`. However that does not solve the missing `drm_irq` issue:
```
$ rpm -qa | grep displaylink
displaylink-1.9.1-2.rawhide.20211221T070835Z.x86_64
$ ack drm_irq /usr/src/evdi-1.9.1
/usr/src/evdi-1.9.1/evdi_drm_drv.h
23:#include <drm/drm_irq.h>
```
Is there something else to change in the build scripts to use the working dev checkout these days?

dave-fl, 2021-12-27:
@viraptor I am not building past the commits from December 2nd - commit https://github.com/DisplayLink/evdi/commit/d6b28414a4ceb41a904077318b48fa8a7d8981d1, I also didn't build using rawhide, just doing the manual build.

I believe commit https://github.com/DisplayLink/evdi/commit/748aff9cdcb0496d42b3ff2fd7433e74b893686c broke things and https://github.com/DisplayLink/evdi/commit/288f4b62b5ee2ed9b4430c32a875da85d03c1a24 fixes them.  But the latter is a 1.10.x release which is why I am staying on https://github.com/DisplayLink/evdi/commit/d6b28414a4ceb41a904077318b48fa8a7d8981d1 for now.