displaylink-rpm/displaylink-rpm #183: New version of Displaylink driver - v5.5

cdonatom, 2021-12-26:
Hi there,

It seems there's a new driver available from synaptics to support Kernel 5.15 plus other enhancements.
https://www.synaptics.com/products/displaylink-graphics/downloads/ubuntu

Could you please update this repo to support it?

Thank you so much!

elguero, 2021-12-27:
Thank you for the notice and we understand the desire to have the latest driver installed.  One thing to note is that this is a beta release and not the final release for v5.5.

We are waiting for Displaylink (Synaptics) to share their plans on how they are going to be doing releases. It is tough to have a game plan around how to support the releases without having the exact details that they mention will be forthcoming.  Please see https://github.com/DisplayLink/evdi/issues/309 for the discussion that has been had so far.

tominglis, 2022-03-15:
The final version of 5.5 came out at the beginning of this month.

It would be great to get support for it on Fedora!

elguero, 2022-03-15:
Hi!  Yes, it just came out last week.  Yesterday, I committed the changes needed into the main branch. Will be working on a release.

Thanks!

closed 2022-03-21 by elguero