displaylink-rpm/displaylink-rpm #193: question for the latest v5.4.1 rpm release

laurapoggio-sptools, 2022-01-04:
*DisplayLink 5.4* and *evdi v1.9.1*

Which kernels does the [latest v5.4.1 release](https://github.com/displaylink-rpm/displaylink-rpm/releases/download/v5.4.1/fedora-35-displaylink-1.9.1-2.x86_64.rpm) support?

Thanks

elguero, 2022-01-05:
Looking at the Displaylink release notes, it is up to version kernel 5.13.

> C.  Bugs fixed in this release
> ==============================
> 
> C.2  Changes in 5.4.1 release
> -----------------------------
> - Added support for kernel 5.13
> - Added beta support for kernel 5.14 beta (33194)
> - DL-6xxx series: improved 5120x1440 - 5K Super Ultrawide resolution selection (33186)
> - Improved performance: added support for damage regions interface from Linux DRM interface (26378
> - Improved performance in Wayland session with AMD GPU machines (33151)
> - DL-6xxx series: fixed blinking mouse cursor while moving across displays (32987)
> 
> 
> C.1  Changes in 5.4.0 release
> -----------------------------
> - Added support for kernel 5.11 and 5.12
> - DL-6xxx series: added support for nVidia G-SYNC monitors
> - DL-6xxx series: added 5K Ultrawide (5120x1440 @ 60 Hz) display support
> - Other reliability improvements

Since you are linking to F35, I have a feeling you are now on kernel 5.15?  There is a beta release of Displaylink.

> C.  Bugs fixed in this release
> ==============================
> 
> C.3  Changes in 5.5 release
> -----------------------------
> - Added support for kernel 5.14 and 5.15
> - Added preliminary support for kernel 5.16
> - Community fixes for evdi kernel module
> - DL-6xxx series: adding RTL8211FD-VX and RTL8211E Ethernet PHYs (WS1597)
> - DL-41xx: improved firmware flashing time, visible from following update onwards

I have been working on testing the changes for the v5.5 release so that we will be ready as soon as it is released.  I haven't committed those changes here yet since I want to do a bit more testing.

laurapoggio-sptools, 2022-01-07:
Thanks you very much for the answer and to point out where to check!

I am still on F34, Kernel is 5.12 and it is not automatically updated with dnf to avoid issues with displaylinks. I would like to upgrade to F35 but I will wait until the 5.5 version is released here.

Thanks

bmreading, 2022-02-27:
@laurapoggio-sptools 

Fedora 35 is still on 5.4.1, but I believe you can use the faster-tracking negativo17 multimedia repo until the official Fedora repo catches up
```
dnf config-manager --add-repo=https://negativo17.org/repos/fedora-multimedia.repo && dnf install -y displaylink-5.5.0-1.fc35.x86_64
```
Be aware that this repo can clash with RPMFusion repos, so if you are using or plan to use those, then you may simply want to manually download and install the following packages from [there](https://negativo17.org/repos/multimedia/fedora-35/x86_64) instead:

1. libevdi-1.10.0-1.20220104gitaef6790.fc35.x86_64.rpm
2. akmod-evdi-1.10.0-1.20220104gitaef6790.fc35.x86_64.rpm
3. displaylink-5.5.0-1.fc35.x86_64.rpm

elguero, 2022-03-21:
The latest release has support for kernels 5.14, 5.15, 5.16 and preliminary support for 5.17.

closed 2022-03-21 by elguero