displaylink-rpm/displaylink-rpm #194: Error: Failed build dependencies

mehs2690, 2022-01-05:
Hi!

I have a problem when exceute the `make` command, so, I have the last version in the master branch and I downloaded as ZIP, the logs are following:

make
wget -O DisplayLink\ USB\ Graphics\ Software\ for\ Ubuntu\ 5.4.1-55.174.zip \
	"https://www.synaptics.com/sites/default/files/exe_files/2021-09/DisplayLink USB Graphics Software for Ubuntu5.4.1-EXE.zip"
--2022-01-04 22:13:33--  https://www.synaptics.com/sites/default/files/exe_files/2021-09/DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu5.4.1-EXE.zip
Resolving www.synaptics.com (www.synaptics.com)... 34.214.248.103, 52.41.97.53
Connecting to www.synaptics.com (www.synaptics.com)|34.214.248.103|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 14500327 (14M) [application/zip]
Saving to: ‘DisplayLink USB Graphics Software for Ubuntu 5.4.1-55.174.zip’

DisplayLink USB Graphics Software for Ubuntu 5.4.1-55.174.z 100%[==========================================================================================================================================>]  13.83M  4.79MB/s    in 2.9s    

2022-01-04 22:13:36 (4.79 MB/s) - ‘DisplayLink USB Graphics Software for Ubuntu 5.4.1-55.174.zip’ saved [14500327/14500327]

rpmbuild -bb --define "_topdir `pwd`" --define "_sourcedir `pwd`" --define "_rpmdir `pwd`" --define "_specdir `pwd`" --define "_srcrpmdir `pwd`" --define "_buildrootdir `mktemp -d /var/tmp/displayportXXXXXX`" --define "_builddir `mktemp -d /var/tmp/displayportXXXXXX`" --define "_release 3" --define "_daemon_version 5.4.1-55.174" --define "_version 1.9.1" --define "_tmppath `mktemp -d /var/tmp/displayportXXXXXX`"  displaylink.spec --target=i386
Building target platforms: i386
Building for target i386
setting SOURCE_DATE_EPOCH=1640736000
error: Failed build dependencies:
	libdrm-devel is needed by displaylink-1.9.1-3.i386
make: *** [Makefile:152: i386/displaylink-1.9.1-3.i386.rpm] Error 11

Does anyone have any idea how to solve such an error?

elguero, 2022-01-05:
> error: Failed build dependencies:
libdrm-devel is needed by displaylink-1.9.1-3.i386

You are missing this package. You can install by running:

`dnf install libdrm-devel`

Do the RPM packages we release not work for you?

mehs2690, 2022-01-10:
Thank you very much, this worked for me, but, I have other error, When I run the following commands:

`sudo modinfo -n evdi /lib/modules/5.10.19-200.fc33.x86_64/extra/evdi.ko.xz`

I get this error: 

```sh
modinfo: ERROR: Module evdi not found.
modinfo: ERROR: Module /lib/modules/5.0.10-200.fc29.x86_64/extra/evdi.ko.xz not found.
```

previously, I had done the Secure boot on Fedora step and the register the MOK with secure boot step but I can't go ahead for this error. Do you have any idea about this error?

I thank you very much for the help and attention

mehs2690, 2022-01-12:
When execute this command `sudo rpm -qa | grep displaylink` I get nothing

It's a weird situation because I was checking the installed rpm's but I can not find anything with "displaylink". I think the rpm failed its installation and I will check what happened. My kernel version is 5.0.10-200.fc29.x86_64 but I'm using Fedora 35, its weird too...

elguero, 2022-07-22:
There is a new RPM that has been released.  I would recommend clearing out any RPMs and any manually installed drivers.  Then give the latest RPM a try.  Please provide some feedback as to whether we can close out this issue.  Thanks.

elguero, 2022-08-12:
Closing for now.

closed 2022-08-12 by elguero