displaylink-rpm/displaylink-rpm #197: rocky-8-displaylink-1.9.1-2.x86_64.rpm fails post-install (evdi compile) on AlmaLInux/EL 8

DrLove73, 2022-01-19:
Hi everyone. 
I have fully updated AlmaLinux 8.5 system but I also have some desktop-focused Fedora29 packages added, but not that changed original files or were installed instead of some original file. If needed, I can produce entire list of packages installed from fedora29 repositories.
Install of evdi modules fails somehow, make.log provides answers.
I also had kernel-ml 5.16 from ElRepo installed but when compilation first failed I rebooted to stock 4.18 kernel and deinstalled kernel-ml kernels, but compilation still fails.

  * Fedora / CentOS / Rocky Linux : AlmaLinux
  * Version 8.5
  * Kernel version `(uname -a)`: 4.18.0-348.7.1.el8_5.x86_64

  * `/var/log/displaylink/displaylink.log`
  * `/var/lib/dkms/evdi/kernel-[version]/log/make.log`
[displaylink-2.log](https://github.com/displaylink-rpm/displaylink-rpm/files/7899466/displaylink-2.log)
[make.log](https://github.com/displaylink-rpm/displaylink-rpm/files/7899467/make.log)

dnf install rocky-8-displaylink-1.9.1-2.x86_64.rpm

Something goes wrong with compilation of evdi module and "some warnings being treated as errors" and module is not compiled.

Complete output of dnf install command:
```
[root@localhost ~]# dnf install /extra/dodaci/DisplayLink/rocky-8-displaylink-1.9.1-2.x86_64.rpm 
Last metadata expiration check: 3:25:43 ago on Wed 19 Jan 2022 03:16:25 PM CET.
Dependencies resolved.
======================================================================================================================================================================================
 Package                                      Architecture                            Version                                     Repository                                     Size
======================================================================================================================================================================================
Installing:
 displaylink                                  x86_64                                  1.9.1-2                                     @commandline                                  5.1 M

Transaction Summary
======================================================================================================================================================================================
Install  1 Package

Total size: 5.1 M
Installed size: 13 M
Is this ok [y/N]: y
Downloading Packages:
Running transaction check
Transaction check succeeded.
Running transaction test
Transaction test succeeded.
Running transaction
  Preparing        :                                                                                                                                                              1/1 
  Installing       : displaylink-1.9.1-2.x86_64                                                                                                                                   1/1 
  Running scriptlet: displaylink-1.9.1-2.x86_64                                                                                                                                   1/1 
Job for displaylink.service failed because the control process exited with error code.
See "systemctl status displaylink.service" and "journalctl -xe" for details.
warning: %post(displaylink-1.9.1-2.x86_64) scriptlet failed, exit status 1

Error in POSTIN scriptlet in rpm package displaylink
/sbin/ldconfig: /etc/ld.so.conf.d/kernel-ml-5.15.12-1.el8.elrepo.x86_64.conf:6: hwcap directive ignored
/sbin/ldconfig: /etc/ld.so.conf.d/kernel-ml-5.16.0-1.el8.elrepo.x86_64.conf:6: hwcap directive ignored
/sbin/ldconfig: /etc/ld.so.conf.d/kernel-ml-5.16.1-1.el8.elrepo.x86_64.conf:6: hwcap directive ignored

  Verifying        : displaylink-1.9.1-2.x86_64                                                                                                                                   1/1 
Installed products updated.

Installed:
  displaylink-1.9.1-2.x86_64                                                                                                                                                          

Complete!
````

elguero, 2022-01-19:
There were some patches that I created for evdi 1.9.1 to work with the latest kernel on EL8 distros.  I just created a release which will contain those patches.

Could you give the latest Rocky package a try?

DrLove73, 2022-01-19:
I logged out of root and started packing laptop jast 10 minutes before your comment :-)

New version installed without issues, I now only have to figure out how to start the damned thing. 
I use this device: Bus 003 Device 008: ID 534d:6021 MacroSilicon VGA Display Adapter
with MATE , will it recognize it?
And do I need to make some configs?

DrLove73, 2022-01-20:
To make things clear:
Installation went good, displaylink service is active but xrandr or MATE (1.26 on X11) Display Settings does not see additional VGA.

elguero, 2022-03-21:
Were you able to figure this out?  There is a new release as well.

DrLove73, 2022-03-21:
I went and bought 120 eur USB-c hub/dock (LAN, PD, HDMI, VGA, several USB 3.1) and it works without any additional drivers, I even moved LAN (office IP is now on separate dock-NIC so no need to select proper connection/IP) and connecting USB sticks to it, so I am golden.
I do not think I will bother with this anymore, Windows works without problem with it and only Linux laptop around is mine.

closed 2022-03-21 by elguero