displaylink-rpm/displaylink-rpm #198: Driver Fedora 35

mr-nightmarelive, 2022-02-13:
More of a question than an issue, but I have a VisionTek dock that uses display link drivers. Worked with Windows and Ubuntu but can't get the dock to work with Fedora. Any thoughts?

ssaavedra, 2022-03-18:
Without more information on the issue, what you tried, what you haven't, what do the logs say... it's difficult for anyone on the internet to help you.

azampatti, 2022-03-19:
I can't get this working on Fedora 35 (5.16.7-200).... Does somebody have it working already? I tried everything I was able to read/attempt.

Maybe a "for dummies" guide would help me :)

mr-nightmarelive, 2022-03-20:
I would have to reinstall Fedora to get the logs to post on here. I had to switch back to Ubuntu in order to use my dock at work. I have been using Ubuntu for about five years and decided to try Fedora. As far as I can remember I installed the GitHub release and was able to use the USB ports on my dock, but the screens did not register. Unfortunately at this moment I am having an issue remembering anything else and have to install Fedora to get a better answer. I will have to install on Monday and get a better answer.

PeaaZee, 2022-03-27:
I have it running now on Fedora 35. Here is what I did.

I did a fresh install after running my head against a wall installing displaylink drivers on OpenSuse Tumbleweed. I would much prefer to use it, but Displaylink drivers a must for me.

After the fresh install I updated the system using
`sudo dnf update`

After rebooting into the new kernel version 5.16, i installed kernel-headers using
`sudo dnf install kernel-headers`
I'm a noob myself and not 100% sure, if this is needed or already installed, I just read about it in a different issue.

I then installed the dynamic kernel modules system using
`sudo dnf install dkms`

I then did a reboot, not knowing if it was necessary.

Afterwards I downloaded the most recent `displaylink-rpm`-package from your github and installed it using
```
cd ~/Downloads
sudo dnf install fedora-35-displaylink-1.10.1-1.x86_64.rpm
```

I checked if everything was running using
`sudo modprobe evdi` which did not throw an error (as I learnt no error is everything you get if it works)
and checking
`systemctl status displaylink-driver.service`, giving me the following output:
```
● displaylink-driver.service - DisplayLink Driver Service
     Loaded: loaded (/usr/lib/systemd/system/displaylink-driver.service; static)
     Active: active (running) since Sun 2022-03-27 12:58:29 CEST; 35min ago
    Process: 1108 ExecStartPre=/sbin/modprobe evdi (code=exited, status=0/SUCCESS)
   Main PID: 1114 (DisplayLinkMana)
      Tasks: 52 (limit: 9318)
     Memory: 120.0M
        CPU: 14min 18.390s
     CGroup: /system.slice/displaylink-driver.service
             └─1114 /usr/libexec/displaylink/DisplayLinkManager
```

I then connected my docking station and it worked.

Upon another reboot ig got the follwing error code: (roughly translated from german to english)

```
Unexpected system error
the system detected an error and recovered itself

name: kernel-core
version: 5.16.16-200.fc35.x86_64
first detected: 16 seconds ago
reported: cant be reported

Es ist ein Kernel-Problem aufgetreten, aber Ihr Kernel ist defekt (flags:GOE). Erklärung: ---> a kernel problem occured, but your kernel is defective (flags: GOE). Explanation:
O - Out-of-tree module has been loaded.
E - Unsigned module has been loaded.
 Kernel-Maintainer können keine Berichte beschädigter Kernel analysieren. Betroffene Module: evdi. ---> kernel maintainers cant analyze reports of defective kernels. Affected module: evdi
```

But (almost) everything works as it should. The only thing I noticed is a graphical glitch only on the main laptop display. When I rightclick the desktop, I can see a complete menu, as soon as I hover over it, all entries that I didn't hover over yet, disappear.

I hope this description helps somebody.

mr-nightmarelive, 2022-03-28:
@PeaaZee Thanks friend I have been fighting with this for a while and just got it working with your help.

s3612985, 2022-03-30:
I attempted to follow those instructions but keep getting this issue when installing the displaylink rpm file.
Running Fedora 35.  Linux envy 5.16.18-200.fc35.x86_64 #1 SMP PREEMPT Mon Mar 28 14:10:07 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux.

"Job for displaylink-driver.service failed because the control process exited with error code.
See "systemctl status displaylink-driver.service" and "journalctl -xeu displaylink-driver.service" for details.
warning: %post(displaylink-1.10.1-1.x86_64) scriptlet failed, exit status 1

Error in POSTIN scriptlet in rpm package displaylink"

Here is the full install dump.

sudo dnf install displaylink-1.10.1-1.x86_64.rpm 

Last metadata expiration check: 0:27:04 ago on Wed 30 Mar 2022 22:02:39.
Dependencies resolved.
========================================================================================================================================
 Package                           Architecture                 Version                        Repository                          Size
========================================================================================================================================
Installing:
 displaylink                       x86_64                       1.10.1-1                       @commandline                       5.1 M

Transaction Summary
========================================================================================================================================
Install  1 Package

Total size: 5.1 M
Installed size: 13 M
Is this ok [y/N]: y
Downloading Packages:
Running transaction check
Transaction check succeeded.
Running transaction test
Transaction test succeeded.
Running transaction
  Preparing        :                                                                                                                1/1 
  Installing       : displaylink-1.10.1-1.x86_64                                                                                    1/1 
  Running scriptlet: displaylink-1.10.1-1.x86_64                                                                                    1/1 
Job for displaylink-driver.service failed because the control process exited with error code.
See "systemctl status displaylink-driver.service" and "journalctl -xeu displaylink-driver.service" for details.
warning: %post(displaylink-1.10.1-1.x86_64) scriptlet failed, exit status 1

Error in POSTIN scriptlet in rpm package displaylink
  Verifying        : displaylink-1.10.1-1.x86_64                                                                                    1/1 

Installed:
  displaylink-1.10.1-1.x86_64                                                                                                           

Complete!
-----

This is the output when I systemctl status. 
Did a reboot, but still can't get the second screen up and running :(

sudo systemctl status displaylink-driver.service
● displaylink-driver.service - DisplayLink Driver Service
     Loaded: loaded (/usr/lib/systemd/system/displaylink-driver.service; static)
     Active: activating (auto-restart) (Result: exit-code) since Wed 2022-03-30 22:36:36 AEST; 427ms ago
    Process: 13398 ExecStartPre=/sbin/modprobe evdi (code=exited, status=1/FAILURE)
        CPU: 26ms

mr-nightmarelive, 2022-03-30:
@s3612985 Looking at the command for installing you posted, try running `sudo dnf install fedora-35-displaylink-1.10.1.x86_64.rpm` in the downloads folder. Might be an argument issue with the file name.

azampatti, 2022-05-09:
Long time getting back here but I figure I wanted to get this running for once :)

So I installed the latest RPM which installs smoothly but I don't get any display out, even after unplugging/plugging the USB Dock.

Service starts normally:
`[mudo@fedora Downloads]$ systemctl status displaylink-driver.service
● displaylink-driver.service - DisplayLink Driver Service
     Loaded: loaded (/usr/lib/systemd/system/displaylink-driver.service; static)
     Active: active (running) since Mon 2022-05-09 17:55:52 CDT; 9s ago
   Main PID: 7934 (DisplayLinkMana)
      Tasks: 14 (limit: 47344)
     Memory: 1.9M
        CPU: 16ms
     CGroup: /system.slice/displaylink-driver.service
             └─7934 /usr/libexec/displaylink/DisplayLinkManager

May 09 17:55:52 fedora systemd[1]: Starting DisplayLink Driver Service...
May 09 17:55:52 fedora systemd[1]: Started DisplayLink Driver Service.
`

Any pointers? anything where I can look at or try? I'm using Fedora 35 with X11 (not Wayland)

mr-nightmarelive, 2022-05-18:
One thing I noticed when I tried was switching between wayland and X11

elguero, 2022-07-22:
Can you try the latest released RPM and provide some feedback?  Thanks

mr-nightmarelive, 2022-07-22:
I will be able to when I am back at my office later today.

elguero, 2022-08-12:
Just checking in to see if there are any updates.  Thanks.

mr-nightmarelive, 2022-08-12:
Not just yet. I had to tear down my multi monitor system at work. Will update as soon as I can.

mr-nightmarelive, 2022-08-30:
I am unable to test currently. My docking station seems to have been misplaced when I got a new PC.

elguero, 2022-08-30:
No worries. Let's close this ticket for now and then when you have a chance to test, you can open a new ticket if it is still an issue.  Does that sound good?

mr-nightmarelive, 2022-08-30:
Sound good to me! I will reach out again when I can.

closed 2022-08-30 by mr-nightmarelive