displaylink-rpm/displaylink-rpm #200: New release 5.5 breaking udev

cdonatom, 2022-03-15:
Hi there,

I just installed the latest version of the driver on Fedora 35, kernel 5.16.13-200.fc35.x86_64 and now the system takes ages to boot.

The output of udev is the following:

`mar 15 12:08:07 cdonato-laptop systemd-udevd[33580]: 4-1.2:1.3: Process '/usr/libexec/displaylink/udev.sh /dev ' failed with exit code 2.
mar 15 12:08:07 cdonato-laptop systemd-udevd[33568]: 4-1.2:1.0: Process '/usr/libexec/displaylink/udev.sh /dev ' failed with exit code 2.
mar 15 12:08:07 cdonato-laptop systemd-udevd[33567]: 4-1.2:1.1: Process '/usr/libexec/displaylink/udev.sh /dev ' failed with exit code 2.
mar 15 12:08:07 cdonato-laptop systemd-udevd[33581]: 4-1.2:1.4: Process '/usr/libexec/displaylink/udev.sh /dev ' failed with exit code 2.
mar 15 12:08:07 cdonato-laptop systemd-udevd[33564]: 4-1.2:1.2: Process '/usr/libexec/displaylink/udev.sh /dev ' failed with exit code 2.
mar 15 12:08:07 cdonato-laptop systemd-udevd[33564]: 4-1.2: Process '/usr/libexec/displaylink/udev.sh /dev /dev/bus/usb/004/016' failed with exit code 2.
`
Not sure whether this is an issue due to the new driver (Displaylink side) or something related to the RPM package creation to be honest. Not sure how to help here.

Thanks!

elguero, 2022-03-15:
Hi! Thanks for testing.

Can you post the contents of `/usr/libexec/displaylink/udev.sh`?

That was something new with the DisplayLink release.

cdonatom, 2022-03-15:
Sure.
`❯ cat /usr/libexec/displaylink/udev.sh
#!/usr/bin/bash
# Copyright (c) 2016 - 2020 DisplayLink (UK) Ltd.
# File autogenerated by udev-installer.sh script

get_displaylink_dev_count()
{
   cat /sys/bus/usb/devices/*/idVendor | grep 17e9 | wc -l
}

get_displaylink_symlink_count()
{
  root=$1

  if [ ! -d "$root/displaylink/by-id" ]; then
    echo "0"
    return
  fi

  for f in $(find $root/displaylink/by-id -type l -exec realpath {} \; 2> /dev/null); do
    test -c $f && echo $f;
  done | wc -l
}

start_displaylink()
{
  if [ "$(get_displaylink_dev_count)" != "0" ]; then
    start_service
  fi
}

stop_displaylink()
{
  root=$1

  if [ "$(get_displaylink_symlink_count $root)" = "0" ]; then
    stop_service
  fi
}

remove_dldir_if_empty()
{
  root=$1
  (cd $root; rmdir -p --ignore-fail-on-non-empty displaylink/by-id)
}

create_displaylink_symlink()
{
  root=$1
  device_id=$2
  devnode=$3

  mkdir -p $root/displaylink/by-id
  ln -sf $devnode $root/displaylink/by-id/$device_id
}

unlink_displaylink_symlink()
{
   root=$1
   devname=$2

   for f in $root/displaylink/by-id/*; do
     if [ ! -e "$f" ] || ([ -L "$f" ] && [ "$f" -ef "$devname" ]); then
       unlink "$f"
     fi
   done
   (cd $root; rmdir -p --ignore-fail-on-non-empty displaylink/by-id)
}

prune_broken_links()
{
  root=$1

  dir="$root/displaylink/by-id"
  find -L "$dir" -name "$dir" -o type d -prune -o -type -l -exec rm {} +
  remove_dldir_if_empty $root
}

main()
{
  action=$1
  root=$2
  devnode=$4

  if [ "$action" = "add" ]; then
    device_id=$3
    create_displaylink_symlink $root $device_id $devnode
    start_displaylink
  elif [ "$action" = "remove" ]; then
      devname=$3
      unlink_displaylink_symlink "$root" "$devname"
      stop_displaylink "$root"
  elif [ "$action" = "START" ]; then
    start_displaylink
  fi
}

start_service()
{
  systemctl start displaylink-driver
}

stop_service()
{
  systemctl stop displaylink-driver
}

if [ "$ACTION" = "add" ] && [ "$#" -ge 3 ]; then
  main $ACTION $1 $2 $3
  return 0
fi

if  [ "$ACTION" = "remove" ]; then
  if [ "$#" -ge 2 ]; then
    main $ACTION $1 $2 $3
    return 0
  else
    prune_broken_links $root
    return 0
  fi
fi
`

elguero, 2022-03-21:
Okay. I was wanting to make sure that the script was extracted properly.

Is the delay happening on every boot?

What does `lsusb` give you?

cdonatom, 2022-03-21:
Yes, it was happening at every boot. This is the lsusb output (filter out those that are not related to this issue):
`Bus 004 Device 004: ID 17ef:a359 Lenovo ThinkPad Lan
Bus 004 Device 003: ID 17e9:6015 DisplayLink ThinkPad Hybrid USB-C with USB-A Dock`

GrzegorzBB, 2022-03-21:
Hi,

small input from my side:
`rpm -qa |grep displaylink
displaylink-1.10.1-1.x86_64`

`uname -a
Linux hp840 5.16.15-201.fc35.x86_64 #1 SMP PREEMPT Thu Mar 17 05:45:13 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux`

`lsusb
Bus 004 Device 003: ID 17e9:6006 DisplayLink Dell Universal Dock D6000
Bus 001 Device 014: ID 06c4:c411 Bizlink International Corp. D6000 Controller

`

`Mar 21 19:01:08 hp840 systemd-udevd[993]: 4-1.1:1.0: Spawned process '/usr/libexec/displaylink/udev.sh /dev usb-004-003-DisplayLink_Dell_Universal_Dock_D6000_1909028749 /dev/bus/usb/004/003' [1032] is taking longer than 59s to complete
Mar 21 19:01:08 hp840 systemd-udevd[963]: 4-1.1:1.0: Worker [993] processing SEQNUM=3659 is taking a long time
Mar 21 19:02:07 hp840 systemd[1]: systemd-udev-settle.service: Main process exited, code=exited, status=1/FAILURE
Mar 21 19:02:07 hp840 systemd[1]: systemd-udev-settle.service: Failed with result 'exit-code'.
Mar 21 19:02:07 hp840 systemd[1]: Failed to start Wait for udev To Complete Device Initialization.
`
`journalctl -u systemd-udev-settle.service
Mar 21 19:00:07 hp840 systemd[1]: Starting Wait for udev To Complete Device Initialization...
Mar 21 19:00:08 hp840 udevadm[981]: systemd-udev-settle.service is deprecated. Please fix multipathd.service not to pull it in.
Mar 21 19:02:07 hp840 systemd[1]: systemd-udev-settle.service: Main process exited, code=exited, status=1/FAILURE
Mar 21 19:02:07 hp840 systemd[1]: systemd-udev-settle.service: Failed with result 'exit-code'.
Mar 21 19:02:07 hp840 systemd[1]: Failed to start Wait for udev To Complete Device Initialization.
Mar 21 19:02:07 hp840 systemd[1]: systemd-udev-settle.service: Consumed 4.184s CPU time.
`

Udev script is also in place:
`[grzegorz@hp840 ~]$ ls -al /usr/libexec/displaylink/udev.sh
-rwxr-xr-x. 1 root root 2140 Mar 21 14:48 /usr/libexec/displaylink/udev.sh
`

laurapoggio-sptools, 2022-03-30:
I had the same problem as described [above](https://github.com/displaylink-rpm/displaylink-rpm/issues/200#issuecomment-1074410114). If installing as suggested [here](https://github.com/displaylink-rpm/displaylink-rpm/issues/193#issuecomment-1053687146) I do not have the boot problem.

cdonatom, 2022-04-19:
it definitely worked! thanks!

tominglis, 2022-05-02:
I installed the 5.5.0 rpm package from this repo, and I have the behaviour described above.

I get "A start job is running for udev wait for Complete Device Initialization" and it takes an extra 90 seconds to boot my laptop.

I have a Dell Latitude 5300 with Fedora Workstation 35.

swm123456, 2022-05-31:
I also experience slow boot on Fedora 35 (takes about 2 minutes). 
`systemd-analyze` tells me that `systemd-udev-settle.service` is causing the slowness.
After some googling:
[](https://github.com/NixOS/nixpkgs/issues/51684)
[](https://support.displaylink.com/forums/287786-displaylink-feature-suggestions/suggestions/41424121-avoid-boot-time-stall-due-to-udev-rules)
I fixed it by adding `--no-block` to `systemctl start displaylink-driver` in udev.sh.

elguero, 2022-07-22:
In the latest release from Displaylink, they made changes to the udev rules.  Can you try the latest release of the rpm and provide feedback?  Thanks.

tominglis, 2022-07-25:
I tried installing the latest build from this repo on Fedora 35 and this issue is still present.

I fix it by going here:

/usr/libexec/displaylink/udev.sh

and adding --no-block to systemctl start displaylink-driver in udev.sh as described above.

commit c7226ca0 (2022-08-14): Add suggestion from community to start displaylink-driver service
with `--no-block`.

Rework el9 patch for v1.12 evdi driver release

Fixes #200
closed 2022-08-14 by elguero