displaylink-rpm/displaylink-rpm #201: Having latest GitHub build installed makes nVidia GPU HDMI port non-functional

Jinux-1, 2022-03-19:
Fedora 35
 Linux fedora 5.16.15-201.fc35.x86_64 #1 SMP PREEMPT Thu Mar 17 05:45:13 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

Displaylink.log
`Creating symlink /var/lib/dkms/evdi/1.10.1/source -> /usr/src/evdi-1.10.1

Building module:
cleaning build area...
make -j16 KERNELRELEASE=5.16.15-201.fc35.x86_64 all INCLUDEDIR=/lib/modules/5.16.15-201.fc35.x86_64/build/include KVERSION=5.16.15-201.fc35.x86_64 DKMS_BUILD=1...
cleaning build area...

evdi.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.16.15-201.fc35.x86_64/extra/
depmod.....
Module evdi-1.10.1 for kernel 5.16.15-201.fc35.x86_64 (x86_64).
Before uninstall, this module version was ACTIVE on this kernel.

evdi.ko.xz:
 - Uninstallation
   - Deleting from: /lib/modules/5.16.15-201.fc35.x86_64/extra/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.
depmod.....
Deleting module evdi-1.10.1 completely from the DKMS tree.`
  * `/var/lib/dkms/evdi/kernel-[version]/log/make.log`
  My dkms folder was empty so I'm not sure what happened to make.log, I did remove the driver after I noticed it wasn't working, but I'm not sure if that had to do with it.

I run a very odd setup so I understand if this never gets fixed lol.
I use an Asus ZenBook Pro Duo, with an HDMI out from the GPU (nVidia RTX 2060M) and VGA out from USB (CPU controlled I;m pretty sure.) I'd also like to have HDMI out through my DisplayLink device (ACP71USZ). When building the latest github release, and installing the x86_64 RPM, the displays reload, and it appears like all the monitors work, however the FPS seems to have halved (I didn't have an FPS counter up, but there was a noticeable difference). I rebooted Fedora and the monitor that is hooked up the HDMI port on the Laptop no longer had any output. Turning it off and on in the GNOME display settings seemed to have made the entire display driver reload, and still had the HDMI out turned off afterwards.

1. Build most recent version of displaylink-rpm from GitHub
2. Install RPM
3. Reboot
4. No more output from HDMI

elguero, 2022-03-21:
```
evdi.ko.xz:

Uninstallation
Deleting from: /lib/modules/5.16.15-201.fc35.x86_64/extra/
Original module
No original module was found for this module on this kernel.
Use the dkms install command to reinstall any previous module version.
depmod.....
Deleting module evdi-1.10.1 completely from the DKMS tree.`
```
```
/var/lib/dkms/evdi/kernel-[version]/log/make.log
My dkms folder was empty so I'm not sure what happened to make.log, I did remove the driver after I noticed it wasn't working, but I'm not sure if that had to do with it
```

If you removed the driver, then yes, this folder would be empty.

There are a lot of reports of "lag" over on the [Displaylink/evdi ](https://github.com/DisplayLink/evdi/issues) repo.

Which driver are you using for nvidia?

Maybe investigate the settings in this file which is installed by the RPM here (https://github.com/displaylink-rpm/displaylink-rpm/blob/master/20-displaylink.conf).

Jinux-1, 2022-03-24:
Updating with more information after reinstalling

displaylink-1.10.1-1.github_evdi.x86_64

displaylink.log 
`Creating symlink /var/lib/dkms/evdi/1.10.1/source -> /usr/src/evdi-1.10.1

Building module:
cleaning build area...
make -j16 KERNELRELEASE=5.16.15-201.fc35.x86_64 all INCLUDEDIR=/lib/modules/5.16.15-201.fc35.x86_64/build/include KVERSION=5.16.15-201.fc35.x86_64 DKMS_BUILD=1...
cleaning build area...

evdi.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.16.15-201.fc35.x86_64/extra/
depmod.....
Module evdi-1.10.1 for kernel 5.16.15-201.fc35.x86_64 (x86_64).
Before uninstall, this module version was ACTIVE on this kernel.

evdi.ko.xz:
 - Uninstallation
   - Deleting from: /lib/modules/5.16.15-201.fc35.x86_64/extra/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.
depmod.....
Deleting module evdi-1.10.1 completely from the DKMS tree.
Creating symlink /var/lib/dkms/evdi/1.10.1/source -> /usr/src/evdi-1.10.1

Building module:
cleaning build area...
make -j16 KERNELRELEASE=5.16.15-201.fc35.x86_64 all INCLUDEDIR=/lib/modules/5.16.15-201.fc35.x86_64/build/include KVERSION=5.16.15-201.fc35.x86_64 DKMS_BUILD=1...
cleaning build area...

evdi.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.16.15-201.fc35.x86_64/extra/
depmod.....`

make.log

`DKMS make.log for evdi-1.10.1 for kernel 5.16.16-200.fc35.x86_64 (x86_64)
Wed Mar 23 10:59:04 PM EDT 2022
make KBUILD_VERBOSE=1 M=/var/lib/dkms/evdi/1.10.1/build SUBDIRS=/var/lib/dkms/evdi/1.10.1/build SRCROOT=/var/lib/dkms/evdi/1.10.1/build CONFIG_MODULE_SIG= -C /lib/modules/5.16.16-200.fc35.x86_64/build modules
make[1]: Entering directory '/usr/src/kernels/5.16.16-200.fc35.x86_64'
test -e include/generated/autoconf.h -a -e include/config/auto.conf || (		\
echo >&2;							\
echo >&2 "  ERROR: Kernel configuration is invalid.";		\
echo >&2 "         include/generated/autoconf.h or include/config/auto.conf are missing.";\
echo >&2 "         Run 'make oldconfig && make prepare' on kernel src to fix it.";	\
echo >&2 ;							\
/bin/false)
make -f ./scripts/Makefile.build obj=/var/lib/dkms/evdi/1.10.1/build \
single-build= \
need-builtin=1 need-modorder=1
  gcc -Wp,-MMD,/var/lib/dkms/evdi/1.10.1/build/.evdi_platform_drv.o.d -nostdinc -I./arch/x86/include -I./arch/x86/include/generated  -I./include -I./arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/compiler-version.h -include ./include/linux/kconfig.h -include ./include/linux/compiler_types.h -D__KERNEL__ -fmacro-prefix-map=./= -Wall -Wundef -Werror=strict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -fshort-wchar -fno-PIE -Werror=implicit-function-declaration -Werror=implicit-int -Werror=return-type -Wno-format-security -std=gnu89 -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -fcf-protection=none -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mtune=generic -mno-red-zone -mcmodel=kernel -Wno-sign-compare -fno-asynchronous-unwind-tables -mindirect-branch=thunk-extern -mindirect-branch-register -fno-jump-tables -fno-delete-null-pointer-checks -Wno-frame-address -Wno-format-truncation -Wno-format-overflow -Wno-address-of-packed-member -O2 -fno-allow-store-data-races -Wframe-larger-than=2048 -fstack-protector -Wimplicit-fallthrough=5 -Wno-main -Wno-unused-but-set-variable -Wno-unused-const-variable -fno-stack-clash-protection -pg -mrecord-mcount -mfentry -DCC_USING_FENTRY -Wdeclaration-after-statement -Wvla -Wno-pointer-sign -Wno-stringop-truncation -Wno-zero-length-bounds -Wno-array-bounds -Wno-stringop-overflow -Wno-restrict -Wno-maybe-uninitialized -Wno-alloc-size-larger-than -fno-strict-overflow -fno-stack-check -fconserve-stack -Werror=date-time -Werror=incompatible-pointer-types -Werror=designated-init -Wno-packed-not-aligned -g -Iinclude/drm  -DMODULE  -DKBUILD_BASENAME='"evdi_platform_drv"' -DKBUILD_MODNAME='"evdi"' -D__KBUILD_MODNAME=kmod_evdi -c -o /var/lib/dkms/evdi/1.10.1/build/evdi_platform_drv.o /var/lib/dkms/evdi/1.10.1/build/evdi_platform_drv.c  ; ./tools/objtool/objtool orc generate  --module  --no-fp   --retpoline  --uaccess  /var/lib/dkms/evdi/1.10.1/build/evdi_platform_drv.o
  gcc -Wp,-MMD,/var/lib/dkms/evdi/1.10.1/build/.evdi_platform_dev.o.d -nostdinc -I./arch/x86/include -I./arch/x86/include/generated  -I./include -I./arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/compiler-version.h -include ./include/linux/kconfig.h -include ./include/linux/compiler_types.h -D__KERNEL__ -fmacro-prefix-map=./= -Wall -Wundef -Werror=strict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -fshort-wchar -fno-PIE -Werror=implicit-function-declaration -Werror=implicit-int -Werror=return-type -Wno-format-security -std=gnu89 -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -fcf-protection=none -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mtune=generic -mno-red-zone -mcmodel=kernel -Wno-sign-compare -fno-asynchronous-unwind-tables -mindirect-branch=thunk-extern -mindirect-br [... cut ...]
Skipping BTF generation for /var/lib/dkms/evdi/1.10.1/build/evdi.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/kernels/5.16.16-200.fc35.x86_64'`

As for the driver I'm using the official nvidia driver from the RPMFusion repo. Version 510.47.03, and CUDA version 11.6

It also seems like for some reason xorg is gone, I'm not sure this is an issue with display link as I have to remove it to test that, so I'll do that after I make this post. It also seems that the monitor functions properly on KDE. On GNOME it doesn't work for some reason.

elguero, 2022-07-22:
Can you try the latest release and provide feedback?

elguero, 2022-08-12:
Closing for now.  If this is still an issue in the latest release, we can attempt to triage.

closed 2022-08-12 by elguero