displaylink-rpm/displaylink-rpm #202: Getting logged off after dock connected

jacekobst1, 2022-04-20:
Hi,

I really appreciate the work you put into maintaining this tool!

Recently I updated my Fedora to version 35, due to new version of displaylink-rpm available. Since then, my docking station with 2 external monitors is working seemingly okay, but every time I connect it to the notebook with already logged in user, the system crash, and over a dozen of seconds I always get logged off. The only way to get it working properly is to connect the cable before logging in. Another bug is that 3-5 times a day my monitors are getting black for several seconds. 

**System**: Fedora
**Version**: 35
**Kernel version**: 5.16.20-200.fc35.x86_64
**Displaylink release**: 5.5.0

**/var/log/displaylink/displaylink.log:**
`Error! DKMS tree already contains: evdi-1.10.1
You cannot add the same module/version combo more than once.
Module evdi/1.10.1 already built for kernel 5.16.20-200.fc35.x86_64 (x86_64).
Module evdi/1.10.1 already installed on kernel 5.16.20-200.fc35.x86_64 (x86_64).
Module evdi-1.10.1 for kernel 5.16.20-200.fc35.x86_64 (x86_64).
Before uninstall, this module version was ACTIVE on this kernel.

evdi.ko.xz:
 - Uninstallation
   - Deleting from: /lib/modules/5.16.20-200.fc35.x86_64/extra/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.
depmod........
Deleting module evdi-1.10.1 completely from the DKMS tree.
Creating symlink /var/lib/dkms/evdi/1.10.1/source -> /usr/src/evdi-1.10.1

Building module:
cleaning build area...
make -j8 KERNELRELEASE=5.16.20-200.fc35.x86_64 all INCLUDEDIR=/lib/modules/5.16.20-200.fc35.x86_64/build/include KVERSION=5.16.20-200.fc35.x86_64 DKMS_BUILD=1....
cleaning build area...

evdi.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.16.20-200.fc35.x86_64/extra/
depmod........`

**/var/lib/dkms/evdi/kernel-[version]/log/make.log:**
`DKMS make.log for evdi-1.10.1 for kernel 5.16.20-200.fc35.x86_64 (x86_64)
śro, 20 kwi 2022, 17:38:12 CEST
make KBUILD_VERBOSE=1 M=/var/lib/dkms/evdi/1.10.1/build SUBDIRS=/var/lib/dkms/evdi/1.10.1/build SRCROOT=/var/lib/dkms/evdi/1.10.1/build CONFIG_MODULE_SIG= -C /lib/modules/5.16.20-200.fc35.x86_64/build modules
make[1]: Entering directory '/usr/src/kernels/5.16.20-200.fc35.x86_64'
test -e include/generated/autoconf.h -a -e include/config/auto.conf || (		\
echo >&2;							\
echo >&2 "  ERROR: Kernel configuration is invalid.";		\
echo >&2 "         include/generated/autoconf.h or include/config/auto.conf are missing.";\
echo >&2 "         Run 'make oldconfig && make prepare' on kernel src to fix it.";	\
echo >&2 ;							\
/bin/false)
make -f ./scripts/Makefile.build obj=/var/lib/dkms/evdi/1.10.1/build \
single-build= \
need-builtin=1 need-modorder=1
  gcc -Wp,-MMD,/var/lib/dkms/evdi/1.10.1/build/.evdi_platform_drv.o.d -nostdinc -I./arch/x86/include -I./arch/x86/include/generated  -I./include -I./arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/compiler-version.h -include ./include/linux/kconfig.h -include ./include/linux/compiler_types.h -D__KERNEL__ -fmacro-prefix-map=./= -Wall -Wundef -Werror=strict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -fshort-wchar -fno-PIE -Werror=implicit-function-declaration -Werror=implicit-int -Werror=return-type -Wno-format-security -std=gnu89 -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -fcf-protection=none -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mtune=generic -mno-red-zone -mcmodel=kernel -Wno-sign-compare -fno-asynchronous-unwind-tables -mindirect-branch=thunk-extern -mindirect-branch-register -fno-jump-tables -fno-delete-null-pointer-checks -Wno-frame-address -Wno-format-truncation -Wno-format-overflow -Wno-address-of-packed-member -O2 -fno-allow-store-data-races -Wframe-larger-than=2048 -fstack-protector -Wimplicit-fallthrough=5 -Wno-main -Wno-unused-but-set-variable -Wno-unused-const-variable -fno-stack-clash-protection -pg -mrecord-mcount -mfentry -DCC_USING_FENTRY -Wdeclaration-after-statement -Wvla -Wno-pointer-sign -Wno-stringop-truncation -Wno-zero-length-bounds -Wno-array-bounds -Wno-stringop-overflow -Wno-restrict -Wno-maybe-uninitialized -Wno-alloc-size-larger-than -fno-strict-overflow -fno-stack-check -fconserve-stack -Werror=date-time -Werror=incompatible-pointer-types -Werror=designated-init -Wno-packed-not-aligned -g -Iinclude/drm  -DMODULE  -DKBUILD_BASENAME='"evdi_platform_drv"' -DKBUILD_MODNAME='"evdi"' -D__KBUILD_MODNAME=kmod_evdi -c -o /var/lib/dkms/evdi/1.10.1/build/evdi_platform_drv.o /var/lib/dkms/evdi/1.10.1/build/evdi_platform_drv.c  ; ./tools/objtool/objtool orc generate  --module  --no-fp   --retpoline  --uaccess  /var/lib/dkms/evdi/1.10.1/build/evdi_platform_drv.o
  gcc -Wp,-MMD,/var/lib/dkms/evdi/1.10.1/build/.evdi_platform_dev.o.d -nostdinc -I./arch/x86/include -I./arch/x86/include/generated  -I./include -I./arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/compiler-version.h -include ./include/linux/kconfig.h -include ./include/linux/compiler_types.h -D__KERNEL__ -fmacro-prefix-map=./= -Wall -Wundef -Werror=strict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -fshort-wchar -fno-PIE -Werror=implicit-function-declaration -Werror=implicit-int -Werror=return-type -Wno-format-security -std=gnu89 -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -fcf-protection=none -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mtune=generic -mno-red-zone -mcmodel=kernel -Wno-sign-compare -fno-asynchronous-unwind-tables -mindirect-branch=thunk-extern -mindirect-branch-register -fno-jump-tables -fno-delete-null-pointer-checks -Wno-frame-address -Wno-format-truncation -Wno-format-overflow -Wno-address-of-packed-member -O2 -fno-allow-store-data-races -Wframe-larger-than=2048 -fstack-protector -Wimplicit-fallthrough=5 -Wno-main -Wno-unused-but-set-variable -Wno-unused-const-variable -fno-stack-clash-protection -pg -mrecord-mcount -mfentry -DCC_USING_FENTRY -Wdeclaration-after-statement -Wvla -Wno-pointer-sign -Wno-stringop-truncation -Wno-zero-length-bounds -Wno-array-bounds -Wno-stringop-overflow -Wno-restrict -Wno-maybe-uninitialized -Wno-alloc-size-larger-than -fno-strict-overflow -fno-stack-check -fconserve-stack -Werror=date-time -Werror=incompatible-pointer-types -Werror=designated-init -Wno-packed-not-aligned -g -Iinclude/drm  -DMODULE  -DKBUILD_BASENAME='"evdi_platform_dev"' -DKBUILD_MODNAME='"evdi"' -D__KBUILD_MODNAME=kmod_evdi -c -o /var/lib/dkms/evdi/1.10.1/build/evdi_platform_dev.o /var/lib/dkms/evdi/1.10.1/build/evdi_platform_dev.c  ; ./tools/objtool/objtool orc generate  --module  --no-fp   --retpoline  --uaccess  /var/lib/dkms/evdi/1.10.1/build/evdi_platform_dev.o
  gcc -Wp,-MMD,/var/lib/dkms/evdi/1.10.1/build/.evdi_sysfs.o.d -nostdinc -I./arch/x86/include -I./arch/x86/include/generated  -I./include -I./arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/compiler-version.h -include ./include/linux/kconfig.h -include ./include/linux/compiler_types.h -D__KERNEL__ -fmacro-prefix-map=./= -Wall -Wundef -Werror=strict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -fshort-wchar -fno-PIE -Werror=implicit-function-declaration -Werror=implicit-int -Werror=return-type -Wno-format-security -std=gnu89 -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -fcf-protection=none -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mtune=generic -mno-red-zone -mcmodel=kernel -Wno-sign-compare -fno-asynchronous-unwind-tables -mindirect-branch=thunk-extern -mindirect-branch-register -fno-jump-tables -fno-delete-null-pointer-checks -Wno-frame-address -Wno-format-truncation -Wno-format-overflow -Wno-address-of-packed-member -O2 -fno-allow-store-data-races -Wframe-larger-than=2048 -fstack-protector -Wimplicit-fallthrough=5 -Wno-main -Wno-unused-but-set-variable -Wno-unused-const-variable -fno-stack-clash-protection -pg -mrecord-mcount -mfentry -DCC_USING_FENTRY -Wdeclaration-after-statement -Wvla -Wno-pointer-sign -Wno-stringop-truncation -Wno-zero-length-bounds -Wno-arra [... cut ...]
kwi 20 17:19:20 teba-jobst kernel: evdi: [I] (card2) Edid property set
kwi 20 17:19:20 teba-jobst kernel: evdi: [W] evdi_painter_connect:882 (card1) Double connect - replacing 00000000e49d2db4 with 0000>
kwi 20 17:19:20 teba-jobst kernel: evdi: [I] (card1) Connected with Task 10649 (DesktopManagerE) of process 10636 (DisplayLinkMana)
kwi 20 17:19:20 teba-jobst kernel: evdi: [I] (card1) Connector state: connected
kwi 20 17:19:23 teba-jobst systemd-coredump[10811]: [🡕] Process 8750 (gnome-shell) of user 1000 dumped core.`

jacekobst1, 2022-04-20:
I checked journalctl after that screens blackout, and i found similiar informations as in the main issue:
`kwi 20 18:44:16 teba-jobst kernel: evdi: [I] (card2) Notifying display power state: on
kwi 20 18:44:16 teba-jobst kernel: usb 2-4.1: Disable of device-initiated U1 failed.
kwi 20 18:44:16 teba-jobst kernel: usb 2-4.1: Disable of device-initiated U2 failed.
kwi 20 18:44:16 teba-jobst kernel: cdc_ncm 2-4.1:1.5 enp0s20f0u4u1i5: unregister 'cdc_ncm' usb-0000:00:14.0-4.1, CDC NCM
kwi 20 18:44:16 teba-jobst NetworkManager[1636]: <info>  [1650473056.2252] device (enp0s20f0u4u1i5): state change: unavailable -> u>
kwi 20 18:44:16 teba-jobst kernel: usb 2-4.1: Set SEL for device-initiated U1 failed.
kwi 20 18:44:16 teba-jobst kernel: usb 2-4.1: Set SEL for device-initiated U2 failed.
kwi 20 18:44:16 teba-jobst kernel: usb 2-4.1: reset SuperSpeed USB device number 7 using xhci_hcd
kwi 20 18:44:16 teba-jobst kernel: usb 2-4.1: Warning! Unlikely big volume range (=767), cval->res is probably wrong.
kwi 20 18:44:16 teba-jobst kernel: usb 2-4.1: [4] FU [Mic Capture Volume] ch = 2, val = -4592/7680/16
kwi 20 18:44:16 teba-jobst kernel: usb 2-4.1: Warning! Unlikely big volume range (=672), cval->res is probably wrong.
kwi 20 18:44:16 teba-jobst kernel: usb 2-4.1: [7] FU [Dell USB Audio Playback Volume] ch = 6, val = -10752/0/16
kwi 20 18:44:16 teba-jobst kernel: cdc_ncm 2-4.1:1.5: MAC-Address: 9c:eb:e8:fc:7c:87
kwi 20 18:44:16 teba-jobst kernel: cdc_ncm 2-4.1:1.5: setting rx_max = 16384
kwi 20 18:44:16 teba-jobst kernel: cdc_ncm 2-4.1:1.5: setting tx_max = 16384
kwi 20 18:44:16 teba-jobst kernel: cdc_ncm 2-4.1:1.5 eth0: register 'cdc_ncm' at usb-0000:00:14.0-4.1, CDC NCM, 9c:eb:e8:fc:7c:87
kwi 20 18:44:16 teba-jobst kernel: usb 2-4.1: usbfs: process 10026 (ActiveCommandQu) did not claim interface 0 before use
kwi 20 18:44:16 teba-jobst NetworkManager[1636]: <info>  [1650473056.5992] manager: (eth0): new Ethernet device (/org/freedesktop/N>
kwi 20 18:44:16 teba-jobst upowerd[1540]: treating change event as add on /sys/devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4.1
kwi 20 18:44:16 teba-jobst upowerd[1540]: treating change event as add on /sys/devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4.1
kwi 20 18:44:16 teba-jobst systemd-udevd[11606]: Using default interface naming scheme 'v249'.
kwi 20 18:44:16 teba-jobst systemd-udevd[11606]: eth0: /etc/udev/rules.d/99-displaylink.rules:13 Failed to write ATTR{/sys/devices/>
kwi 20 18:44:16 teba-jobst systemd-udevd[11606]: eth0: /etc/udev/rules.d/99-displaylink.rules:14 Failed to write ATTR{/sys/devices/>
kwi 20 18:44:16 teba-jobst kernel: cdc_ncm 2-4.1:1.5 enp0s20f0u4u1i5: renamed from eth0
kwi 20 18:44:16 teba-jobst NetworkManager[1636]: <info>  [1650473056.6219] device (eth0): interface index 12 renamed iface from 'et>
kwi 20 18:44:16 teba-jobst systemd-udevd[11606]: eth0: Process '/bin/sh -c 'echo 65536 > /sys/module/cdc_ncm/parameters/ntb_max_tx'>
kwi 20 18:44:16 teba-jobst systemd-udevd[11606]: eth0: Process '/bin/sh -c 'echo 65536 > /sys/module/cdc_ncm/parameters/ntb_max_rx'>
kwi 20 18:44:16 teba-jobst NetworkManager[1636]: <info>  [1650473056.6320] device (enp0s20f0u4u1i5): state change: unmanaged -> una>
kwi 20 18:44:16 teba-jobst NetworkManager[1636]: <info>  [1650473056.6337] settings: (enp0s20f0u4u1i5): created default wired conne>
kwi 20 18:44:16 teba-jobst systemd-udevd[11608]: tx-0: Process '/bin/sh -c 'echo 65536 > /sys/module/cdc_ncm/parameters/ntb_max_tx'>
kwi 20 18:44:16 teba-jobst systemd-udevd[11609]: rx-0: Process '/bin/sh -c 'echo 65536 > /sys/module/cdc_ncm/parameters/ntb_max_tx'>
kwi 20 18:44:16 teba-jobst systemd-udevd[11608]: tx-0: Process '/bin/sh -c 'echo 65536 > /sys/module/cdc_ncm/parameters/ntb_max_rx'>
kwi 20 18:44:16 teba-jobst systemd-udevd[11609]: rx-0: Process '/bin/sh -c 'echo 65536 > /sys/module/cdc_ncm/parameters/ntb_max_rx'>
kwi 20 18:44:16 teba-jobst systemd-udevd[11609]: Using default interface naming scheme 'v249'.
kwi 20 18:44:16 teba-jobst systemd-udevd[11609]: enp0s20f0u4u1i5: /etc/udev/rules.d/99-displaylink.rules:13 Failed to write ATTR{/s>
kwi 20 18:44:16 teba-jobst systemd-udevd[11609]: enp0s20f0u4u1i5: /etc/udev/rules.d/99-displaylink.rules:14 Failed to write ATTR{/s>
kwi 20 18:44:19 teba-jobst ModemManager[1610]: <info>  [base-manager] couldn't check support for device '/sys/devices/pci0000:00/00>
kwi 20 18:44:23 teba-jobst kernel: evdi: [I] (card2) Disconnected from Task 7121 (DesktopManagerE) of process 7111 (DisplayLinkMana)
kwi 20 18:44:23 teba-jobst kernel: evdi: [I] (card2) Removing i2c adapter bus number 9
kwi 20 18:44:23 teba-jobst kernel: evdi: [I] (card2) Closed by Task 7121 (DesktopManagerE) of process 7111 (DisplayLinkMana)
kwi 20 18:44:23 teba-jobst kernel: evdi: [W] evdi_painter_connect:882 (card1) Double connect - replacing 0000000030d666e2 with 0000>
kwi 20 18:44:23 teba-jobst kernel: evdi: [I] (card1) Connected with Task 7121 (DesktopManagerE) of process 7111 (DisplayLinkMana)
kwi 20 18:44:23 teba-jobst kernel: evdi: [I] (card1) Connector state: connected
kwi 20 18:44:23 teba-jobst kernel: evdi: [I] (card2) Opened by Task 7121 (DesktopManagerE) of process 7111 (DisplayLinkMana)
kwi 20 18:44:24 teba-jobst kernel: evdi: [I] (card2) Added i2c adapter bus number 9
kwi 20 18:44:24 teba-jobst kernel: evdi: [I] (card2) Connected with Task 7121 (DesktopManagerE) of process 7111 (DisplayLinkMana)
kwi 20 18:44:24 teba-jobst kernel: evdi: [I] (card2) Connector state: connected
[... cut ...]
kwi 20 18:44:24 teba-jobst gnome-shell[8035]: Window manager warning: Overwriting existing binding of keysym 39 with keysym 39 (key>
kwi 20 18:44:24 teba-jobst gnome-shell[8035]: Window manager warning: Overwriting existing binding of keysym 31 with keysym 21 (key>
kwi 20 18:44:37 teba-jobst gnome-shell[8035]: meta_window_set_stack_position_no_sync: assertion 'window->stack_position >= 0' failed
kwi 20 18:44:37 teba-jobst gnome-shell[8035]: meta_window_set_stack_position_no_sync: assertion 'window->stack_position >= 0' failed
`

elguero, 2022-05-06:
Here is someone mentioning that they are getting blackouts and have been working with the folks there in the evdi repo.

https://github.com/DisplayLink/evdi/issues/347

elguero, 2022-07-22:
Can you try out the latest release and provide feedback?

jacekobst1, 2022-07-22:
Sorry, it was too annoying and I couldn't fix it, so I already switched to Ubuntu, and the problem has gone.
There was also the blackouts issue, but I figured it out, after making some changes to Pulseaudio (or some similar sound package) configuration file.

closed 2022-07-22 by jacekobst1