displaylink-rpm/displaylink-rpm #204: Regression - kernel 5.17.5-200.fc35.x86_64 mode rejection.

chiddekel, 2022-05-05:
On kernel 5.17.5-200.fc35.x86_64 mode rejection:
(...)
[W] evdi_mode_valid:75 Mode 1900x1200@60 rejected
(...)
Why default mode is 75 and not the range of for example starting from 60?

cdonatom, 2022-05-06:
I think this is an issue with the kernel provided on Fedora 36. I am using akmods and it refuses to compile for this particular version whilst it does work and compile for 5.17.4.

Olyol95, 2022-05-06:
I'm seeing the same mode rejection issue on 5.17.4-200.fc35.x86_64, with [evdi](https://github.com/DisplayLink/evdi/commits/devel) compiled from `39da217`.

elguero, 2022-05-06:
An issue was created by @chiddekel upstream.

https://github.com/DisplayLink/evdi/issues/357

elguero, 2022-07-22:
Looking over the comments on issue at DisplayLink/evdi, it looks like this commit is an attempt to fix the problem: https://github.com/DisplayLink/evdi/commit/dbe48b051682c6dc4e3688b35d599c90d8dd5a4f

Not sure if anyone wants to try it out.  DisplayLink just released evdi 1.12.0.  That probably means a newer DisplayLink driver release is in the works.

chiddekel, 2022-07-23:
Confirm - evdi 1.12.0 work on Fedora 35 with kernel 5.18.11-100.fc35.x86_64.

closed 2022-08-14 by elguero