displaylink-rpm/displaylink-rpm #205: Fedora 36 Support

michael-manley, 2022-05-11:
Since Fedora 36 was released yesterday, I will mention probably only needs a change to your guy's build automation to add fc36, but I did take the Fedora 35 SRPM and built it successfully on Fedora 36.

codehammer10, 2022-05-14:
I was able to get Fedora 35 SRPM to work on Fedora 36 as well. Had some minor issues with the monitors powering on and off and i used [https://www.displaylink.org/forum/showthread.php?t=65476&page=2](this) solution to resolve my issue.

ebousse, 2022-05-19:
@codehammer10 If you have time could you point to the exact solution from the page you linked? It seems to contain several attempts/solutions and I am not sure what to try there.

codehammer10, 2022-05-25:
> 

@ebousse the comment on that page from edclement

"After a few days of struggling, I managed to find a solution on Ubuntu 17.10. Hope this works for everyone having problems with the D6000 dock. Below format is markdown (to lazy to convert it)

## Instructions for fixing the blanking monitor on dock issue within Ubuntu.
**Make sure to reboot after making the changes!**

#### Disable USB AutoSuspend
* `sudo gedit /etc/default/tlp` and change the below value from `1` to `0`

```
USB_AUTOSUSPEND=0
```

* `sudo gedit /etc/udev/rules.d/01-usb-dock.rules`. This will create a new file.

```
ACTION=="add", SUBSYSTEM=="usb", ATTR{idVendor}=="1bcf", ATTR{idProduct}=="2b95", TEST=="power/control", ATTR{power/control}="on"
ACTION=="add", SUBSYSTEM=="usb", ATTR{idVendor}=="1bcf", ATTR{idProduct}=="0005", TEST=="power/control", ATTR{power/control}="on"
ACTION=="add", SUBSYSTEM=="usb", ATTR{idVendor}=="1d6b", ATTR{idProduct}=="0002", TEST=="power/control", ATTR{power/control}="on"
ACTION=="add", SUBSYSTEM=="usb", ATTR{idVendor}=="1d6b", ATTR{idProduct}=="0003", TEST=="power/control", ATTR{power/control}="on"
ACTION=="add", SUBSYSTEM=="usb", ATTR{idVendor}=="04d9", ATTR{idProduct}=="0143", TEST=="power/control", ATTR{power/control}="on"
ACTION=="add", SUBSYSTEM=="usb", ATTR{idVendor}=="04f3", ATTR{idProduct}=="24a1", TEST=="power/control", ATTR{power/control}="on"
ACTION=="add", SUBSYSTEM=="usb", ATTR{idVendor}=="05e3", ATTR{idProduct}=="0608", TEST=="power/control", ATTR{power/control}="on"
ACTION=="add", SUBSYSTEM=="usb", ATTR{idVendor}=="05e3", ATTR{idProduct}=="0610", TEST=="power/control", ATTR{power/control}="on"
ACTION=="add", SUBSYSTEM=="usb", ATTR{idVendor}=="05e3", ATTR{idProduct}=="0620", TEST=="power/control", ATTR{power/control}="on"
ACTION=="add", SUBSYSTEM=="usb", ATTR{idVendor}=="06c4", ATTR{idProduct}=="c411", TEST=="power/control", ATTR{power/control}="on"
ACTION=="add", SUBSYSTEM=="usb", ATTR{idVendor}=="17e9", ATTR{idProduct}=="6006", TEST=="power/control", ATTR{power/control}="on"
```

#### Disable Display Power Management
* `sudo gedit /etc/X11/xorg.conf` Append the below options to the file.

```
Section "Monitor"
Identifier "DVI-I-3-2"
Option "DPMS" "false"
EndSection

Section "Monitor"
Identifier "DVI-I-2-1"
Option "DPMS" "false"
EndSection
```

#### Disable audio suspend
* `sudo gedit /etc/pulse/default.pa`. Comment out the line indicated below.

```
### Automatically suspend sinks/sources that become idle for too long
# load-module module-suspend-on-idle
```

#### Disale USB wake in BIOS
Boot into the BIOS by pressing the F12 key as the machine turns on. Under `Power Management` >> `USB Wake Support` uncheck both boxes."

PR #213 (2022-06-05): Fedora 36 has been released; enable CI
Fixes #205

Note that this almost certainly depends on #207 to cover the updated kernel in Fedora 36
commit 6faa0de2 (2022-06-05): Fedora 36 has been released; enable CI (#213)

Fixes #205
closed 2022-06-05 by elguero