displaylink-rpm/displaylink-rpm #206: Install fails on RHEL 8, postinstall scriptlet error.

tbalbers, 2022-05-17:
Hi,

Using Release DisplayLink 5.5.0 and Bundled EVDI 1.10.1

On RHEL 8.6 x86_64, fresh install, no security profile.

I face an issue that shows up like so:

dnf install displaylink-1.10.1-1.x86_64.rpm
---------------------------------------------------------------------------------------------------------
...
Running transaction
  Preparing        :                                                                            1/1 
  Installing       : dkms-3.0.3-1.el8.noarch                                                    1/2 
  Running scriptlet: dkms-3.0.3-1.el8.noarch                                                    1/2 
  Installing       : displaylink-1.10.1-1.x86_64                                                2/2 
  Running scriptlet: displaylink-1.10.1-1.x86_64                                                2/2 
Job for displaylink-driver.service failed because the control process exited with error code.
See "systemctl status displaylink-driver.service" and "journalctl -xe" for details.
warning: %post(displaylink-1.10.1-1.x86_64) scriptlet failed, exit status 1

Error in POSTIN scriptlet in rpm package displaylink
  Verifying        : dkms-3.0.3-1.el8.noarch                                                    1/2 
  Verifying        : displaylink-1.10.1-1.x86_64                                                2/2 
Installed products updated.

Installed:
  displaylink-1.10.1-1.x86_64                        dkms-3.0.3-1.el8.noarch                       

Complete!
---------------------------------------------------------------------------------------------------------

If I check the displaylink log file, I see:
---------------------------------------------------------------------------------------------------------
...
Building module:
cleaning build area...
make -j12 KERNELRELEASE=4.18.0-372.9.1.el8.x86_64 all INCLUDEDIR=/lib/modules/4.18.0-372.9.1.el8.x86_64/build/include KVERSION=4.18.0-372.9.1.el8.x86_64 DKMS_BUILD=1...(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.18.0-372.9.1.el8.x86_64 (x86_64)
Consult /var/lib/dkms/evdi/1.10.1/build/make.log for more information.
---------------------------------------------------------------------------------------------------------

And in /var/lib/dkms/evdi/1.10.1/build/make.log 
---------------------------------------------------------------------------------------------------------
DKMS make.log for evdi-1.10.1 for kernel 4.18.0-372.9.1.el8.x86_64 (x86_64)
Tue 17 May 13:12:29 CEST 2022
make KBUILD_VERBOSE=1 M=/var/lib/dkms/evdi/1.10.1/build SUBDIRS=/var/lib/dkms/evdi/1.10.1/build SRCR
OOT=/var/lib/dkms/evdi/1.10.1/build CONFIG_MODULE_SIG= -C /lib/modules/4.18.0-372.9.1.el8.x86_64/bui
ld modules
make[1]: Entering directory '/usr/src/kernels/4.18.0-372.9.1.el8.x86_64'
test -e include/generated/autoconf.h -a -e include/config/auto.conf || (                \
echo >&2;                                                       \
echo >&2 "  ERROR: Kernel configuration is invalid.";           \
echo >&2 "         include/generated/autoconf.h or include/config/auto.conf are missing.";\
echo >&2 "         Run 'make oldconfig && make prepare' on kernel src to fix it.";      \
echo >&2 ;
...
/var/lib/dkms/evdi/1.10.1/build/evdi_modeset.c:397:19: error: initialization of ‘void (*)(struct drm_plane *, struct drm_atomic_state *)’ from incompatible pointer type ‘void (*)(struct drm_plane *, struct drm_plane_state *)’ [-Werror=incompatible-pointer-types]
  .atomic_update = evdi_plane_atomic_update,
                   ^~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/evdi/1.10.1/build/evdi_modeset.c:397:19: note: (near initialization for ‘evdi_plane_helper_funcs.atomic_update’)
/var/lib/dkms/evdi/1.10.1/build/evdi_modeset.c:401:16: error: ‘drm_gem_fb_prepare_fb’ undeclared here (not in a function); did you mean ‘drm_gem_fb_create’?
  .prepare_fb = drm_gem_fb_prepare_fb
                ^~~~~~~~~~~~~~~~~~~~~
                drm_gem_fb_create
/var/lib/dkms/evdi/1.10.1/build/evdi_modeset.c:406:19: error: initialization of ‘void (*)(struct drm_plane *, struct drm_atomic_state *)’ from incompatible pointer type ‘void (*)(struct drm_plane *, struct drm_plane_state *)’ [-Werror=incompatible-pointer-types]
  .atomic_update = evdi_cursor_atomic_update,
                   ^~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/evdi/1.10.1/build/evdi_modeset.c:406:19: note: (near initialization for ‘evdi_cursor_helper_funcs.atomic_update’)
   ./tools/objtool/objtool orc generate  --module --no-fp  --retpoline "/var/lib/dkms/evdi/1.10.1/build/.tmp_evdi_platform_dev.o";
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:315: /var/lib/dkms/evdi/1.10.1/build/evdi_modeset.o] Error 1
make[2]: *** Waiting for unfinished jobs....
   ./tools/objtool/objtool orc generate  --module --no-fp  --retpoline "/var/lib/dkms/evdi/1.10.1/build/.tmp_evdi_connector.o";
   ./tools/objtool/objtool orc generate  --module --no-fp  --retpoline "/var/lib/dkms/evdi/1.10.1/build/.tmp_evdi_encoder.o";
   ./tools/objtool/objtool orc generate  --module --no-fp  --retpoline "/var/lib/dkms/evdi/1.10.1/build/.tmp_evdi_drm_drv.o";
/var/lib/dkms/evdi/1.10.1/build/evdi_painter.c: In function ‘evdi_log_pixel_format’:
/var/lib/dkms/evdi/1.10.1/build/evdi_painter.c:733:29: error: storage size of ‘format_name’ isn’t known
  struct drm_format_name_buf format_name;
                             ^~~~~~~~~~~
/var/lib/dkms/evdi/1.10.1/build/evdi_painter.c:735:2: error: implicit declaration of function ‘drm_get_format_name’; did you mean ‘drm_get_format_info’? [-Werror=implicit-function-declaration]
  drm_get_format_name(pixel_format, &format_name);
  ^~~~~~~~~~~~~~~~~~~
  drm_get_format_info
/var/lib/dkms/evdi/1.10.1/build/evdi_painter.c:733:29: warning: unused variable ‘format_name’ [-Wunused-variable]
  struct drm_format_name_buf format_name;
                             ^~~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:315: /var/lib/dkms/evdi/1.10.1/build/evdi_painter.o] Error 1
   ./tools/objtool/objtool orc generate  --module --no-fp  --retpoline "/var/lib/dkms/evdi/1.10.1/build/.tmp_evdi_fb.o";
   ./tools/objtool/objtool orc generate  --module --no-fp  --retpoline "/var/lib/dkms/evdi/1.10.1/build/.tmp_evdi_gem.o";
make[1]: *** [Makefile:1577: _module_/var/lib/dkms/evdi/1.10.1/build] Error 2
make[1]: Leaving directory '/usr/src/kernels/4.18.0-372.9.1.el8.x86_64'
make: *** [Makefile:33: all] Error 2
---------------------------------------------------------------------------------------------------------

I tried cleaning out as per the instructions, but that didn't change anything.
OS: Red Hat Enterprise Linux
Version: 8.6
Kernel ver.: 4.18.0-372.9.1.el8.x86_64

[displaylink.log](https://github.com/displaylink-rpm/displaylink-rpm/files/8707747/displaylink.log)
[make.log](https://github.com/displaylink-rpm/displaylink-rpm/files/8707750/make.log)

elguero, 2022-05-17:
Thank you for the info.

It looks like there may have been some backports put into the latest 8.6 EL kernel.  We will have to look at making changes to EVDI in order for it to compile on EL8.

elguero, 2022-07-22:
This should be fixed in the latest release.  Some patches have been added to support EL 8 and EL 9 distros.

closed 2022-07-22 by elguero

CryptoKiddies, 2023-06-08:
This issue still occurs in Fedora 37